meerk40t/meerk40t #389: Items merged between raster operations when saving and reloading project.

fr-og, 2021-03-12:
Items are merged between raster operations upon saving and reloading a project.

Tested in versions 0.6.20, 0.6.21 & 0.7.0 beta 12

See image for example. On the left is configuration before saving, on the right is configuration after save and reloading.

[image]

tatarize, 2021-03-12:
#209 Yeah, they reclassify wrong or rather don't save the very specific classification data. I'll fix this likely in 0.7.0 sometime fairly soon.  Saving and loading  of somethings is still an outstanding issue there and referencing the data more correctly for many files is going to be strongly preferred.

tatarize, 2022-07-20:
This is corrected in 0.8.x since it saves the references and restores them correctly.

closed 2022-07-20 by tatarize