meerk40t/meerk40t #391: Feature Request:  Double Click on Camera updates Bed Image

inspectionsbybob, 2021-03-18:
Just trying to save some keystrokes.  Don't know how hard it would be to implement.

Because the Camera Window being open still seems to be the pre-cursor to "pauses" and "Camera Not Found" and the occasional "USB Failure" I try to keep the Camera window closed as much as possible.  With that in mind, I would suggest an option for a quick way to update the Bed Photo.

**Double Click on the Camera Icon on the top toolbar to update the Bed Photo**.  
This would NOT open the camera window, it would, Put a msg on the screen, in the background, capture the camera session, let it focus (if needed), update the screen msg,  update the Bed Photo. and release the camera, and update the screen msg, then delete the screen msg.

tatarize, 2021-03-19:
In 0.7.0 this is already a thing. The window in 0.7.0 isn't actually the camera which runs independently. In console "camera0 open" will start the camera without opening the window. Then `camera background` will update the background image from the camera, without the window opening. In theory you could even string a bunch of stuff together. `camera export image wizard Gravy vectrace fill none stroke red engrave -s 12.2` would take a camera export frame make it a image, then apply rasterwizard script Gravy to it, turn that into a vector traced outline of the image, fill that with no fill and set the stroke to red. Then put that in an engrave operation with a speed of 12.2mms.

For your purposes if you did `camera0 open` then `camera0 background` it would do that. You could even bind a key to update the frame from the camera without the camera window open.

inspectionsbybob, 2021-03-19:
Great minds...
How much processor does the Camera routine take?  I have a LOT less
problems when I don't have a camera window open...

Bob

tatarize, 2021-03-19:
I dunno, but there's other resources it could cause issues. You could check the CPU it depends a bit on other things and takes up some GPU, etc.

inspectionsbybob, 2021-03-20:
I tried to look at the "Performace" tab in Win10, but there was a long list
of USB functions, and none of them looked like it would tell me what I
wanted to know.  I may play a bit tomorrow PM and see if I can see
something that might be out of place...

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

closed 2021-08-13 by tatarize