meerk40t/meerk40t #393: Warn about closing app during burn

Sophist-UK, 2021-03-21:
I am truly an idiot - once is a mistake, twice is stupidity. Twice now I went to close the spooler window and instead closed the main window.

Can we please have an "Are you sure?" confirmation if we are closing Meerk40t whilst a burn is in progress?

tatarize, 2021-03-23:
There used to be such a setting, but it got lost in one of the updates/conversions. It also has to go kinda out of the way to kill a process in progress. Otherwise the GUI closes down but the thread stays there sending data without fully killing the processes. I'll be sure to add it back.

tatarize, 2021-08-03:
This is in the 0.6.23 branch.

tatarize, 2021-09-11:
Published.

https://github.com/meerk40t/meerk40t/releases/tag/0.6.23

closed 2021-09-11 by tatarize