meerk40t/meerk40t #399: usb connection failed

Irishangell, 2021-04-09:
get the message stating usb connection failed. uSB connection works fine with k40 whisperer. have reinstalled the drivers and deleted and redownloaded meerk40t to no avail. only happens since i moved to v7 of the software. usb works fine. meerk40t log below

Using LibUSB to connect.
Attempting connection to USB.
Attempting connection to USB.
Using LibUSB to connect.
Finding devices.
K40 device detected:
DEVICE ID 1a86:5512 on Bus 000 Address 001 =================
	 bLength                :   0x12 (18 bytes)
	 bDescriptorType        :    0x1 Device
	 bcdUSB                 :  0x110 USB 1.1
	 bDeviceClass           :   0xff Vendor-specific
	 bDeviceSubClass        :    0x0
	 bDeviceProtocol        :    0x2
	 bMaxPacketSize0        :    0x8 (8 bytes)
	 idVendor               : 0x1a86
	 idProduct              : 0x5512
	 bcdDevice              :  0x304 Device 3.04
	 iManufacturer          :    0x0 
	 iProduct               :    0x0 
	 iSerialNumber          :    0x0 
	 bNumConfigurations     :    0x1
	  CONFIGURATION 1: 100 mA ==================================
	   bLength              :    0x9 (9 bytes)
	   bDescriptorType      :    0x2 Configuration
	   wTotalLength         :   0x27 (39 bytes)
	   bNumInterfaces       :    0x1
	   bConfigurationValue  :    0x1
	   iConfiguration       :    0x0 
	   bmAttributes         :   0x80 Bus Powered
	   bMaxPower            :   0x32 (100 mA)
	    INTERFACE 0: Vendor Specific ===========================
	     bLength            :    0x9 (9 bytes)
	     bDescriptorType    :    0x4 Interface
	     bInterfaceNumber   :    0x0
	     bAlternateSetting  :    0x0
	     bNumEndpoints      :    0x3
	     bInterfaceClass    :   0xff Vendor Specific
	     bInterfaceSubClass :    0x1
	     bInterfaceProtocol :    0x2
	     iInterface         :    0x0 
	      ENDPOINT 0x82: Bulk IN ===============================
	       bLength          :    0x7 (7 bytes)
	       bDescriptorType  :    0x5 Endpoint
	       bEndpointAddress :   0x82 IN
	       bmAttributes     :    0x2 Bulk
	       wMaxPacketSize   :   0x20 (32 bytes)
	       bInterval        :    0x0
	      ENDPOINT 0x2: Bulk OUT ===============================
	       bLength          :    0x7 (7 bytes)
	       bDescriptorType  :    0x5 Endpoint
	       bEndpointAddress :    0x2 OUT
	       bmAttributes     :    0x2 Bulk
	       wMaxPacketSize   :   0x20 (32 bytes)
	       bInterval        :    0x0
	      ENDPOINT 0x81: Interrupt IN ==========================
	       bLength          :    0x7 (7 bytes)
	       bDescriptorType  :    0x5 Endpoint
	       bEndpointAddress :   0x81 IN
	       bmAttributes     :    0x3 Interrupt
	       wMaxPacketSize   :    0x8 (8 bytes)
	       bInterval        :    0x1
Config Set
Config Set: Success
Setting Active Config
Active Config: Success.
Kernel detach: Not Implemented.
Attempting to claim interface.
Interface claim: Failed. (Interface is in use.)
Attempting to release interface.
Interface released.
Attempting to claim interface.
Interface claim: Failed. (Interface is in use.)
Connection to USB failed.

Connection to USB failed.

tatarize, 2021-04-09:
Yeah, interface is in use. Means something claimed the USB and didn't let it go. This is either MeerK40t screwed up previously and the hook to the USB got stuck, you are running some other thing like Whisperer or another MeerK40t that still has the hook or something. It's been known to happen that MeerK40t might have some thread get stuck and not fully close from the previous iteration, so check your tasks and make sure there's not a second MeerK40t in there, that or hope it forces reset of the USB device there which sometimes will just happen kinda randomly.

tatarize, 2021-04-09:
There is however some kind of USB issue we've been hunting a bit, and this might be part of it.  Make sure this is beta 19+. There was a bad connection handler in there in beta-18 and maybe beta-17. https://github.com/meerk40t/meerk40t/releases/tag/0.7.0-beta19

I'm not sure other than something else currently connected to it like whisperer what would do that. Bob reports that sometimes he gets his stuck and uses the Reset and Dispose stuff in Whisperer to fix it. Does that fix it? If so I'll implement those commands.

Irishangell, 2021-04-09:
I restarted the pc and that seems to have fixed the usb issue except now the bloody thing won't run the laser. It says everything is connected but the job just goes to spool and sits there.it doesn't automatically.home before a job either even when the command is there.  I've given up in frustration tonight after 2 hours pissing about with it but I will try again tomorrow and see if I run into any further usb issues or anything else at which point I'll open a different issue.

tatarize, 2021-04-09:
If everything went to the spooler without processing odds are good either the usb didn't connect or the interpreter didn't start up. You can hit the buttons in Controller and it'll usually force the connections and make sure things are processing the data correctly.

closed 2021-05-05 by tatarize