meerk40t/meerk40t #402: Feature request: dithered ppi

kgwatson, 2021-04-20:
I've been experimenting with different settings and noticed at some speed/ppi combinations a repeating pattern is visible. Would it be possible to add a dithering option to the ppi settings so that any resultant patterns are less noticeable?

tatarize, 2021-04-20:
Hm. I tended to *add* more patterning with dot length etc, so you could do more interesting ppi patterns, and cut dashed lines etc. I guess it would be possible to avoid patterning by reframing how ppi works. It adds the ppi value to a counter, then if above 1000 fires the laser for that dot. So at 100 ppi it *will* always fire a laser exactly every 10th step. It would be possible to instead randomize the PPI so that rather than add ppi, it simply has a ppi/1000 chance of firing the laser. At small enough scales it would seem quite likely to avoid any inadvertent patterning caused by the highly deterministic nature of how ppi works.

kgwatson, 2021-04-20:
Random might work, but what about some sort of Floyd-Steinberg error diffusion? That should remove any chance of patterning.

tatarize, 2021-04-20:
While Floyd-Steinberg works it's built around the idea of dithering a 2d image. This is certainly a step with any raster preprocessing event. And it has a pretty good ability to avoid artifacting with fairly minimal amounts of work, though there are better ones out there. For PPI it doesn't have this sort of look-ahead within Eulerian space. PPI just knows there's an official on-off laser signal and makes a carry ahead patterning. In effect, it's dithering in 1 dimension. And for some things like drawing a rectangle outline that's the only method of dithering that is available. It seems like carry ahead with a degree of randomness would also be quite possible where I could tweak the math to adjust the amount of random pattern breaking involved but the dithering would need to be 1d plot processing adjusting the on/off values sent to the beam.

kgwatson, 2021-04-21:
Sounds good.

tatarize, 2021-05-20:
The Option for Random PPI was added to the various guis. No underlying code for this exists however. But, preparations are being made.

closed 2024-10-04 by jpirnay