meerk40t/meerk40t #404: Camera LAGS or is slow

inspectionsbybob, 2021-04-26:
Had a chance to play with 6.22 a bit more as other project/devices are busy.

1. It looses "Perspective Corrections" between power cycles.  Not much, but enough to require re-calibration between power cycles.  
2. The camera is SLOW.  Even after a reboot and opening up a new, empty, session to re-calibrate the perspective, dragging the camera around was painfully slow, and trying to get the corner marks onto my calibration sheet with dots in the corners was tough being soo laggy/slow...

tatarize, 2021-04-27:
I'll check the perspective corrections thing. I can't really do much with the camera speed. I assume you fiddled with the Framerate to see if there's a faster one or something? It by default doesn't update things that fast, but the slider is framerate.

inspectionsbybob, 2021-04-27:
It wasn't the camera, it was MeerK40T, I would grab something and start to
drag it, and nothing would happen for 3-6 seconds, and then it move... to
where I was 5 seconds ago... most frustrating

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

tatarize, 2021-04-28:
At the lowest setting in the framerate slider it actually will capture a frame every 5 seconds.

inspectionsbybob, 2021-04-28:
it was a combination or slow frame rate and a very very complicated (lots
of pieces) object... redid the object as a single raster rather than a
hundred tiny pieces to be individually rendered and it was MUCH a happier...

The artist had sent a SVG with the text as individual line segment since
she knew we didn't have the font she used...  I told her to send me the
text as a PNG image I could drop into the workpiece whole, and then add the
CUT lines... went from over 200 objects to 8.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-05-03:
MOST of it was user issues...

closed 2021-05-03 by inspectionsbybob