meerk40t/meerk40t #406: How do we "Tune" the USB controller so it pauses less?

inspectionsbybob, 2021-05-03:
As I had a medium-sized project with a lot of Rastering and Cutting of small things, stuff that keeps the controller really busy.

It paused on both Rastering operations, and also on CUT operations  (lots of circles)

I was watching the USB queue or buffer.... and of 900 bytes it spent a lot of time at 902... and when it got STUCK, I would disconnect the USB (using the buttons) and reconnect and it would pick up where it left off without missing a beat...

The question is "How to -tune- the USB interface for better throughput..."

joerlane, 2021-05-03:
What does the "Rejected Packets" count in the Controller window show? 

Also; what is the total length of the USB cable used to connect to the controller board?

inspectionsbybob, 2021-05-03:
Rejected packets normally sits at ZERO

Rather than just have a USB extension, I have an ACTIVE REPEATER cable that
is about 16'.

joerlane, 2021-05-04:
Is there a file that can reproduce this problem reliably?

inspectionsbybob, 2021-05-04:
Try this file, it paused, stopped, paused and then paused again.

Rejected was ZERO

Bob Sisson

inspectionsbybob, 2021-05-04:
I had to make 50+ of these medallions.  So I would run them this side up,
Flip/left shift, and run the Raster ONLY. So this got run 10 times
The image should be RASTERED, I had to edit it each time I restarted
Meerk40T even though I saved it as Raster, not image...

Bob Sisson

joerlane, 2021-05-04:
Neither of those messages contained an attachment. You can email it to tatarize or whatever if it won't attach directly here. Just looking to run it on another machine to reproduce the issue reliably, or look into what else might ultimately be involved.

tatarize, 2021-05-04:
Is it a classic stutter? If so you need to either run the image at a higher DPI or slower. The buffer can only get new commands so often and when it runs out because the commands take like no time at all to execute you get the buffer to bottom out which causes a stutter. If your long usb cable isn't at fault it's likely that you're running a high density raster very quickly. This means the head moves quickly and takes very little time to finish.

https://www.youtube.com/watch?v=blhbbq22sGU

inspectionsbybob, 2021-05-04:
question is then about slowing down, means less power as running slower
means more time on target, deeper burns, etc...  I just got the material
dialed in at the desired power level, speed, step and PPI.

Question, at 1000 (100%) PPI, can't we NOT try to cycle the beam, as it
should be ON all the time, ie for cuts...  I can see slowing the raster
down a bit, along with the PPI so I don't burn things, but cuts should be
continuous ON and not taxing the poor little boards brain...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

tatarize, 2021-05-05:
> Question, at 1000 (100%) PPI, can't we NOT try to cycle the beam, as it
> should be ON all the time, ie for cuts...  I can see slowing the raster
> down a bit, along with the PPI so I don't burn things, but cuts should be
> continuous ON and not taxing the poor little boards brain...

It should be left at 1000 for cuts. You can slow things down as needed. The PPI doesn't mess with the board much, unless it causes stutter but it's not very taxing to just flicker the laser. The option is there because you can use it. Cut and Engrave are really the same operation except for the cut-inner-first option. There's no real utility in setting PPI to anything other than 1000 for cuts.

You can affect the only property of laser in several different ways. The only property of lasers is how much laser and where. Laser per unit of area. If you slow it down it does the same thing as increasing the power. If you turn up the power it does the same thing as slowing it down. If you turn up the PPI it does the same thing (mostly) as making the laser go faster. So if you need to cut something more lightly and you can't make your laser cut faster (because it'll tear itself apart). PWM for boards that support that does the same thing as reducing the power any other way at your disposal. There's also focus that with expertise you can play around with, but that's basically it.

You can usually turn the power down a bit and slow the laser down and get about the same results.

inspectionsbybob, 2021-05-05:
I understand all that... I was just wondering why if we cut at 1000 why we
need to by cycling at 1000 times a second rather than just "ON".  As I had
pauses when cutting at 1000ppi of circles which should be easy on its
little brian.... I can understand pauses in uneven edges with a zillion
way-points, but a circle, or an arc, shouldn't be causing pauses...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

tatarize, 2021-05-05:
1000ppi doesn't really mean 1000 cycles it means 1000 on bits per 1000 bits. It is *just on*.

tatarize, 2021-05-20:
In 0.7.0 there is a "Shift Pulses" option in Preferences. This should actually increase the ceiling of the raster stutters about 3x. So it should stutter at around 300 mm/s or so at max. Assuming it works correctly. This may distort the raster by up to 1 mil in a right or left direction. Tests of the quality have not been done.

closed 2021-08-12 by tatarize