meerk40t/meerk40t #407: Crash report MeerK40t 0.6.22 on UBUNTU 20.04

Olivier33m, 2021-05-04:
Hello,

MeerK40t crash log. Version: 0.6.22
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "/home/olivier/Appilcations/meerk40t-master/DeviceManager.py", line 229, in on_button_properties
    uid = self.devices_list.GetItem(item).Text
wx._core.wxAssertionError: C++ assertion ""item.m_itemId >= 0 && (size_t)item.m_itemId < GetItemCount()"" failed at /home/wxpy/wxPython-4.1.1/ext/wxWidgets/src/generic/listctrl.cpp(3758) in GetItem(): invalid item index in GetItem

tatarize, 2021-05-05:
https://github.com/meerk40t/meerk40t/commit/6be250d7b44311e6a8de5e322a3ede74b35669a5

Had fixed the copy of that bug in 0.7.x just yesterday when I did the bug logs. This will be fixed in 0.6.23.

tatarize, 2021-09-11:
This is published now.

https://github.com/meerk40t/meerk40t/releases/tag/0.6.23

closed 2021-09-11 by tatarize