meerk40t/meerk40t #408: SVG CLI error: object of type 'LaserOperation' has no len()

giladrom, 2021-05-06:
Trying to use the command line options to parse and execute a job:

```
meerk40t -zaS 100 ~/circle.svg
```

I get the following error:

```
Traceback (most recent call last):
  File "/usr/local/bin/meerk40t", line 8, in <module>
    sys.exit(run())
  File "/usr/local/lib/python3.9/site-packages/meerk40t/main.py", line 322, in run
    device("plan validate\n")
  File "/usr/local/lib/python3.9/site-packages/meerk40t/kernel.py", line 136, in __call__
    return self._kernel.console(data)
  File "/usr/local/lib/python3.9/site-packages/meerk40t/kernel.py", line 2231, in console
    self._console_parse(command, channel=self._console_channel)
  File "/usr/local/lib/python3.9/site-packages/meerk40t/kernel.py", line 2316, in _console_parse
    data, remainder, input_type = command_funct(
  File "/usr/local/lib/python3.9/site-packages/meerk40t/kernel.py", line 1173, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/meerk40t/core/cutplanner.py", line 296, in plan
    self.execute()
  File "/usr/local/lib/python3.9/site-packages/meerk40t/core/cutplanner.py", line 400, in execute
    cmd()
  File "/usr/local/lib/python3.9/site-packages/meerk40t/core/cutplanner.py", line 469, in strip_rasters
    if len(op) == 1 and isinstance(op[0], SVGImage):
TypeError: object of type 'LaserOperation' has no len()
```

Here's the SVG file:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   width="60mm"
   height="60mm"
   viewBox="0 0 60 60"
   version="1.1"
   id="svg8"
   inkscape:version="1.0.2 (e86c8708, 2021-01-15)"
   sodipodi:docname="circle.svg">
  <defs
     id="defs2">
    <rect
       x="71.06275"
       y="119.13209"
       width="32.594937"
       height="12.257925"
       id="rect855" />
  </defs>
  <sodipodi:namedview
     id="base"
     pagecolor="#ffffff"
     bordercolor="#666666"
     borderopacity="1.0"
     inkscape:pageopacity="0.0"
     inkscape:pageshadow="2"
     inkscape:zoom="0.94972573"
     inkscape:cx="134.40268"
     inkscape:cy="43.665962"
     inkscape:document-units="mm"
     inkscape:current-layer="layer1"
     inkscape:document-rotation="0"
     showgrid="false"
     fit-margin-top="0"
     fit-margin-left="0"
     fit-margin-right="0"
     fit-margin-bottom="0"
     inkscape:window-width="1252"
     inkscape:window-height="943"
     inkscape:window-x="0"
     inkscape:window-y="25"
     inkscape:window-maximized="0" />
  <metadata
     id="metadata5">
    <rdf:RDF>
      <cc:Work
         rdf:about="">
        <dc:format>image/svg+xml</dc:format>
        <dc:type
           rdf:resource="http://purl.org/dc/dcmitype/StillImage" />
        <dc:title></dc:title>
      </cc:Work>
    </rdf:RDF>
  </metadata>
  <g
     inkscape:label="Layer 1"
     inkscape:groupmode="layer"
     id="layer1"
     transform="translate(-57.690468,-82.636902)">
    <circle
       style="fill:none;fill-rule:evenodd;stroke-width:0.264583;stroke:none;stroke-opacity:1"
       id="path10"
       cx="87.690468"
       cy="112.6369"
       r="30" />
    <text
       xml:space="preserve"
       id="text853"
       style="font-style:normal;font-weight:normal;font-size:10.58329999999999949px;line-height:1.25;font-family:sans-serif;white-space:pre;shape-inside:url(#rect855);fill:#0000ff;fill-opacity:1;stroke:none;text-align:center"
       x="13.492674"
       y="0"><tspan
         x="73.86251"
         y="128.78828"><tspan
           style="text-align:center;text-anchor:middle;fill:#0000ff;fill-opacity:1">TEST</tspan></tspan></text>
  </g>
</svg>
```

tatarize, 2021-05-08:
```  File "/usr/local/lib/python3.9/site-packages/meerk40t/core/cutplanner.py", line 469, in strip_rasters
    if len(op) == 1 and isinstance(op[0], SVGImage):
TypeError: object of type 'LaserOperation' has no len()
```

LaserOperation used to have len() but it got switched when the tree got switched. This is an oversight. Found another with stripping text, and another at the cusp of `element engrave`. The text in the file gets classified in raster, but since cli mode has rendering engine it strips the raster out.  Upgrading pip should do the stripping of the test text out correctly. Giving you nothing to engrave and exiting normally.

---

Note: use post27+ since I got rid of the couple bugs I hit getting the cobwebs out of this code.

Now if you want to do this, cli mode doesn't process rasters but I assume you maybe want to cut. Though 100mm/s is a *fast* engrave.

`meerk40t -qze "inkscape locate input circle.svg text2path load" -e "element* engrave -s 100" -e "plan copy preprocess validate list  blob list spool"`

This will load up meerk40t, tell it to quit when finished, do not load the gui. And execute these three commands:

* "inkscape locate input circle.svg text2path load"
     * inkscape: we are going to use inkscape.
     * locate: find inkscape on the system
     * input <file>: use this file for inkscape
     * text2path: perform text to path
     * load the file inscape created (temp.svg)
* "element* engrave -s 100"
     * element*: Find all elements
     * engrave: all elements passed to this command get put in an engrave operation.
          * -s 100: Speed 100, the speed of the engrave is 100mm/s
* "plan copy preprocess validate list  blob list spool"
     * plan: we are making a new job
     * copy: copy the data from operations
     * preprocess: check if anything needs to be done
     * validate: do those things.
     * list: show me the plan currently
     * blob: convert this plan to cutcode
     * list: show me the plan again
     * spool: send this cutcode to the laser.

If you needed to make an image inkscape might also have you covered there too.
From console: `help`
```
--- inkscape Commands ---
image           image <operation>*
input           input filename
load            load processed file
locate          find inkscape
makepng         make png
simplify        simplify path
text2path       text to path
version         determine inkscape version
```

You could rather than `text2path` do `makepng` and load that. You could also do both. `text2path load makepng load` which would make that into a vector and a image.

---

Alternatively, I think it used to be the case that it would make the rasters if wxPython was merely installed even if the gui was suppressed. I could certainly restore that functionality. You could also load up actual images and then they won't need to convert.

giladrom, 2021-05-08:
> Now if you want to do this, cli mode doesn't process rasters but I assume you maybe want to cut. Though 100mm/s is a _fast_ engrave.

Actually, I'm interested in raster engraving - I'd like to automate a process for engraving names on personalized items for my employees, without them having to do anything but place the item and press a button, with an auto-generated SVG. Currently using K40Whisperer with manually generated SVGs through Inkscape.

tatarize, 2021-05-08:
Seems like a worthwhile project. It might be able to be mostly automated with a batch file. Which is basically just a file of console commands. Since you can run it pretty easily with -b from the command line. Though it's a static file which is to say means it couldn't do the customizable aspect.

---

The command:

`text "Jack Jackson" scale 50 50 translate -a 0 20% raster -s 100`

In MeerK40t will create the text "Jack Jackson" make it fairly big, move it absolute position 0, and 20% down heightwise, then put that in a raster operation at speed 100.

`plan copy preprocess validate blob spool`

Would then start the process running, and send it right to the laser.

You could then do `element* delete` afterwards to get rid of that. It would take using the gui to do the rastering of the text. Though I could likely make that properly accessible from the CLI if wxPython is merely installed.

This generally would allow a lot of automation in the process but there's still some obvious gaps. 

---

There's no obvious way to take the user crafted input. I guess this could be scripted outside of Meerk40t to ask for the name and put it in whatever set of commands gets run, there's still that slight issue of the lack of rastering without gui. I'll probably fix that shortly. Then you really could just run a more or less automated process.

You'd need to figure out asking the name stuff, but it's pretty close out of the box. Most of the interest is with #226, the cli stuff should be enough to get most of any rather typical workflow done, with just the CLI.

In theory this could be done (if I fix the raster thing) with a single command to MeerK40t, and maybe a couple other little things.

meerk40t -qze "text \"Jack Jackson\" scale 50 50 translate -a 0 20% raster -s 100" -e "plan copy preprocess validate list blob list spool"

I might have to check the use of quotes within the commandline there since I'm not sure that'll work, since you need to quote the text there Jack_Jackson would also work but that might be a bit meh. And the CLI raster access rather than stripping the rasters.

You could *almost* do this with an alias. So you bind a key in MeerK40t, but there's no way to strictly do the text input. There's a tiny amount of variable access with $x and $y in bind referring to the current position of the laserhead, but there's nothing to parse a request for this information.

I'll focus on the cli raster and commandline double-quote issue.  The user input is a bit iffier but there's clearly some need to say bind a key to ask you a name and raster it.

tatarize, 2021-05-08:
`meerk40t -mcze -e "text \"Jack Jackson\" scale 50 50 translate -a 0 20% raster -s 100" -e "plan copy preprocess validate list blob list spool"`

I ran a bunch of tweaks and this execute now works. Though it will require a pip update to post28.
https://github.com/meerk40t/meerk40t/commit/fe5a3e24ad42655c41baa26fe88fe0fdbf42d039

* Makeimage split into multiple bits.
* GUI App launches without Main Window for Render
* Lifecycles throw a console or gui flag.
* run_later moved to main window rather than app. (thread doesn't actually work until MainLoop() is called.
* thread.daemon's do not wait for a shutdown.
* console_thread is a thread daemon.
* bootstrap ready occurs before the batch and executes of the CLI.

It took a bit to work through the issues. Turns since images need a size and the rendering engine in wxPython sets the size of the image it had an issue if you tried to make an image before drawing it the first time as the image created had no size. The ready lifecycle kicks off the controller so it needed to be moved until after the execute commands in the commandline. And loading up the wxApp without hitting mainloop wrote the run_later value to the kernel but then did not actually have a thread to use to execute things, so the signals stopped working which is what was needed for the controller. Also, the controller kicks off officially before the info is spooled so setting the -q flag will quit before the laser runs. In theory this is fixable if we set the "_quit" on done flag with an -e command. But, we can't actually set with an -e command.

We could however spool a quit command, with plan_alias.

tatarize, 2021-05-08:
Pushed another update to fix that crash. That command actually works now.

giladrom, 2021-05-08:
Will I be able to use the original SVG workflow, or is that not recommended/doable?

giladrom, 2021-05-08:
This is the error I get with the SVG now (after updating to the latest post):

```
19646.441834 -> 5928.508109 reduced -13717.933725 (-69.824011%)
MeerK40t crash log. Version: 0.7.0 Beta-21 on darwin
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/meerk40t/kernel.py", line 1420, in run
    func_result = func()
  File "/usr/local/lib/python3.9/site-packages/meerk40t/device/basedevice.py", line 400, in _interpret_threaded
    self._process_spooled_item()
  File "/usr/local/lib/python3.9/site-packages/meerk40t/device/basedevice.py", line 411, in _process_spooled_item
    if self.plotplanner_process():
  File "/usr/local/lib/python3.9/site-packages/meerk40t/device/lhystudios/lhymicrointerpreter.py", line 376, in plotplanner_process
    on = int(on)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
```

giladrom, 2021-05-08:
> Pushed another update to fix that crash. That command actually works now.

When I execute this, I get the following output but the laser doesn't seem to do anything:
```
% meerk40t -mcze -e "plan-alias finish quit" -e "text \"Jack Jackson\" scale 50 50 translate -a 0 20% raster -s 100" -e "plan copy preprocess validate list blob append finish list spool"

plan-alias finish quit
text "Jack Jackson" scale 50 50 translate -a 0 20% raster -s 100
plan copy preprocess validate list blob append finish list spool
Copied Operations.
----------
Plan:
0: 0
----------
Plan 0:
0: Image 140mm/s =T2B 1000ppi ±20
1: Raster1 140mm/s =T2B 1000ppi ±20
2: Engrave 100mm/s 1000ppi
3: Cut 10mm/s 1000ppi
4: Engrave 110mm/s 1000ppi
5: Raster1 100mm/s =T2B 1000ppi ±20
6: <function Planner.beep at 0x10d36df70>
Commands 0:
----------
----------
Plan:
0: 0
----------
Plan 0:
0: CutCode(1 items)
1: <function Planner.beep at 0x10d36df70>
2: <function Planner.attach.<locals>.plan_alias.<locals>.user_defined_alias at 0x11cfd0310>
Commands 0:
----------
Spooled Plan.
>
```

giladrom, 2021-05-08:
> ```
>     if len(op) == 1 and isinstance(op[0], SVGImage):
> TypeError: object of type 'LaserOperation' has no len()
> ```
> 
> 
> LaserOperation used to have len() but it got switched when the tree got switched. This is an oversight. Found another with stripping text, and another at the cusp of `element engrave`. The text in the file gets classified in raster, but since cli mode has rendering engine it strips the raster out.  Upgrading pip should do the stripping of the test text out correctly. Giving you nothing to engrave and exiting normally.
> 
> 
> 
> Note: use post27+ since I got rid of the couple bugs I hit getting the cobwebs out of this code.
> 
> 
> Now if you want to do this, cli mode doesn't process rasters but I assume you maybe want to cut. Though 100mm/s is a _fast_ engrave.
> 
> 
> `meerk40t -qze "inkscape locate input circle.svg text2path load" -e "element* engrave -s 100" -e "plan copy preprocess validate list  blob list spool"`
> 
> 
> This will load up meerk40t, tell it to quit when finished, do not load the gui. And execute these three commands:
> 
> 
> 
> * "inkscape locate input circle.svg text2path load"
>   
>   * inkscape: we are going to use inkscape.
>   * locate: find inkscape on the system
>   * input : use this file for inkscape
>   * text2path: perform text to path
>   * load the file inscape created (temp.svg)
> 
> * "element* engrave -s 100"
>   
>   * element*: Find all elements
>   * engrave: all elements passed to this command get put in an engrave operation.
>     
>     * -s 100: Speed 100, the speed of the engrave is 100mm/s
> 
> * "plan copy preprocess validate list  blob list spool"
>   
>   * plan: we are making a new job
>   * copy: copy the data from operations
>   * preprocess: check if anything needs to be done
>   * validate: do those things.
>   * list: show me the plan currently
>   * blob: convert this plan to cutcode
>   * list: show me the plan again
>   * spool: send this cutcode to the laser.
> 
> 
> 
> If you needed to make an image inkscape might also have you covered there too.
> 
> From console: `help`
> 
> 
> ```
> --- inkscape Commands ---
> image           image <operation>*
> input           input filename
> load            load processed file
> locate          find inkscape
> makepng         make png
> simplify        simplify path
> text2path       text to path
> version         determine inkscape version
> ```
> 
> 
> You could rather than `text2path` do `makepng` and load that. You could also do both. `text2path load makepng load` which would make that into a vector and a image.
> 
> 
> 
> Alternatively, I think it used to be the case that it would make the rasters if wxPython was merely installed even if the gui was suppressed. I could certainly restore that functionality. You could also load up actual images and then they won't need to convert.

So this gives me the following crash:

```
% meerk40t -qze "inkscape locate input two.svg text2path load" -e "element* engrave -s 100" -e "plan copy preprocess validate list blob list spool"
inkscape locate input two.svg text2path load
----------
Finding Inkscape
Searching: /Applications/Inkscape.app/Contents/MacOS/Inkscape -- Result: Success
Searching: /Applications/Inkscape.app/Contents/Resources/bin/inkscape -- Result: Fail
----------
Making plain_svg with Inkscape.
quit
b''
Loading...
MeerK40t crash log. Version: 0.7.0 Beta-21 on darwin
Traceback (most recent call last):
  File "/usr/local/bin/meerk40t", line 8, in <module>
    sys.exit(run())
  File "/usr/local/lib/python3.9/site-packages/meerk40t/main.py", line 363, in run
    device(v.strip() + "\n")
  File "/usr/local/lib/python3.9/site-packages/meerk40t/kernel.py", line 136, in __call__
    return self._kernel.console(data)
  File "/usr/local/lib/python3.9/site-packages/meerk40t/kernel.py", line 2302, in console
    self._console_parse(command, channel=self._console_channel)
  File "/usr/local/lib/python3.9/site-packages/meerk40t/kernel.py", line 2382, in _console_parse
    data, remainder, input_type = command_funct(
  File "/usr/local/lib/python3.9/site-packages/meerk40t/kernel.py", line 1180, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/meerk40t/extra/inkscape.py", line 19, in load
    e.load(filename)
AttributeError: 'Context' object has no attribute 'load'
```

tatarize, 2021-05-09:
Best bet:
 `meerk40t -ze "plan-alias finish quit" -e "text \"Jack Jackson\" scale 50 50 translate -a 0 20% raster -s 100" -e "plan copy preprocess validate list blob append finish list spool"`
---

> Will I be able to use the original SVG workflow, or is that not recommended/doable?

Should be fine. The general point is the meerk40t should be able to adapt your chosen workflow, it doesn't need to go the other way around. Loading a file and sending it to the laser is the most obvious workflow ever and certainly should be supported. Anything that stops you from getting that done is a bug and needs to be corrected. Doubly so for crashes which are defacto bugs.

> This is the error I get with the SVG now (after updating to the latest post):
>>File "/usr/local/lib/python3.9/site-packages/meerk40t/device/lhystudios/lhymicrointerpreter.py", line 376, in  plotplanner_process
>    on = int(on)
> TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

This happens weirdly because the raster thing didn't work on the text object and it returned an image size of 1x1 which is the real bug. Though having None in there is another bug I just didn't think it would commonly happen after I went about fixing it. I guess I missed a spot.

>When I execute this, I get the following output but the laser doesn't seem to do anything:
> % meerk40t -mcze -e "plan-alias finish quit" -e "text \"Jack Jackson\" scale 50 50 translate -a 0 20% raster -s 100" -e "plan copy preprocess validate list blob append finish list spool"

Apologies. I had remembered I didn't correct that but I somehow forgot I was going back to fix that. Remove the `m` from `-mcze` in fact you could also drop the `c`. The m does mock. I didn't want it sending to the laser but I forgot to remove that from the posted confirmed that worked. The c loads up text console which is usually fine but because of the way `input()` work it can't die without an extra return push. It's great if you know your way around the console since you can get lots of information. `channel open 1/pipe/usb_send` etc, gives you some nice information. Etc.

TLDR: `meerk40t -ze "plan-alias finish quit" -e "text \"Jack Jackson\" scale 50 50 translate -a 0 20% raster -s 100" -e "plan copy preprocess validate list blob append finish list spool"`

```% meerk40t -qze "inkscape locate input two.svg text2path load" -e "element* engrave -s 100" -e "plan copy preprocess validate list blob list spool"
inkscape locate input two.svg text2path load
----------
Finding Inkscape
Searching: /Applications/Inkscape.app/Contents/MacOS/Inkscape -- Result: Success
Searching: /Applications/Inkscape.app/Contents/Resources/bin/inkscape -- Result: Fail
----------
Making plain_svg with Inkscape.
quit
b''
Loading...
```

The problem here is the -q it means quit when the spooler empties out. At issue though is that spooler *is*  empty so it quits too early and you can see the command there 3 from the bottom. That kills the process by shutting down meerk40t. It's supposed to help but I think maybe I need to add shutdown and just have -a add a shutdown like the version ended up with for the command line. And then make the -a trigger deadlast like even after the commands. It would likely make things a lot more consistant.

---

So 1-2 crash bug, and a better workflow idea.

tatarize, 2021-05-09:
I moved some stuff around and now the rather obvious command:

`python meerk40t.py -qaze "text \"Jack Jackson\" scale 50 50 translate -a 0 20% raster -s 100"` works.

So would loading image or svg, or using inkscape to process the image and sending it to the backend. The ordering changed a bit so the `-a` flag goes last. After `-e` commands or a `-b`. The `-o` origin (resets the laser) and the `-q` either occurs within the auto spooling or just sets the flag if `-a` isn't set. The end result is this ordering is more well thought out and plays well with others.

Several of those attempts will now work fine. -q isn't going to inject a random quit since it only occurs *after* spooling. And you don't need to do "plan-alias finish quit" since there's a proper shutdown command. And the auto will do a `plan append shutdown` if you requested a -q happen. And you'll finish things like a -o "return to 0,0" at the more correct time.

CLI stuff is important and doesn't get nearly enough TLC. Thank you for raising this issue. I'll try to get to that crash bug you found. The second one might be a one-off that happened because quit hit pretty randomly and the load command was suddenly gone because everything was tearing itself down.

giladrom, 2021-05-10:
Is there a way to move the laser to a certain position before starting the job? It seems as though -e "move 3000 3000" has unexpected behavior (it will move sometimes, but not all the time). Using the GUI console yields the expected results.

tatarize, 2021-05-11:
The -q does weird things at times but -e "quit" is a bit better at times. You could also put it on a timer. `timer 1 60 quit` (1 time 60 seconds between). That or spool the shutdown to make sure the stuff has finished. The commands should generally be in there, and kinda working. But, CLI stuff doesn't get the TLC it needs, console commands get a lot and should work for most things.

It might be a race condition if you do -q and say -e "move 3in 3in" it'll end up doing both and if the 3in 3in movement doesn't get sent before the shutdown kills the controller meant to send it, that stuff could certainly be a thing.

giladrom, 2021-05-11:
Okay! So I was able to use timers to sequence the following workflow:

1. Go home
2. Move to 4000 4000 (middle of laser tray)
3. start laser
4. Go back home
5. quit

However, #3 (start laser) always goes back home before starting engraving, so my positioning is not maintained. Even when I move the text element around the SVG, it still goes back home every time, and so I can't position it properly.

Here's the output from meerk40t:
```
% meerk40t -ze "home" -e "timer 1 5 move_absolute 4000 4000" -e "inkscape locate input two.svg text2path load" -e "element* engrave -s 110" -e "timer 1 15 plan copy preprocess validate list blob list spool" -e "timer 1 60 home" -e "timer 1 70 quit"
home
timer 1 5 move_absolute 4000 4000
inkscape locate input two.svg text2path load
----------
Finding Inkscape
Searching: /Applications/Inkscape.app/Contents/MacOS/Inkscape -- Result: Success
Searching: /Applications/Inkscape.app/Contents/Resources/bin/inkscape -- Result: Fail
----------
Making plain_svg with Inkscape.
b''
Loading...
element* engrave -s 110
timer 1 15 plan copy preprocess validate list blob list spool
timer 1 60 home
timer 1 70 quit
```

And the SVG:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   xmlns:osb="http://www.openswatchbook.org/uri/2009/osb"
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:xlink="http://www.w3.org/1999/xlink"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   width="13in"
   height="7.5in"
   viewBox="0 0 330.20002 190.5"
   version="1.1"
   id="svg8"
   inkscape:version="1.0.2 (e86c8708, 2021-01-15)"
   sodipodi:docname="two.svg">
  <defs
     id="defs2">
    <linearGradient
       id="linearGradient893"
       osb:paint="solid">
      <stop
         style="stop-color:#0000ff;stop-opacity:1;"
         offset="0"
         id="stop891" />
    </linearGradient>
    <linearGradient
       inkscape:collect="always"
       xlink:href="#linearGradient893"
       id="linearGradient895"
       x1="40.52813"
       y1="120.54301"
       x2="56.68288"
       y2="120.54301"
       gradientUnits="userSpaceOnUse" />
  </defs>
[... 54 lines cut ...]
    <text
       xml:space="preserve"
       style="font-style:normal;font-variant:normal;font-weight:normal;font-stretch:normal;font-size:3.88056px;line-height:1.45;font-family:Arial;-inkscape-font-specification:Arial;font-variant-ligatures:normal;font-variant-caps:normal;font-variant-numeric:normal;font-variant-east-asian:normal;text-align:center;text-anchor:middle;fill:#000000;fill-opacity:1;fill-rule:evenodd;stroke:none;stroke-width:0.2;stroke-miterlimit:4;stroke-dasharray:none;stroke-opacity:1"
       x="121.95787"
       y="180.4541"
       id="text879"><tspan
         sodipodi:role="line"
         id="tspan877"
         x="121.95787"
         y="180.4541"
         style="font-style:normal;font-variant:normal;font-weight:normal;font-stretch:normal;font-size:3.88056px;font-family:Arial;-inkscape-font-specification:Arial;font-variant-ligatures:normal;font-variant-caps:normal;font-variant-numeric:normal;font-variant-east-asian:normal;text-align:center;text-anchor:middle;fill:#000000;fill-opacity:1;fill-rule:evenodd;stroke:none;stroke-width:0.2;stroke-miterlimit:4;stroke-dasharray:none;stroke-opacity:1">HAZEL RYAN</tspan></text>
    <text
       xml:space="preserve"
       style="font-style:normal;font-variant:normal;font-weight:normal;font-stretch:normal;font-size:3.88056px;line-height:1.45;font-family:Arial;-inkscape-font-specification:Arial;font-variant-ligatures:normal;font-variant-caps:normal;font-variant-numeric:normal;font-variant-east-asian:normal;text-align:center;text-anchor:middle;display:inline;fill:#000000;fill-opacity:1;fill-rule:evenodd;stroke:none;stroke-width:0.2;stroke-miterlimit:4;stroke-dasharray:none;stroke-opacity:1"
       x="121.95788"
       y="185.59337"
       id="text879-7"><tspan
         sodipodi:role="line"
         id="tspan877-4"
         x="121.95788"
         y="185.59337"
         style="font-style:normal;font-variant:normal;font-weight:normal;font-stretch:normal;font-size:3.88056px;font-family:Arial;-inkscape-font-specification:Arial;font-variant-ligatures:normal;font-variant-caps:normal;font-variant-numeric:normal;font-variant-east-asian:normal;text-align:center;text-anchor:middle;fill:#000000;fill-opacity:1;fill-rule:evenodd;stroke:none;stroke-width:0.2;stroke-miterlimit:4;stroke-dasharray:none;stroke-opacity:1">443-604-3886</tspan></text>
  </g>
</svg>
```

tatarize, 2021-05-12:
With the gui partially loading like it is, it will have access to the persistent settings. If you have pre-home in your settings set it will do a home before starting with the rest of the settings. I'd make sure you don't have that set, the list of the plan should also tell if there's such a command in there.

giladrom, 2021-05-12:
Couldn't find any kind of persistent setting for pre-home in the UI or elsewhere? Is there a place I should look - or rather, is there a way to move the laser BEFORE firing it, even if it has a pre-home setting?

tatarize, 2021-05-13:
If it's adding a pre-home it should exist when you list your settings, which doesn't do printing of the console information by default. If you put in a `-c` you could see it. Mine certainly did since I have it add in autobeep. Though you could also do `-e "channel print console"` even without the `-c` and it would send the console information to standard-out.

You can run it with the console command: `set -p / prehome False` and it would turn prehome off. It's in the JobPreview, in the menu items to turn it off normally. If you turn it off with console you'd only need to do it one time since `-z` has access to persistent settings (I added -Z to hard remove the gui completely when that's needed). This command would likely be changed in a version or so since I'm tearing apart devices (though it would likely just be `set prehome False` after the devices update).

You don't really need to change the position regardless of the home if the text is actually located in the right position. It should move to the location of the text and raster it there, regardless where there is. In some cases you'd to `element* transform 3000 3000` to move all your element data down and right 3in (you can also do 'transform 3in 3in').

* `element*` -- is select all elements.
* `transform tx ty` is move the element passed to the command by the specified amount.

tatarize, 2021-05-13:
[image]

giladrom, 2021-05-15:
This menu doesn’t show up on MacOS - all I get is the dropdown.

giladrom, 2021-05-15:
Also, when trying to add element* transform I get:

element* transform 3000 3000
transform is not a registered command in this context: elements

giladrom, 2021-05-15:
Ok I figured it out! the only thing left to do now is to only engrave specific elements instead of the whole SVG. What I'm using is -e "timer 1 20 element3 engrave element6 engrave -s 110", but it seems to still engrave all elements, as if I did element* engrave (I just want to engrave the text elements, but have other elements in the SVG for positioning).

tatarize, 2021-05-15:
Sorry the command there would be "element* translate" rather than transform. Translate means to move it but it is one of the affine transformations. You seemed to have figured it out.

----

In theory `element3,6 engrave -s 110` (no space between 3 and 6) should likely work. However it might be better if you put this directly into the plan. The plan does a full copy so every single operation which may be classed into several operations gets put into the plan and processed, that's what `copy` does in a plan context there. What you actually want is to is take this engrave operation and send it and *only* it to the plan.

`element3,6 engrave -s 110 plan preprocess validate list blob list spool` will go from the engrave `ops` context to `plan` which does a something similar to `copy-selected` but actually just copies the information specifically in the pipeline. Each command copies some data to the next operation in the commands. In this case going from an ops to the plan command just copies the operations it was given into the current plan. Which is *just* the given engrave.

tatarize, 2021-05-15:
`element3,6 engrave -s 110 plan preprocess validate list append home blob list spool` might also work and remove the need to put that home on the time, since it would home directly after the finishing the rest of the job. `append home` would just add in a home command there. While the shutdown command might have issues (though it really shouldn't) especially if added *after* home.
```
>> plan command
Plan Commands:
physicalhome
home
origin
unlock
wait
beep
interrupt
shutdown
```

You can `append` or `prepend` any of these you want. So in theory adding `append home append shutdown` before the spool *should* home the device and shutdown things when that goes across the spooler.

---

I'm sure a lot of the documentation needs to be improved here. The help in console is pretty good especially `help <command>` with some extra bits, but I don't think I've done any really good explanation of the chaining or contexts. But, it's always been a do that in the future thing.

```
engrave
        (elements) -> engrave -> (ops)
        Option: float ('--speed', '-s')
        Option: float ('--power', '-p')
        Option: int ('--step', '-S')
        Option: Length ('--overscan', '-o')
        Option: Color ('--color', '-c')
        Option: int ('--passes', '-x')
```

tatarize, 2021-05-15:
> This menu doesn’t show up on MacOS - all I get is the dropdown.
> […](#)

@joerlane Does this not actually show up on OSX?

joerlane, 2021-05-15:
The menus do exist in Mac OS X. They conform to Mac OS standards instead though; and appear in the regular menu bar at the top of the screen; next to the Apple menu, and MeerK40t menu. They are only active when the Jobs window is in front. 

The Camera window will also override menus to show those are specific to the camera window as well.

giladrom, 2021-05-17:
> `element3,6 engrave -s 110 plan preprocess validate list append home blob list spool` might also work and remove the need to put that home on the time, since it would home directly after the [...]

Still no change - ```element3,6 engrave -s 110 plan preprocess validate list append home blob list spool``` still insists on engraving the entire file including all elements.

tatarize, 2021-05-19:
Sorry for the delay.

I guess maybe delete the other elements. I'll track down the reason this is happening and make those commands a bit more robust since I couldn't use them to see if it was working weirdly.

`element1,2,4,5,7 delete` will get rid of them from the list. Then they certainly can't end up engraving.

giladrom, 2021-05-19:
I figured it out! I just ended up deleting the actual operations, since deleting the elements didn’t really work. 

Here’s the final command line:

```
meerk40t -z -e "timer 1 1 home" -e "inkscape locate input 304705.svg text2path load" -e "element* translate 0 0"  -e "element* translate 3in 3in" -e "timer 1 2 operation2,3 delete" -e "timer 1 3 elements* engrave -s 110" -e "timer 1 10 plan copy preprocess validate list append home blob list spool" -e "timer 1 50 quit"
```

tatarize, 2021-05-20:
`timer 1 3 elements* engrave -s 110` you mean element* right, or was that a reason things didn't work. elements is like `element* list` (when I get that added in).

I wouldn't fiddle with working as you can only get something less functional but it should tell you elements* is not a registered command or whatever.

giladrom, 2021-05-20:
Oh, that could be it. Although I did mess around with the command line so probably a typo, I did try element3,6 engrave -s 110, but I’ll try again.

closed 2021-07-17 by tatarize