meerk40t/meerk40t #409: 0.7.0b22 USB won't connect - failed to load dynlib/dll 'CH341DLL.dll'

Sophist-UK, 2021-05-08:
Meerk40t 0.6.x connects just fine.

When I try to use 0.7.0b22 (or earlier) USB won't connect and I get the following message in the USB window:

`Failed to load dynlib/dll 'CH341DLL.dll'. Most probably this dynlib/dll was not found when the application was frozen.`

tatarize, 2021-05-12:
One bit of feedback said it worked. Though it had some bad false positives on the virus front. I can recompile the bootloaders a few more times and try to get something that hits fewer issues on that front.

Sophist-UK, 2021-05-13:
b23 connected fine first time.

Sophist-UK, 2021-05-13:
But I mistakenly closed Meerk40t 0.7b23 and when I reopened it then it did not connect - had to turn K40 off and on to ghet it to connect.

tatarize, 2021-05-14:
Hm. Seems like a pretty repeatable workflow. I'll give that a shot when I start testing the new devices code. I added a new controller thing that has reset and release to the menubar.

[image]

tatarize, 2021-05-20:
This is corrected in b23 and b24.

closed 2021-05-20 by tatarize