meerk40t/meerk40t #410: Feature request : pulse on dots.

FabienTregan, 2021-05-08:
Hi,

  Sometimes, I do not want to cut a shape, but just make a pulse without moving the head (e.g. if I want to make the smallest possible hole in a sheet of black paper to make a pinhole camera). I currently do it manually by
  - puting a segment starting where I want to make each hole in the dxf
  - selecting the segment 
  - moving the head to top/left of the segment
  - sending a short pusle
 
  Would it be hard to automate this in MeerK40t using DXF Points and adding a "pulse duration for dots" parameter for each color (in addition to speed and PPI that would still be used for non-dots) ?
  
  I attache a .dxf file with points (renamed in .dxf.txt because github won't accept .dxf).

Best Regards.
[points.dxf.txt](https://github.com/meerk40t/meerk40t/files/6446521/points.dxf.txt)

tatarize, 2021-05-09:
In the console there's a command called pulse. It's not a mode but you can do `pulse 100` to fire your laser in one spot for 100 ms without moving. You can also do `bind space +laser` in console and then you can move your laser around with the WASD keys and hit space to fire the laser in one spot. It would be bound by default but letting cats burn down your house is a bad idea.  You could also bind a key to do `pulse 100` or whatever value fits your need. 

It is possible to code this up if I had a series of points, though there's nothing in there currently since this would be a different sort of operation. It's not strictly enough to add in a pulse duration for dots though there is one of those in navigator to let you fire pulses from there. Since you can only do this pulse duration trick in default/rapid mode and most of the other operations run in the sort of compact/program mode.

There's also not strictly a dots structure in svg, and those are the geometry data structures I'm currently using. I'm not sure how often this would get used and it might be more reasonable to do this as a sort of one-off buried function. That or just as a separate operation type.

tatarize, 2021-05-09:
I've got the first step. DXF Point objects show up now. The issue is they are closed moves. `Path('M0,0z')` it's a technically valid path and is just a point. I do not know how you could move it since it's 0x0 in size. If you try to laser these it will actually go there in an engrave or whatever. But, it's 0 pixels long, and consequently doesn't do anything. It goes there and then goes to the next point. There's nothing to dwell at that point... at least not yet.

tatarize, 2021-05-09:
And I added the rest of the dots stuff.

You'll need the latest pull of 0.7.0, but if you type `dots -s <time-in-ms>` in console it will make a dot operation. This will exclusively classify dots, which are literally 0 length position objects. These will dwell for the set amount of time.

---

Also, rather than do all this you could have done very tiny rectangles, and set a passes value to adjust for time. I think this is what the one person who requested this before ended up doing, though that might mess with your workflow.

--

I gave dots an icon, and classification scheme, and some coverage for how they should work. But, I kept them *mostly* hidden. They don't need much in the way of typical operation setting, so I just sort of blanked that out. They are normal operations so they will get saved like normal operations. And anything that is a dot will classify in them (unless it qualified for another classification first, eg red and red cut is above them).

You could safely leave dots in your operations, they can only classify a dot. If you put something not a dot in there. It will go to the first point and dwell their for the allotted time.

tatarize, 2021-05-13:
I managed to patch in the correction for moving the Dot Operation objects and having it still work.

https://www.youtube.com/watch?v=gu5YoVJGlj8

tatarize, 2021-08-06:
This exists in the latest versions.

closed 2021-08-06 by tatarize