meerk40t/meerk40t #414: Meerk40t 0.6.22: Vector time estimate in spooler way under

Sophist-UK, 2021-05-12:
When performing a cut, estimate was 44s, actual was 53s.

tatarize, 2021-05-13:
That's not that far under. Some of those estimates are basically absolute voodoo and not remotely close.

I think to get them right would require some work timing the various ticks of the laserhead. Figuring out when it slows down. How long a raster scanline takes to reverse course. And the intro and outro acceleration timing. Had I properly accounted for the slower real-speed (for unknown reasons the Chinese software goes at 92% of the actual requested speed) it would have estimated 48.76. I'd think maybe the intro and outro waste about 1.5 seconds each assuming they did both.

It might be doable to get that timing a lot more dialed in. #342

tatarize, 2021-09-25:
Needs some better timed data to estimate this stuff better.

tatarize, 2022-07-20:
#1165 This stuff might still be wrong. I guess at some point I really need to fix up that code or whatever.

closed 2023-05-29 by Sophist-UK