meerk40t/meerk40t #415: Spline crashing on 0.7.0-beta23

FabienTregan, 2021-05-13:
I made the simple possible spline and I can not open it :

```
MeerK40t crash log. Version: 0.7.0 Beta-23 on linux
Traceback (most recent call last):
  File "/home/fabien/projets/meerk40t-0.7.0-beta22/meerk40t-0.7.0-beta23/meerk40t/gui/wxmeerk40t.py", line 2317, in on_click_open
    self.load(pathname)
  File "/home/fabien/projets/meerk40t-0.7.0-beta22/meerk40t-0.7.0-beta23/meerk40t/gui/wxmeerk40t.py", line 1827, in load
    svg_ppi=self.context.svg_ppi,
  File "/home/fabien/projets/meerk40t-0.7.0-beta22/meerk40t-0.7.0-beta23/meerk40t/core/elements.py", line 4033, in load
    results = loader.load(self.context, self, pathname, **kwargs)
  File "/home/fabien/projets/meerk40t-0.7.0-beta22/meerk40t-0.7.0-beta23/meerk40t/dxf/dxf_io.py", line 64, in load
    elements, dxf, entity, scale, bed_dim.bed_height * MILS_PER_MM
  File "/home/fabien/projets/meerk40t-0.7.0-beta22/meerk40t-0.7.0-beta23/meerk40t/dxf/dxf_io.py", line 298, in entity_to_svg
    (b[1][0], b[1][1]), (b[2][0], b[2][1]), (b[3][0], b[3][1])
IndexError: list index out of range
```
[spline.dxf.txt](https://github.com/meerk40t/meerk40t/files/6471021/spline.dxf.txt)

tatarize, 2021-05-13:
In testing I never ran across a quad spline and figured they were all cubic. So that code tries to access a 4th point when it's limited to three. I'll fix it in short order.

tatarize, 2021-05-13:
https://github.com/meerk40t/meerk40t/commit/38de7cbc4f6c54caf38a752b14abeefc382749f6

And I'll patch it in 0.6.23+

closed 2021-08-12 by tatarize