meerk40t/meerk40t #416: 0.6.22 PNG file - white / transparent background still burned as black

Sophist-UK, 2021-05-13:


tatarize, 2021-05-13:
Hm. 0.6.22 should have fixed issue #390 which tended to have some minor issues there. There also might be an OS component to the problem since sometimes you get different backgrounds based in part on what color a particular implementation calls default.

tatarize, 2021-05-13:
Are the images being classed as rasters for you and combined? Or is it kicking in when classed as images? The head seems to be rastering okay in image. Unless it's somehow doing the proper smaller region but then coloring the transparency as black within the smaller region.

tatarize, 2021-05-13:
Also, if it's in a raster, can you try right clicking and go to the bottom with Make Raster Image and see if it produces an image with black there for you. I can't reproduce this failing. Also, what OS since that sometimes make a difference. I think Linux had some regions getting blacks where they shouldn't have had them.

Sophist-UK, 2021-05-13:
Sorry - it appears that the background was always transparent - and I mistakenly had not changed it to white.

O/S is Windows 10 1909.

Before #355 I used to get black background burns on every transparent PNG background. After #355 it worked fine until today and this specific image. I cannot find a menu item in 0.6.22 for Make Raster Image.

Sophist-UK, 2021-05-13:
Here is the dragon head with transparent bg.
[Circular keyfob dragon bee.zip](https://github.com/meerk40t/meerk40t/files/6474354/Circular.keyfob.dragon.bee.zip)

tatarize, 2021-05-14:
[image]

[image]

Make raster still went to white. So I'm thinking if it's a thing, that it'll be in the actual images themselves. The rastering routines have methods for different image types, so 1-bit black and white works like on and off, and grayscale L works like PPI applied to a shade. RGB should convert to grayscale sort of and work a bit like that. RGBA might have an issue within that. But, I'm pretty sure it should be RGB with the A value being used to attenuate the values.

`return (1.0 - (pixel[0] + pixel[1] + pixel[2]) / 765.0) * pixel[3] / 255.0` where 0, 1, and 2 are RGB and 3 is A. That seems like it should be correct. A value of 0.

Well, it's certainly something. Tested and it clearly is cutting that as black.

tatarize, 2021-05-14:
Okay, the dragon head image in a mode P image which gets:

```python
                        v = p[pixel * 3] + p[pixel * 3 + 1] + p[pixel * 3 + 2]
                        return 1.0 - v / 765.0
```

As the mode, but that discounts the chance of there being an alpha layer, but there clearly is one. Which identifies the bug. I'll lookup how to deal with that type of mode, when it has alpha.

tatarize, 2021-05-14:
I made the seemingly required update but it's weird and I'm going to have to debug it harder to figure out what's wrong. It weirdly goes over to the right hand side seemingly for no reason. But, if I did "image remove blue" on the image (the image has no blue and should be fine) it somehow rastered fine since it converted to RGBA. I still have something to get to the bottom of there, but it's closer.

diamened, 2021-09-11:
I'm seeing the same problem here. The attached image classifies as 'image' and has a PNG with transparent background in it. But the laser engraves the transparent areas black. The white parts are ok.

[Viking Runestone 02.svg.zip](https://github.com/meerk40t/meerk40t/files/7147129/Viking.Runestone.02.svg.zip)

tatarize, 2021-09-11:
Checked in the 7.1 version and this simulates those areas as black, implying it would absolutely want to raster the edges there with a hard laser-on.

tatarize, 2021-09-11:
The 0.6.23 version appears to have treated the image correctly though that version is still beta, and where this issue was expected to be fixed.

tatarize, 2021-09-11:
0.6.22 looked to do that correctly. It, however, did not.

[image]

Testing. 0.6.23 fixes this issue. 
[image]

Should push up the publication and test the rest of the code. And test that this is not an issue in 0.7.x

tatarize, 2021-09-11:
I tested and published the old 0.6.23 code that fixed this.

https://github.com/meerk40t/meerk40t/releases/tag/0.6.23

Please confirm that corrects it (It does). And I'll go ahead and close this issue.

diamened, 2021-09-11:
The 0.6.23 version works perfectly. Thank you

closed 2021-09-12 by tatarize