meerk40t/meerk40t #417: 0.7.0b23: USB errors when burning

Sophist-UK, 2021-05-13:


Sophist-UK, 2021-05-13:
Got 7 of these at the beginning of a vector operation.

tatarize, 2021-05-14:
Likely because it timed out. Did it recover successfully because that's the metric worth noting. Failing to recover from usb timeouts is a problem, having them exist, isn't really an issue, or isn't one that I could address.

tatarize, 2021-05-14:
Also, as a thing sometime the timeout is set low enough the M2 doesn't properly conform to best practices with USB. Sometimes you can not get a signal back until it's finished with a process... like home. It won't send you a reply until it's over which can sometimes take a while. And USB is supposed to give a reply after 250 ms, but it upwards of seven seconds. I moved up the timeouts.

Sophist-UK, 2021-05-14:
I don't think I get these under 6.23 - I will have to test it.

But the rest of the burn went fine.

Sophist-UK, 2021-05-14:
If there was a timeout it was of the order of miliseconds rather than seconds. The 5 or 7 messages appeared almost simultaneously.

tatarize, 2021-05-14:
0.6.x gets those but it doesn't report them. In 0.7.0 one of I think Tiger's great suggestions was to relay any errors in the libusb to the front end. So even if they are non-critical, they provide a lot more information and that's shown in 0.7.0 but not in 0.6.x.

Sophist-UK, 2021-05-14:
Well that explains why they are appearing in 070 and not 0622 - but I think that there may be a bug which sends some odd USB signals early on - if we are able to track it down we could stop these.

closed 2021-08-12 by tatarize