meerk40t/meerk40t #418: 0.7.0b23: Duplicate image operation - result has each image twice and then crashes

Sophist-UK, 2021-05-13:
```
MeerK40t crash log. Version: 0.7.0 Beta-23 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3148, in on_item_selection_changed
AttributeError: 'SVGImage' object has no attribute 'node'
```

Sophist-UK, 2021-05-13:
Crash when attempt to select any of the first set of duplicates in tree.

tatarize, 2021-05-14:
```
            if node.type == "opnode":
                selected[i] = node.object.node
```

I can't make this crash. Maybe I fixed it already. There were some issues there I fixed. Lemme try in the released version. --- Can't crash it there either.

tatarize, 2021-05-15:
* Find an operation that has children
* Right click: duplicate operation.
* Click and child within the duplicated operation.
* Crash.

tatarize, 2021-05-16:
Corrected in both branches of 0.7.0.

closed 2021-07-22 by tatarize