meerk40t/meerk40t #419: 0.7.0b23: Can we show Inkscape/svg friendly names in tree rather than internal names?

Sophist-UK, 2021-05-13:


tatarize, 2021-05-14:
If the object has an id it displays that. It's not too weirdly cryptic. What better name should it have instead.

[image]

Sophist-UK, 2021-05-14:
In Inkscape in the Object list you can rename an object to something more meaningful.

So from "image1234" to "Dragon Head" for example.

tatarize, 2021-05-14:
You're supposed to be able to double click a path and change the name in the large box at the top. But, I checked it's pretty lackluster. And doesn't do things like allow groups to be correctly renamed.

Sophist-UK, 2021-05-14:
I am not looking to be able to rename them in Meerk40t - just for Meerk40t to use the nice names I have already added in Inkscape. I assume that they are in the SVG XML somewhere.

tatarize, 2021-05-15:
Checked from a couple svgs that were posted for various issues the id which is svg spec. There's an `inkscape:label` value that has the fancier name you gave them this is true for group and other things you gave inkscape labels to. It wouldn't be too hard to prefer that as a name over id. Though it would be a bit inkscape specific. Assuming other graphics programs do something different it could certainly opt for those names over other names.

tatarize, 2021-05-15:
Added it.

https://github.com/meerk40t/meerk40t/commit/5f1243b3c1ebb49f9d18749f106abfbdbb81f724

[image]

It's a bit inkscape specific but it will prefer `inkscape:label` over `id`. I assume some other graphic editors also have a namespace label or some such thing. Those can be added through the same methods to also give them priority in naming. And I might actually switch meerk40t over to using them rather than rewriting the ids. Since the ids are functional and tend to require uniqueness within the spec.

closed 2021-08-12 by tatarize