meerk40t/meerk40t #420: AttributeError: 'Arc' object has no attribute 'reversed' (0.7.0 1f82339aa82a01cbeff483c03b06b9ac76de88db )

FabienTregan, 2021-05-14:
The joint .dxf (made with libre cad, using the Fillet tool to round some corners), gives an `"AttributeError: 'Arc' object has no attribute \'reversed\'"` in the optimization step^before engraving.

The "triangles" are made of a spline and a polyline which starts and ends at the same position but are not connected. If I remove either the polylines or the spline, the error doesn't rise.

Crash Report :

```
MeerK40t crash log. Version: 0.7.0 Beta-23 on linux
Traceback (most recent call last):
  File "/home/fabien/projets/meerk40t/meerk40t/gui/jobpreview.py", line 537, in on_button_start
    self.context("plan%s optimize\n" % self.plan_name)
  File "/home/fabien/projets/meerk40t/meerk40t/kernel.py", line 136, in __call__
    return self._kernel.console(data)
  File "/home/fabien/projets/meerk40t/meerk40t/kernel.py", line 2302, in console
    self._console_parse(command, channel=self._console_channel)
  File "/home/fabien/projets/meerk40t/meerk40t/kernel.py", line 2388, in _console_parse
    _=_,
  File "/home/fabien/projets/meerk40t/meerk40t/kernel.py", line 1180, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "/home/fabien/projets/meerk40t/meerk40t/core/planner.py", line 452, in plan
    self.execute()
  File "/home/fabien/projets/meerk40t/meerk40t/core/planner.py", line 566, in execute
    cmd()
  File "/home/fabien/projets/meerk40t/meerk40t/core/planner.py", line 686, in optimize_travel
    opt.optimize()
  File "/home/fabien/projets/meerk40t/meerk40t/core/optimizer.py", line 14, in optimize
    self.optimize_travel()
  File "/home/fabien/projets/meerk40t/meerk40t/core/optimizer.py", line 41, in optimize_travel
    self.cross(j, k)
  File "/home/fabien/projets/meerk40t/meerk40t/core/optimizer.py", line 112, in cross
    cutcode[q].reverse()
  File "/home/fabien/projets/meerk40t/meerk40t/core/cutcode.py", line 307, in reverse
    self.arc = copy(self.arc).reversed()
AttributeError: 'Arc' object has no attribute 'reversed'
```
[reproduce bug.dxf.txt](https://github.com/meerk40t/meerk40t/files/6476840/reproduce.bug.dxf.txt)

tatarize, 2021-05-14:
I removed the arcs from svgs and replaced them with bezier curves, but I didn't do that for dxf when they happen to include such things, they can issues that won't crop up elsewhere. But, arcs should reverse just fine. It might just be `reverse()` and messed up there.

tatarize, 2021-05-14:
https://github.com/meerk40t/meerk40t/commit/6641376bbc9d2a0ca1a531838aab312c5ef8db94

Found and fixed...

closed 2021-07-17 by tatarize