meerk40t/meerk40t #421: More stats for current operation in status bar

Sophist-UK, 2021-05-14:
Can w current operation in the status bar i.e.

###### Current
* USB status
* Controller active / inactive

####### Proposed
* USB Status
* Name, speed and settings for current operation (same as spooler table) - inactive when idle
* Percentage complete
* Time estimate remaining and total

tatarize, 2021-05-15:
Current is actually 3 items. USB status, controller status, and spooler status. Though spoolers don't really have a status anymore. So nothing ends up in that spot. Replacing that with percent complete would be pretty reasonable. The time estimate code needs a bit of work, I'd need to consider how to make the spooler give a more reasonable API for querying that since it's sort of needed, for several things.

There might end up being multiple usbs and multiple spoolers when versions gets finished. Though I guess there would be currently selected one at any particular time and that would be the one to be displayed.

The spooler tends to be filled with cutcode in 0.7.x versions which generally just says cutcode and doesn't give anything really about the current operation. This could be changed giving cutcodes information names or other useful bits of info. Each cut has its own settings so you could have current speed or power info. This stuff should be more compiled information than operations, though I could likely specify what the original operation that gave rise to the cutcode was, assuming there was one. Ruidaserver takes ruidacode which is basically just cutcode and was never defined as elements within an operation. This sort of intermediate step is the reason for cutcode.

We could give more specific information about the exactly which cut it's on. `Cutting 30/306 Bezier Curve...` as it readily has that information.

The important bit seems to be the complete amount and the time estimation. That seems like it might be better accomplished with a gauge rather than a status bar at all.

0.7.x has a AUI gui, it's not used for much currently but you can move stuff into their own windows and arrange them however you like. And then have them save. A particular completion gauge seems like it might better accomplish this main goal. I haven't messed around with making a bunch of small dockable AUI windows but, in theory any number of them would be pretty easy and possible.

Sophist-UK, 2021-05-15:
I am definitely not voting for yet another window - there are already too many windows and too much information on many of them. My vote would be to keep showing the current operation in the spooler window (whereas currently it disappears once it starts). This would also allow you to cancel the current operation and move to the next one.

As an aside I think we should distinguish between user information / buttons and debugging information / buttons and only display the former by default - with the latter made available using a menu option in the window. 

I am also unsure what the use case is for having multiple spoolers (or spool queues). If we wanted to support multiple K40s, can't we run multiple instances of Meerk40t?

tatarize, 2021-05-16:
A pane not a window. In 0.7.0 you'll notice you can move the tree around or that weird stop button. Those are panes. These are now saving in the 0.7.0-devices branch which means if you didn't want that it wouldn't need to be there but would be a part of the main window that could be cut into chunks of UI that you might like to know. And save your preferences allowing you to have that kind of toolbar if you wanted. Or a UI gauge for the particular project.

The spooler needs to fixed so those bits of info don't disappear so quickly and wrongly.

---

I do have a bit of an information overload where I think it better to have the information than not. I often tuck those debugging parts into console or otherwise hide them.

---

You can run multiple instances of MeerK40t. Especially if you use the "--profile x" cli option which will keep all your settings apart so they don't overlap and kinda fight each other. And that would be the typical method. Especially if you have other other differences like a different bed size from one device to the other.

There are some reasons you might want to do this. For example if you wanted to save a M2Nano datastream to disk currently the procedure is whacky (you need to process the entire operation then save the egv with either `egv_export` or from BufferView within the Controller. Or if you wanted to send data over your internal network to your raspberry pi that is listening on a particular port for your connection. So you'd just send your job to a particular spooler and thus to right spot on your network. Or if you wanted to do something like write some gcode out. That stuff gets easier if it's hooked to a different spool queue. In theory you can click job, select what device you want to send the job to, and have it dealt with. The analogy is printers where even if you only have one printer, you might still want to print to a PDF or something.

It might be a little bit overkill, but devices with mixing and matching output destinations and drivers has been a goal for a while now. You could check the 0.7.0-devices branch for how this is sort of shaping up, the job preview let you select the device you're sending the data to. And you'd typically only need the one, and the last one you used is the default so it works pretty reasonably. 

There was some multi-device support even back in 0.6.x but it tends to suck and makes two main windows and is weird and confusing. Here you can launch multiple meerk40ts with non-overlapping persistent settings and generally define a lot of stuff that could happen to spooled data. Going to two different K40s being just one of those things.

Sophist-UK, 2021-05-16:
1. It does seem like overkill to me, but it is your software and your decision and if the architecture makes it easy and reliable then it's not a problem.

2. Perhaps we can rename the USB window (which is USB logging debugging) to Debug and include other tabs holding other information...

tatarize, 2021-05-16:
1. It might actually be. But, I want some stuff that saves files to disk and sends stuff over the net and it seems like it needs an interface like that.
2. The newer devices code has custom windows for some of the main bits to configure preferences and the controller. And the point there was to give all the information for just lhystudios boards which included the usb log. If my backend is saving to a file, a USB log is meaningless to me. As such it could be that USB window could just be deleted.

[image]

Sophist-UK, 2021-05-16:
The USB window is IMO useful for debugging USB connection issues - and keeping debugging information hidden like this until it is needed prevents a novice user from being overwhelmed by excessive technicalities which will make MeerK40t seems daunting to use. 

However it does make sense for USB debugging information to be shown in the controller window because that is what it relates to. So perhaps we simply need to have several modes to run MeerK40t in with differing levels (and types) of information shown. As an example - and clearly there would need to be discussion about what stuff can be hidden from a Novice user etc. - you might have:

* Novice User
* Intermediate User (if there really are a lot of differences between Novice and Expert - if not don't bother)
* Expert User
* Debugging
* Developer

Each of these would show different panels by default in each Window - perhaps with a View Menu allowing panels to be toggled.

Have I described this sufficiently well to be understandable?

tatarize, 2021-05-16:
Yes. It certainly makes sense.

I just can't decide what each thing would be useful at what level. I would guess in preferences, Configuration would be disabled, since 98% of the time clicking anything in there will break your laser. But, if you need that stuff you shouldn't have to ask where it is. You build a funky laser that homes in the upper-right... okay.

"Fix rated to actual speed" and certainly Acceleration Charts. Everyday novice users might only need board setup (and that's actually a bit rare to need, since M2s are so common) and I could see a strong argument for "Pulse Shifting" since it'll have the effect of just making the laser stutter go away.

To some extent there is something like this within the operations properties. There's a whole second half of options within operation properties that control some things only Joe tends to know what they do.

I move some of the fancier and rarer settings into console since they are easier to write, easier to add things, and much more hidden. Which seems like a reasonable option. Rather than code up the acceleration chart, I could either put them in a custom hidden window without direct hooks into the gui and in console more directly. Though then it might feel like I'd have a lot of windows still even if a lot of them have no direct visible hooks.

And really there's a *lot* of features tucked into console already, interactions with inkscape, embroidery fills, shape drawing, path optimization circa 0.6.x, all image manipulations possible, image to vector tracing... a lot of stuff.

I certainly need to adjust some stuff to make more places to hide features.

Sophist-UK, 2021-05-17:
Lots of people have an aversion to command lines. IMO (for what its worth etc.) I think we should put everything into a GUI somewhere, which is why you need to hide the options for novices.

tatarize, 2021-05-19:
While a lot of things do have a gui, the console is mostly to make sure you can do everything with command line in case you need that for your workflow.

There are hundreds of little hidden features. I do not have the UX/UI skills needed to make that many weird options have GUI, such that they wouldn't seem obtrusive and cluttered. For example, since I have access to PIL there's literally a little ImageMagick loaded up in there, all the image things PIL can do, have a command. There's a ruidaserver (ruidacontrol, ruidadesign) that is able to make your Meerk40t pretend to be a Ruida so you can control it with lightburn or RDWorks. There's an eulerian embroidery fill routine. There's a set of routines that will take find inkscape and perform some inkscape commandline functions on an SVG before loading it. And that's not including a bunch of stuff that are beyond the sum of their parts. For example: `image vectrace engrave -s 20 plan blob spool driver -n lhystudios outfile daisy.txt` would take the image currently selected, trace the outline of the image into a vector path, put that path into an egrave operation, then plan that for the laser, turning it into cutcode and sending it to a spooler which converts it to lhymicro-gl code (lhystudios driver) which saves that into a file called daisy.txt. There's no way a GUI could do that. There's also basic things like draw a rectangle that you can do from the console or move the laser to some spot.

Though a lot of the typical methods is to use the keybinds that let you bind keys for the commands. Which is kinda gui access, all the bound keys do things so if you wanted to set a series of alternative home locations with a keypress you could do that, or rotate things while you hold key. There's a lot of power under the hood, and I have done a reasonable job mapping out a lot of most common things. When something new is needed it's a couple of lines of code or a clever bind away from being done.

Sophist-UK, 2021-05-19:
Yes - you have partially convinced me. I agree that these examples include a LOT of esoteric things that are so technical that a command line is essential.

But simple things like inserting a beep or a pause between operations should not need a knowledge of command lines, and another reason for a GUI is that the user may want to do it quickly before the current operation ends.

tatarize, 2021-05-20:
Yeah, still in the stalled feature thing (#264). I should maybe make a push to correct it and make spooler useful. It needs a couple reasonable things like not consuming the top element since I might want to see the information there. Or maybe cancel just that job and have it kill that one spooled object and move on the next (after maybe finishing the last cut). Or cancel that element, pause the rest, and perform a home before resuming because you're cutting your losses on that piece but the rest would be fine and independant.

tatarize, 2022-07-20:
There's a lot of stuff in status bar now.