meerk40t/meerk40t #422: Raster scripts turn transparent into white - can we keep it transparent?

Sophist-UK, 2021-05-15:
This is not a burn issue - white is same as transparent from burn perspective, but it is a UI issue when the white hides another object.

tatarize, 2021-05-15:
White is not transparent from a burn perspective, they actually differ notably when you deal with the `invert` button in raster-wizard and elsewhere. White in that case is the stuff that gets burnt, black is non-burn, and regardless of the invert or non-invert transparent is do-not-burn.

That said, console command `image remove -d 1 white` would remove all the white from your image. Though this is a lot slower than it needs to be. But, this converts from pure black and white into RGBA. There's not an easy method image type that lets you do black and transparent. One could do alpha mask pretty easily, but we're generally going to require some pretty fast conversion code. In theory if `putalpha()` works to make `mode='LA'` image and those generally would work this would generally be rather fast.

In theory this could be done giving us a view option for our images, sort of like images have a cached bitmap for speed speed and just have a white stripped image. 

---

The other aspect here beyond the couple images being somewhat weird at times is that this doesn't fix the bigger UI issue. Namely, if you have a stack of images or a overlapping objects you really can't see them. It seems a bit like it might be better to make everything not-selected appear semi-transparent or have some ways to toggle this stuff or hide layers from view or something. 

There's a feature called Z-Depth which basically is an option on images that will turn images into a series of grayscale images. So that darker colors don't just render to black but they render to black in most images. And lighter colors might appear in the lighter bands of the image but not in any of the darker ones. This allows you to perform rasters on all of these images and make genuine 3d images. There's also other options here to highlight a particular raster with a second pass of some selected cropped region.  See #24 for examples. Things like chopping an image into multiple smaller images for either rastering faster (there are some ways to optimize this stuff), or allowing overlapped objects, to emphasize some parts. All of this makes objects to be rastered *really* hard to see. And not all of them would be solved with a simple couple things of displaying white as transparent.

Though I'll see if there's an easy way to do that.

Sophist-UK, 2021-05-15:
My vote, for what it is worth (which is only small), would be:

A. For raster scripts to keep transparent as transparent and white as white. This would mean that the dithering functionality needs to work differently - an easy fix would be to check if all the pixels combined in the dither were all originally fully transparent, and if so revert the result to transparent.

B. Display all images in the UI with white translated to transparent, except when a single image is selected in which case show it as-is and on top of all other images. This way the UI image would always show the resulting burn, but if you wanted to examine a specific image more closely you could select it and zoom in.

tatarize, 2021-05-15:
I added a quick command (devices branch which is still rather broken) for `dewhite` which converts all the white to transparent in native PIL (it's fast without loops). https://github.com/meerk40t/meerk40t/commit/0057c074b1f1e08047821877fed924857a5bd765

A. This doesn't mean dithering has to change. Currently the way this is done is if there's an alpha on an image, it's peeled off and then applied on at the end to ensure those pixels are white. The wizard code calls trim which gets rid of pure white edges already. When these get transferred to transparent it'll have a perfectly reasonable bounding box for that.

---

B. That image could still be covered with other images, especially blacks in another image on top of it. I'm thinking fixing this aspect of UI is a bit harder. 

---

A wouldn't be too hard to implement. It would just be rasterwizard followed with a dewhite. I might even be able to put a checkbox in the dither. And I'd likely want to write a rastering filter for "LA" image mode. But, even the current code converting it to RGBA would work fine for those image.

closed 2021-08-12 by tatarize