meerk40t/meerk40t #424: 0.6.22: Passes

Sophist-UK, 2021-05-15:


tatarize, 2021-05-15:
Yes, this is the optimization. If an operation is flagged with passes which get processed during the conversion to cutcode. This means that the cutcode gets X copies of the commands. During the optimization phase all cutcode is processed to find a highly efficient path. This makes no distinction between first pass and second pass, and consequently it will cut them together. And doubling back on the same path is going to be a highly useful effective way of doing things since it has a distance of 0.

The operations are currently converted each to their own batch of cutcode. So every engrave becomes a cutcode. So if you duplicated the operations. I just tried this and managed to figure out the method to confirm #418. So if you make a second operation with the same properties and color. And refresh the classification, you'll get a copy of the cuts in both operations. (or duplicate the operation when it doesn't have any children).

Also turning the optimization off would run them in the correct order.

The optimization routines currently need some constraints so inner parts of cuts must cut before the outer-parts and to allow some kinds of priority rankings. This would clearly fall into that. It used to all pack into a single cutcode block and optimize across that block switching speeds as needed, which was a bit of a weird effect. But, if you had engraves and cuts you couldn't be sure the cuts would occur after the engraves which would have been weird if you had drop-outs so I put them into their own blobs/cutcode. I could probably do that for the passes too. At least until I finish the cut optimization part up. It's one of the two major outstanding bits in 0.7.0 left.

Sophist-UK, 2021-05-16:
But it doesn't. It will do simple closed paths 1, 2, 3, 1, 2, 3 but it splits combined paths and does 1a, 1b, 1a, 1b, 2, 3, 1c, 2, 3, 1c when according to your description it should do 1a, 1a, 1b, 1b, 1c, 1c, 2, 2, 3, 3.

tatarize, 2021-05-16:
It depends on how close those paths are to each other. It seeks to minimize the travel distance if doubling back helps with that it will do that. It's optimized with a two-opt algorithm which should find near close to the optimal algorithm (though not really provable given its TSP). It's allowed to double back but it's not necessarily the shortest path. The optimization depends entirely on the path distances and how to minimize those.

Sophist-UK, 2021-05-16:
If 1a, 1b and 1c are closed paths, then the nearest path to the end of 1a is ... um .. 1a because the start point of the 2nd pass of 1a is the exact same point as the end point of 1a.

There does appear IMO to be some some sort of bug.

tatarize, 2021-05-16:
There might be a bug in the optimization with regard to moving the first or last element so it might be the case that it's starting on 1a so it forced to keep that. And without starting on 1c it can't do 1c 1b 1a 1a 1b 1c 2 3 3 2... that or the jump from 2,3,1c twice isn't enough. And I think there's an iteration limit, so it might have been optimizing and then gave up.

I think the section is going to be rewritten with some options. First you can instantly get moderate results with the greedy algorithm. It's a fraction of the time, but it's like 80-90% of the improvements in travel time with very little timeloss. Though it does take an acceleration structure to do log(n) nearest point algorithm. And secondly I wrote a pretty fast version of this algorithm in numpy and I might port that in semi-optionally.

After that stuff, I'd need constraints between different objects within the algorithm for inner first and passes. And I think 0.7.0 would be good to go... assuming devices is finished and most of the bugs are checked off.

--

I'm thinking the optimization there is probably prematurely terminating, or has a particular end stuck and unmoving so it can't really fix that, or both.

Sophist-UK, 2021-05-17:
numpy is good - because with correct usage it can allow MeerK40t to do the compute-heavy optimisations outside the constraints of the GIL i.e. running multiple threads simultaneously on multiple-cores.

I am in the process of braindumping my thoughts on route optimisations - and will send them to you by email once complete.

tatarize, 2021-05-18:
Took like 80 lines, but it's a pretty solid algorithm for two-opt.

https://github.com/tatarize/vpype/blob/7b69a9b469847ac3bad67ba75b597dc99a942aad/vpype_cli/operations.py#L187

Tiger and I had a pretty fruitful discussion so hopefully Project Foresight (https://github.com/meerk40t/meerk40t/milestone/8) can be completed. Part of the issue with optimizations is that you basically walk into it blind. Regardless how good or bad the optimization are, not letting a person know what they are getting before we start firing the laser is a problem. 

Hopefully this window with a fast planning will solve that. Do all default operations that would be done in plan. Show the user for the accept/reject (or some other tweaks with cut planning). This could also be offered as an alternative instead of spool in the job planning.

[image]

In theory the optimizations for the cutcode can be done here, or tweaked to be more acceptable, or the first 74% of the cuts thrown away and just start a job of the last 26% of a job. There's a lot of potential there, but also letting people tweak and optimize things in place, so that if we get around to fancier cutplanning it will likely have a usable home.

Sophist-UK, 2021-05-18:
Hopefully you got my optimisations braindump. :-)

tatarize, 2021-05-18:
Your brain dumps are much better organized than mine...

closed 2021-08-12 by tatarize