meerk40t/meerk40t #425: SVG load order

Sophist-UK, 2021-05-17:
I use Inkscape for vector editing, and MK is loading the objects in reverse order to the object list in Inkscape.

I have tracked this down to Inkscape saving them in reverse order - so this is **not** a bug in MK.

But it would be great to have an MK option to load the objects in reverse order to that in the SVG file. :-)

tatarize, 2021-05-18:
Hm. Yeah, this would be pretty easy. I have this pretty worthlessly working for Operations. In case you want to reverse those for some reason. I'll add this for files and groups, and the elem-branch itself. (actually it's already in elem-branch) in theory if you ungroup everything even the file, you could reverse them in the element branch itself.

Done. https://github.com/meerk40t/meerk40t/commit/42bf12b5f74960f841b1e2a02ebbf6332293a59e

Sophist-UK, 2021-05-18:
Yes - I was aware that you can already reverse the order in an Operation. 

But what I was actually looking for was a Preference which says whenever you load an SVG file, load the items in the file object list, the Elements list and Operations in reverse order.

tatarize, 2021-05-19:
Hm. Yeah, okay. That'd go in the settings with the other quirky loading options.

tatarize, 2021-05-19:
Added it.

closed 2021-07-17 by tatarize