meerk40t/meerk40t #426: Layout for multiple SVG files

Sophist-UK, 2021-05-17:
I find that I want to cut leather out for multiple objects at a time. Currently I do this by combining SVG files through copying and pasting in inkscape in order to lay them out in the best way.

It would be great if MK allowed you to open multiple SVG files and for MK to then lay them out in the most efficient manner (smallest rectangular area in the defined bed size).

tatarize, 2021-05-18:
What you're describing is cut nesting. The algorithms to do it can be a bet heavy since you're doing a lot of different geometry to do this see #108. The general consensus of that issue is that we would be better off outsourcing this work to 3rd party software. I believed at the time that some abilities to call other software especially with a CLI would be extremely helpful here. After I wrote the `inkscape.py` plugin that lets you call some nifty inkscape functions I'm pretty sure that that would be the way forward.

https://deepnest.io/ had some pretty good work done on this topic. I think that's about where I was looking before it was tabled. If there's some commandline ability call such things a quick internal plugin could be added to call DeepNest or whatever other software exists. 

I'm not completely opposed to writing such algorithms but unlike a lot of things I have no experience dealing with nesting algorithms and from the research I did at the time, I wasn't sure the ultimate difficulty.

Sophist-UK, 2021-05-18:
I agree that as written this is a duplicate of #108.  And I agree that making a web services call is probably the best approach to deep nesting. Could we do something simpler in MeerK40t i.e. nothing clever, simply treat the SVG as a rectangular area and see of there is an empty area at least as big that we can put it in as an initial position (rather than loading all SVG files at the origin and making the user disentangle them)?

tatarize, 2021-05-18:
Hm. Yeah, there is certainly grounds for something as simple as load this file but put don't overlap it with the other stuff. I could maybe see putting this as a menu item on file. So "reload" and "reload without overlap" kinda of thing. 

Though in theory if you select the file in elements branch, you should just select all the elements of that file and can just move them from the selection box without anything else getting in your way.

closed 2021-08-12 by tatarize