meerk40t/meerk40t #427: Ratio for xy not in rotary mode ?

jhindes521, 2021-05-18:
Would it be possible to implement the custom ratio of x and y like in the rotary mode but for normal engraving / cutting for dimensionally accurate parts. 

This works with the rotary but it disables the auto home after completion  

Regards 
J

tatarize, 2021-05-18:
As a temporary work around you can add auto-origin (Return to Origin) after completion. This doesn't trigger the device's home function but rather sends the device to (0,0) or where meerK40t presumes 0,0 should be. 

In addition to the nerfed home, there's other reasons this slight difference matters usually with regard to the visualizations. The screen says what 1mm is but if your x steps are 0.93 mils rather than the 1:1 assumed, it's going to be an issue.

Also, you could scale the vectors manually, without rotary. In 0.6.x if you hit F9 it'll let you perform a transformation on the selected objects: `scale(0.93,1)` would for example reduce the x width by 7%

---

Also see #281

tatarize, 2022-07-20:
This should be mostly settled with the coordinate system in 0.8.x

closed 2022-07-20 by tatarize