meerk40t/meerk40t #429: Simulation and Optimizations, optimize until user stop.

tatarize, 2021-05-21:
Lightburn has order by groups and order by level or something. I think this is why. I gave additional consideration and I've ruled out simple constraints like giving them a number and that number must occur before the other number for each cut piece. Really you need to run the innermost closed shape first and anything that occurs completely within another closed shape. So this could be done in levels or groups. You could run everything that does not enclose anything, and do those first. Then move up to the next level of hierarchy because once you're done with that, anything that encloses an already cut element no longer encloses anything. Now if you call everything that encloses nothing to be level 1 then anything that encloses anything that is level 1 to be level 2, etc. You could build the hierarchy of enclosed objects.

[image]

So there are a lot of orders that these could be cut in. d, a, b, h, e, g, f, c ... etc.If you did so in pure levels you would do the bottommost elements of the cut hierarchy first. So d, b, e, g... then f, h, a then c. If you did this in groups... you would do (d,a), ((b), (g,f), (h), e, c). There's still work to be done here. Clearly there's a better way to represent these relationships and I'll need to do so.

https://user-images.githubusercontent.com/3302478/119078575-531eb680-b9ab-11eb-9249-b64948fef035.mp4

Sophist-UK, 2021-05-21:
Copied and pasted from email about this:

OK – suppose we didn’t have SVG groups:

I think that you should do the what’s-inside-what analysis first and group the paths accordingly, and then use TSP Greedy in a special way:
```
G1
    a
    G2
        d
b
G3
    c
    G4
        e
        G5
            f
            G6
                g
        h
```
My first cut algorithm is as follows: You start from home, and find the closest path in the top level i.e. `G1`, `b` or `G3` including the single closed path in the first level groups i.e. `a`, `b` or `c`. Assuming that you start at home in top left, this will be `a`. But before you laser `a` you need to do any contained insideness groups. So you iterate down the tree with the same algorithm and find `d`. Then, knowing how far `d` is from the current position, you look for other bottom-level paths that are closer in this case none. So you laser `d` and then repeat.

Using the same algorithm, but with home in bottom-left and ignoring `e` and `h`, you find `c` and then find `f` and then `g`, you check how near `g` is to bottom-left and then check lowest level groups for a closer item which you find is `a`. So you laser `a` and repeat.

I guess as a simpler alternative you just start with all paths that are not in a group which contains another nested group, and find the closest to the current point and laser that. Then repeat.

**However** I am unclear how you would then constrain the 2-opt to respect the cut-inside first constraint. 

It just gets more complicated with SVG groups. I suspect that **if** you are going to do inside-first optimisation, then you have to ignore groups entirely when analysing inside-out, creating the insideness groups as above - and then add as much of the SVG grouping to this as you can. As an edge example, if you grouped `a` and `g` together and `d` and `f` together, then this is incompatible with the inside-first groups above and would need to be ignored. But supposing `h` was actually a series of small circles in a group, in the grouping above `G4` would contain `e`, `G5` and all the individual items of `h` and you could then create `G7` containing the all the `h` paths i.e.

SVG groups
```
a
b
c
d
e
f
g
H
        h1
        h2
        h3
```
results in inside groups:
```
G1
    a
    G2
        d
b
G3
    c
    G4
        e
        G5
            f
            G6
                g
        h1
        h2
        h3
```
and then you merge in the SVG groups to get:
```
G1
    a
    G2
        d
b
G3
    c
    G4
        e
        G5
            f
            G6
                g
        H
            h1
            h2
            h3
```
You would probably need to note which groups were insideness groups and which were SVG groups, and I have not yet thought through the algorithm to decide how to handle groups where SVG group items are part of more than one insideness-groups.

tatarize, 2021-05-22:
My first choice of object is d, b, g, e, or h. I may not laser a, f, or c because they contain unlasered-enclosing objects. However, if I choose to laser d first then my choices at that point are a, b, g, e, or h. My lasering of d permits the newly permitted lasering of a.So it seems like the topology should be that of groups from the bottom up. Whereby you may only laser the candidate objects which are at the bottom most. Then when you laser an object all of those direct children get added as siblings within the tree. Except that it can't really work like that because sometimes objects will have additional additional constraints. Just because I lasered h doesn't mean I can laser c. Because e and (g->f) are still required. I think your datastructure is about right. And what I need to do is a search for all leaf nodes in the tree. I don't consider 'a' because it's not permitted. I set up the tree as you listed.

Root:
* A        
     * D
* B
* C
    * E
    * H         
    * F                 
        * G   

And the rule is I am only allowed to cut somebody if they have no children. I assume this is a would-be morality-system of a highwayman, you gotta cut the children first. Then when the parents no longer have children, you can cut them. So The order You take them in doesn't matter, just whether it's a leaf node. You could do D, B, E H, G. If you do G you free up F. If you do H you free up nothing. Now I can fairly trivially see how this would work with a greedy algorithm. You find the nearest object of D, B, E, H, G. Then if D, you find the nearest object of A, B, E, H, G. I don't however see how this can be accessed in a meaningful way by a 2-opt, algorithm since rastering an object removes a restriction, but the reverse of that operation doesn't actually work, so I can't go through there flipping the objects forwards and back without consequences. If you have a tree like this you could take it depth first or breadth first. So D, A, B, E, H, G, F, C (depth first) or B, D, E, H, G, F, A, C (breadth first) moving down levels until you find an unencumbered node. Or you could do a distance based selection while respecting the constraints. You could run two-opt on subpaths where constraints don't play a role. In breadth first, for example you could two-opt the  (B, D, E, H, G) and (F,A) and then (C). But, you could also optimize (B, D, E, H, G, F) for example if and only if F occurs *after* G. So (G, B, F, E, H) is a perfectly valid path, and it remains a valid path so long as F comes after G. You could add in A to that path with the caveat that A must come after D. And finally you can include C with the caveat that C must come after (E, H, F) it must come after G too but by definition F comes after G, so G need not be considered because if C is after F, C is by definition after G (so long as the path is valid).

Which leads to a somewhat useful conclusion. The tree itself defines the constraints. Valid paths are any paths where the parent's cutting occurs after it's immediate children (you need not consider grandchildren because if they got cut they didn't have any). And while the Parent -> children relationship is 1 to Many. The child to parent relationship is *always* 1:1 (except for root). So our constraints can be defined as the object that immediately encircles you, must be cut after you. So each object has 1 or 0 objects that cannot be after it. With 2-opt, you are reversing sublists within the main primary path, the constraints that you cannot reverse a path wherein any member is listed as a constraint of any other member (We would start with a valid greedy path and make no illegal moves, so you wouldn't be flipping these back into the correct order), though you could just search the subpath of each object in one direction. And the constraint tree might be easier in the Parent -> Children fashion. Since you could index the children and if any child is within the range being flipped it violates and that operation is disallowed.

tatarize, 2021-05-22:
The point isn't that you can't cut them in a single burn, but that you might be better off jumping ship burning something else and coming back around to that burn. What's the fastest way (shortest travel times) to burn this? The correct answer involves several partial complete cuts. If you were forced to select an object and burn it in one go. You would necessarily have a suboptimal path. In fact the optimal path clearly starts on the side of b closest to e.
 
  
[image]

----

Copied from email, making a topology point that is also relevant. Optimizations should include a burn closed shape in single burn option.

tatarize, 2021-06-04:
@Sophist-UK Writing here rather than an email, mostly for posterity sake. And as a break from the repeated bashing my head again this other crash bug.

## An overview of the current algorithm.

## `blob`

Each operation found in the plan is calls `as_blob` with the state of the `opt_inner_first` value that is set. If the previous item in our new plan is a cutcode, this cutcode is extended to the end of the previous cutcode. Which means that if there's a command between the cuts like a beep. If you did one pass, beep, another pass, beep. These would be required to be done separately. 

For each pass we do the following.
* For Cut or Engrave, we convert each PathSegment into a CutObject this is done by calling the relevant initializations.
* For `ImageOp` and `Raster`s, we convert the images into a CutObject specifically a RasterCut. If we set crosshatch on, then this actually creates two different raster cuts. Each image in ImageOp gets its own RasterCut, There should only be 1 object in a `Raster` because of the nature of the op.

Each operation's LaserSettings is given to each CutCode object. They technically have their own LaserSettings, but these are often the same object. And the code in the plotplanner checks for the sameness of the memory locations to know if it should switch up the properties for the next part of the cut. If one cut needs a different setting it can clone the LaserSettings object (to avoid sideeffect changes to the same object in memory) and change the property and the plotplanner will see if something like speed, accel, or dratio was changed, and adjust accordingly.

If there is a cut operation and we have flagged `cut_inner_first` the cutcode object gets flagged as `constrained` rather than `grouped`.

We `correct_empty` and delete any empty CutGroups, and if our Cutcode is found to include nothing, we return None, and have this ignored during the `blob` stage.

## We call inner_first_cutcode()

### Inner first cutcode:
* We extract all closed groups() these closed groups are placed at the start of the cutcode.
     *  Closed groups are any groups marked closed. This is imported from the original SVG object.
* We place all cuts that are not closed groups after this in a flat file.
* We brute force search every cut object and check if it is inside another object. This is true for cut objects and engrave objects alike. Any object found to be inside another object is marked with a list called `.contains` and the other object is marked in the other object in a list called `.inside`. This is done at the level of the svg elements of single non-merged paths.

The resulting cutcode from this algorithm is a cutgroups of closed object (max one level deep), and a flat list of non-closed items. These are marked as to whether they contain or inside each other.

## We call `short_travel_cutcode()`

### Short Travel cutcode:

1. We start at either 0,0 or a start defined in the cutcode object.
2. Of candidate cutcode, we brute force search all of them to find the nearest point.
3. We go to that point. Reversing the entry if needed. We proceed from the other endpoint. We mark this chose cutcode as `permitted=False`, repeat step 2

### Candidate cutcode:
* To be a candidate cutcode the we search whether the the current cutgroup contains any uncut objects. And that it is flagged as permitted=True.
* Containing any uncut object means every group group flagged in the `.contains` must not have `.permitted` set to True.

Note that the list of candidate cutcode can expand to include previous non-candidate cut code if *everything* in that cutcode has been consumed by the algorithm already.

### If, we did not optimize travel, we call `inner_selection_cutcode`
We find a list of candidate cutcode and add it to our new cutcode. We repeat until there are no candidates. This will cut everything innermost first, then the next level out, etc.

### If we don't optimize travel or inner cuts.
We had the cutcode over calling `.flat()` which does a depth first iteration of all the elements.

---

Our final CutCode in the plan is always just a flat list of cuts and the order they should be cut in.

---

My current plan. I'll go through and force the cutcode into competent orders and subdivide them into relevant sections based on user preferences. I will flag the .start value of each cutcode object as the optimize travel is done, ignoring operations like beep, that would not change the laser head's position.
* Users would get an option to cut passes in series rather than parallel, and this would make each pass it's own cutcode object.
* Users would get all their image/raster operations first.
* Their engrave operations second.
* Their cut operations third.
    *  Cut Operations could be further subdivided in the same manner as the stages of inner_selection_cutcode. Basically we find the initial list of candidates and we call that a cutcode object. We find the next list of candidates. We call that another cutcode object. We proceed in that manner until all cuts are spoken for.

We would, in essence be kneecapping the algorithm a little bit by making everything it's own cutcode level. We'd make up for a bit of this loss by using the `.start` flag in the cutcode to optimize for the transitions. We would then have purely non-constrained cutcode. Inside each cutcode object we could perform two-opt without any issue since nothing inside any cutcode would need to be before any other bit of the cutcode. The user options would effectively be deciding what gets made into it's own cut object. If inner-first is off then cutcode derived from Cut Operations can be a single set of objects. If passes is on, and passes is series is set, then each pass gets its own cut object. Otherwise they are all left in the same cutcode object (since you can take them in any order).

Sophist-UK, 2021-06-05:
@tatarize Kneecapping is IMO an understatement. 

IMO this loses the best part of the current algorithm namely that it optimises across operations. 

The current optimisation works pretty well - with the only issue being multiple passes happening too rapidly. We have tested with multiple Operation Passes, but I am not sure whether we have tested what happens with Add Passes or Duplicate Operation or Duplicate Elements  though I would expect the same thing to happen. And whilst it is trivial to identify paths which are being burned again with Operation Passes, it is presumably more difficult when the objects have been duplicated by using one of the other 3 options.

I think the ***golden rule*** we need to follow is to avoid optimisation **repeating the same path** any quicker than it would happen if the operations were done as displayed in the tree. And I think that what this means is that you can merge the first Operations of each type (treating Operation Passes like Duplicate Operations) and optimise them, and then see what operations (and Operation Passes) are left and merge those etc.

This would in most real-life cases pretty much guarantee that duplicate burns will not be done faster than in the tree - the exceptions that occur to me being A) Duplicate a Cut Operation and then change it to Engrave, B) do a cut as first operation and a duplicate of the Cut as last operation with (say) an image in between because the image might be optimised to do first followed by the first cut which would then be immediately followed by the second cut (which might well be the reverse of the first cut because of the head position - this immediately burning the same path twice in quick succession).

Perhaps the way out of this is to pop-up a warning if we detect that the type and number of operations is such that optimisation may do repeat burns in quick succession.

So, how about this for an algorithm which will give most of the benefit for simple cases, and still give reasonable benefits for more complex cases:

1. From an Optimisation perspective, treat Operation Passes the same as if they were Duplicate Operations.
2. Select **only** the first Operation of each type (Image, Raster, Engrave, Cut) and merge them for optimisation.
3. Then do a virtual Home (to try to get Duplicate Operations to cut the same route) and repeat with any remaining operations until they are all done.

(This is essentially a sophisticated example of mult(opt(objects)).)

As you have previously pointed out, one or more special operations (Home, Beep, Interrupt) between two normal operations would also be a break point for merging operations to optimise across them. So

* Image
* Raster
* Beep
* Engrave
* Cut

would merge the Image and Raster operations and optimise them and then merge the Engrave and Cut operations and optimise them.

BUT ... I do think that the user needs more control over e.g. merging of operations regardless. They may want them done separately and in the order in the tree. So the optimisation options we need might be:

* [ ] Fix closed paths
* [ ] Inside-Out
* [ ] Optimise travel time
* [ ] Optimise across Operations

As previously requested, I would still like multiple spool jobs to be created for:

* Unmerged Operations
* Each of the merged operations groups above

Is there any way of getting the Name of the Spool job to indicate the operations that are included (like 0.6.22)?

tatarize, 2021-06-09:
> Kneecapping is IMO an understatement.

It's not that striking. Good optimizations will have major gains, a few roadblocks make for minor reductions to that. 

>Note: I believe that the issues so far have resulted from multiple Operation Passes, but I am not sure whether we have tested what happens with Add Passes or Duplicate Operation or Duplicate Elements.

They should be about the same. The code will end up adding cutcode to the same general places.

>I think the golden rule we need to follow is to avoid optimisation repeating the same path any quicker than it would happen if the operations were done as displayed in the tree.

By definition that requires no optimization ever occur.

1. From an Optimisation perspective, treat Operation Passes the same as if they were Duplicate Operations.

We already do that. However, since operations are mixed and matched in the optimizations it'll end up with the same result.

> 2. Select only the first Operation of each type (Image, Raster, Engrave, Cut) and merge them for optimisation.

If you want cut then a different set of engraves it shouldn't force the one engrave into the other engrave.

>3. Then do a virtual Home (to try to get Duplicate Operations to cut the same route) and repeat with any remaining operations until they are all done.

This would be fine for step&repeat. Just do the same operation multiple times and you can do this with a real shift origin point which work pretty nicely. It also keeps a lot of things simple.

> Is there any way of getting the Name of the Spool job to indicate the operations that are included (like 0.6.22)?

0.6.22 actually put the operations in the spooler. They were a bit different and could spool the operations themselves. It wasn't a giving it the name but actively actually was the operation. During the morphing of the cutcode this could certainly be done to assign it a name.

Sophist-UK, 2021-06-09:
> > Kneecapping is IMO an understatement.
> 
> It's not that striking. Good optimizations will have major gains, a few roadblocks make for minor reductions to that.

Cross operation optimisation is an amazing benefit. Please don't remove it.

> > Note: I believe that the issues so far have resulted from multiple Operation Passes, but I am not sure whether we have tested what happens with Add Passes or Duplicate Operation or Duplicate Elements.
> 
> They should be about the same. The code will end up adding cutcode to the same general places.

That is my expectation too. The key point is that without optimisations they all do the paths in an operation and then do the same paths in the same order again. Add passes and Duplicate Elements both add the same elements to the end of the same operation, having the same effect as Operation Passes. Duplicate Operation is similar if the new duplicate immediately follows the operation that was duplicated.

> > I think the golden rule we need to follow is to avoid optimisation repeating the same path any quicker than it would happen if the operations were done as displayed in the tree.
> 
> By definition that requires no optimization ever occur.

Well - not strictly true, but it does demonstrate that my own statement was a stupid one and didn't explain what I was trying to say.

If the user wants to be explicit about the order of paths, they will turn optimisation off. So we should assume that they are **not** worried about the path order and that optimisations can reduce the time taken for one pass of the original elements (before Add passes or Duplicate Elements). Let's call the original set of elements for the operation, S0 and the time taken if you optimise this set of burns T0.

So long we do convert Operation Passes, and Add passes to Cutcode by repeating the same optimised code again we guarantee not to cut the same path in less than T0. 

Ideally we would know that a Duplicate Operation is the same and use the same optimised route BUT the user may have changed the elements in the duplicate so this may not be possible to achieve - so I think that we should probably treat a Duplicate Operation separately from an optimisation perspective and warn the user in the Wiki help that Duplicate Operations will be optimised separately and may result in elements being burned in a different order resulting in reduced interval between the repeats, in a worst case the same path being burned again immediately.

Duplicate Elements is more difficult because it could be a subset. - and I am not yet absolutely sure how this should be handled. But my first gut reaction is to use the optimised path from the full set of elements and just omit the ones which are not in the subset because this would result in a similar delay to an unoptimised burn of the duplicated elements.

Using this as a base, optimising across operations should never result in a burn interval shorter than above. So, if we optimise across Image (1 pass) and engrave (2 passes) and cut (3-passes) we would start by optimising across all three operations, and then take the optimised path and for pass 2 simply omit the images, and for pass 3 omit the images and engraves. This is still likely to be reasonably well optimised.

To avoid Duplicate Operations being optimised merged, as previously suggested we would only ever merge one engrave and one cut operation. And having done those as above we would then repeat with another set of merged operations from whatever was left.

Obviously we can find edge cases where the optimisation would not achieve much, but in the vast majority of real-life burns, this would probably be pretty effective.

> > 1. Select only the first Operation of each type (Image, Raster, Engrave, Cut) and merge them for optimisation.
> 
> If you want cut then a different set of engraves it shouldn't force the one engrave into the other engrave.

Not sure what you mean.

> > 1. Then do a virtual Home (to try to get Duplicate Operations to cut the same route) and repeat with any remaining operations until they are all done.
> 
> This would be fine for step&repeat. Just do the same operation multiple times and you can do this with a real shift origin point which work pretty nicely. It also keeps a lot of things simple.

I am not sure I explained myself clearly. By "virtual home" I mean you don't actually move the head to home, but when doing greedy optimisation, you assume that it has moved there in order to try to get a consistent optimisation of paths for Uplicate Operations. If you start each optimisation as if the head was at the Home position, each optimisation should come out the same. 

> 0.6.22 actually put the operations in the spooler. They were a bit different and could spool the operations themselves. It wasn't a giving it the name but actively actually was the operation. During [...]

Again, I am not sure if I explained myself clearly. We engrave and cut leather. Leather is a natural but processed material, which comes in different thicknesses, different densities, different colours, different processing (i.e. chemicals) and different finishes. Burning this is part science, part art, and you need to watch the burn and adjust the number of cut operations in real time. In 0.6, I get a lot of operations in the spooler, so if I Duplicate Operation the cut operations, then when the laser has started the Operation that actually finally cuts through the leather I can go into the spooler and del [... cut ...]
I also tend to do cut multiple pieces of leather from a single piece of material - and many other people do the same e.g. the inpain's birdhouse. And an alternative use case is to have the spooler separate out each separate piece (definition: there is an outermost closed-path **cut** with no burns joining or crossing) and complete all operations on that piece before moving onto another piece, and having separate spool jobs for each piece. This is a more difficult use case to achieve, and perhaps should not be attempted in 0.7.0.

So not only would I like a better description of what is in a spooled burn, I would like better granularity.