meerk40t/meerk40t #430: 0.7beta: Terminal Window help needs splitting into sections

Sophist-UK, 2021-05-23:
Terminal Window help is now IMO too big to be displayed as a whole.

We need `Help` to display a list of sections with a brief description and an annotation that `Help All` will give what you get today.

`Help section` should give the current list of commands for the relevant section as is.

Also a clickable link to the relevant Terminal Commands Wiki page (via an unmutable redirect page) would also be very useful at the end of every Help output.

tatarize, 2021-05-23:
https://github.com/meerk40t/meerk40t/commit/7b2c576fcd8a6d33ad9a2b874c51d435f54ad5af

Typing `webhelp` would show the uris that a couple webhelp things go to, and thus are clickable.

Okay. URIs will show up as clickable. Help might need to be redesigned a bit to make it easily navigated. It's automatically generated so it's not just trivial. The help <topic>are if they have extended help in them like rect does for example, are doc-strings. A lot of the sections need to require that you query them rather than showing everything at once. Or something. 

I can't really hard code uris directly into help because in theory `kernel` is intended to be a stand alone project. But redesigning the help system could certainly be easily done. Registering additional bits of information in it. It's modeled a little bit after Click and a bit after Vpype's command pipeline, for console stuff. Individual commands can easily have extended help.

Also, there's some issues with the presentation of the material within help that could be changed as needed. If you see any really useful help system for something similar. Since help is built within console it would be pretty easy to turn the commands into longer run of type help. I think registering some command groups (not just going by the command data input output stuff) could be useful. Then writing additional help pages (they could be registered in the kernel pretty easily) might be easily doable here. With URIs to additional help, and overarching concepts like how command stringing work. Etc.

tatarize, 2021-05-23:
To help with the terms that get muddied I made Terminal window be called Console window. It was initially a context thing since Console started as an actual class rather than a core part of kernel. So Terminal got the name because it was a terminal into the console. But, this creates the oddity that you open terminal because I wanted something typed in the console. While it might be more confusing to developers to say console vs. Console window it's likely a better way of doing since its called console everywhere.

Sophist-UK, 2021-05-23:
Console is fine as far as I am concerned. As for developers, if they cannot cope with this small level of confusion then what hope do they have of developing anything (esp. given the discussion we had last night which I found confusing and I have a pretty good level of technicality).

I am going to have to find the time to create a run-from-source environment with git so I can test this sort of stuff without you needing to create another beta executable.

But I cannot stress how important I believe the documentation and in-app help and usability are. IMO they sit between basic functionality that everyone wants (1st priority) and advanced functionality that only a few people want (3rd priority) - in other words usability is pretty darn important.

Despite MeerK80t already being superior to Whisperer (IMO) and Whisperer being functionally stabilised and having significant bugs, MeerK40t is still not the software that most K40 users use and recommend to others. And usability will be key to changing that.

tatarize, 2021-05-24:
Yeah, there some methods of looking for help that redirects to the Wiki. And more is fine. Part of the reason is that with rather constant development 0.7.0 was notably hard to document, since it was in fairly constant flux. Guides built for earlier forms of the beta would be missing significant aspects of functionality. The hope is to finish up 0.7.0 and go ahead and then try to stabilize everything.

Sophist-UK, 2021-05-24:
Sounds good.

tatarize, 2021-06-20:
Console help for channels is kinda crap. 

```
help channel
[21:08:36] 	channel 
[21:08:36] 	(None) -> channel -> (channel)
```

Sophist-UK, 2023-04-25:
I envisaged the help command simply using the existing internal data and presenting it differently.

`help` by itself would list commands and one-line descriptions of those commands that have `(None) -> ...` or those currently considered `Base Commands` plus some suffix text saying how to get more detailed help for any of the above commands listed.

`help command` would list the full description for the actual command and one-liners for all commands that take the output type of that command as an input type.

So as an example:
```
 > help
The following Base Commands can be used to kick off console functions:

...
channel         Channels are the means that MK can communicate internally and with the user
...

Base commands return a type of data that can be chained into another command that takes that type of data as input.

For more help on any of the above use 'help <base command>'.

For more help on the help command itself use 'help help'.

> help channel

channel        Channels are the means that MK can communicate internally and with the user
    	(None) -> channel -> (channel)

The 'channel' base command can be followed by any of the following channel-commands:
     list              list the channels open in the kernel (DEFAULT)
     close           stop watching this channel in the console
     open          watch this channel in the console
     print           print this channel to the standard out
     save            save this channel to disk
 
Here is the current 'Channels Active' list:
       1: console
       2: kernel-lifecycle
       3: lookup
       4: module-lifecycle
       5: service-lifecycle
       6: optimize
       7: m2nano/events
       8: m2nano/usb
       9: m2nano/usb_send
       10: m2nano/recv
       11: threads
       12: scene
       13: scene-events
       14: keybinds
       15: profiler
       16: signals
       17: load
       18: classify

For more help type 'help channel <channel-command>' or help <channel-command>'.

> help channel list
     	list            list the channels open in the kernel
     	(channel) -> list -> (channel)

> help list
     	wordlist list <key:str>
     	(wordlist) -> list -> (wordlist)
     	Argument: str 'key':
     		Wordlist value
     
     	materials list <name:str>
     	(materials) -> list -> (materials)
     	Argument: str 'name':
     		Name to display the materials from
     
     	ops list 
     	(ops) -> list -> (ops)
          
     	elements list 
     	(elements) -> list -> (elements)
     
     	tree list 
     	(tree) -> list -> (tree)
     
     	clipboard list 
     	(clipboard) -> list -> (None)
     
     	window list 
     	(window) -> list -> (window)
     
     	spooler list 
     	(spooler) -> list -> (spooler)
     
    	camera list 
     	(camera) -> list -> (None)
     
     	channel list 
     	(channel) -> list -> (channel)

>
```

The key internal structures for this already exist. Some tweaks to text will be needed. Ideally colours can be included to help guide the eye and emphasise certain text.

tatarize, 2023-04-27:
There's a command for `help -i elements` for example that lets you search all the commands that take `elements` context as input. And a couple other things that makes help more usable. There's also a bunch of ways to write additional help for commands but isn't done that much or that extensively. There's minor and extended help etc. But, it's still a giant list of random-ish commands for the most part.

Sophist-UK, 2023-04-27:
Yes - I do know about `-i` but this is about user friendliness - and the point of my post is that `-i` shows that the data already exists to support my suggestion - it is simply using the existing data to provide help in a more friendly form for user who are new to MK or who e.g. might find a command line intimidating.

By creating a more friendly version of `help` you will position MK for incremental improvements to the help text itself.

Sophist-UK, 2023-12-16:
Well today I wanted to use the `speed` console command, and I tried (yet again) to use the console `help` function and it has not improved one iota since my April suggestion.

`help speed` says I need a `lihuiyu` input to `speed`, but `help -o lihuiyu` lists nothing (and to rub it in it doesn't give me a message to tell me there are no outputs). I tried a few obvious commands like `spool` but experimentation didn't solve it either. So I have no idea how to do this - the help **functionality** was useless in this case.

Leaving aside the functionality of `help`, the usability is still crap. This is partly because the way `help` works is unfriendly, and partly because the actual help isn't helpful to the average user (and I would say that I am more experienced with MK than many). The help is inconsistent in approach, uses too much jargon (which assumes you already understand what the command does - which is pretty darn silly for a help system). Examples:

```
find            find <substr>
flush           flush current settings to disk
``` 
So for `flush` I get a helpful description of what the command does, but for `find` I have no idea what it is using as a base within which to find the substring. This is a VERY common inconsistency whereby some commands get a description of what it does, and some a syntax of how to use it - and these are completely different use cases.

If I ask for detailed help on these two commands I get:
```
help find
    	'find' will display the list of accepted commands that contain a given substr.
     
     
     	find <substr:str>
     	(None) -> find -> (None)
     	Argument: str 'substr':

help flush
     	flush 
    	(None) -> flush -> (None)
```

Leaving aside the minor issue of how many blank lines separate the output sections, the `find` command now has the description which would have been useful earlier (though `help-find` would be a better command name IMO), whilst flush has now lost the description and the help is a useless tautology, adding absolutely nothing not someone's knowledge.

In another example, how does this add **any** knowledge to the user about the command???:
```
help
penbox          Penbox base operation
help penbox
    	penbox <key:str>
    	(None) -> penbox -> (penbox)
    	Argument: str 'key':
    		Penbox key
```
I had zero knowledge of what Penbox was when I started, and I still have zero knowledge after asking for help.

```
find speed
    Commands:
    move_at_speed
    lihuiyu speed
    ops speed
    
    Params:
    move_at_speed speed dx dy
    lihuiyu speed speed
    ops speed speed
```
I am unclear why `find` essentially tells me everything twice. But I tried `lihuiyu speed` as it suggested and got `lihuiyu is not a registered command in this context: Base` so that was pretty unhelpful.

```
help
     wait
help wait
   	Provide a wait time. This is executed within the current thread. If called from the gui thread this will lag the redrawing of frames etc. This command is intended to be called by drivers or in other special events where calling the console commands is entirely acceptable.
  

    	wait <sleeptime:float>
    	(None) -> wait -> (None)
     	Argument: float 'sleeptime':
   		Wait time in seconds
```
Here is some actually useful help (well, it would be if I was actually interested in the `wait` command).

I would also suggest that there are some `base` commands which should not be `base` commands.

`abort` and `estop` are base commands but if I am running several jobs on different lasers at the same time, which one is supposed to be aborted? Also, are these the same commands? The help is no help in understanding which job(s) get aborted, whether `abort` is identical to or different from `estop` etc.

`acceleration` seems to me to most likely only apply to m2 nano and similar, but again which job or op is it supposed to apply to?

Finally, since MK wants you to string commands together, I would suggest a new means of help. Example:

1. I type `wordlist ?`, and it gives me a numbered list of the commands that I can use after wordlist:
    ```
    1. add
    2. addcounter
    3. backup
    4. get
    5. index
    6. list
    7. load
    8. restore
    9. set
    ```
2. It then prompts in the command input area with `wordlist ` and I type `5` and immediately it replaces the `5` with `index ` and then gives me the syntax definition of `wordlist index` so I can add the arguments. If I had selected `6` instead it would replace `6` with `list ` and since that has no arguments it should then wait for me to press enter.

I should add that wordlist without any following arguments should assume `list` rather than output literally nothing.

Summary: If you want users to be able to harness the power of the console, then you have to provide much much much better help than this.

Sophist-UK, 2023-12-16:
P.S. Changing the subject to refer to v0.9.x since this has not changed or improved in that time.

Sophist-UK, 2024-10-04:
This is a set of well thought out ways to improve the help command that use the existing help data but present it better.

The console commands are a very powerful tool, but users need to know how to use them - and the current help system is no help.

This is, IMO, another example of a "fuck you" moment - closing and effectively losing from sight valuable suggestions just to make the queue look better.

jpirnay, 2024-10-05:
While you might be right, the current workload will not allow it, but reopened to allow some one else to deal with it at a later time.