meerk40t/meerk40t #432: Arrange RibbonBar in more robust fashion. Should have an method to madness.

tatarize, 2021-05-24:
[image]

Currently I think maybe Controller and Preferences should be renamed and tasked. Preferences are specifically the Driver Preferences of the current device and Controller monitors the Output of the current Device.

Sophist-UK, 2021-05-24:
Better - but I would suggest:
* Start Job -> Execute Job
* Compile and Simulate -> Simulate
* Controller I am fine with.
* Preferences -> Configuration
* Settings -> Preferences
* Group: Main -> Project
* Group: Window -> Control
* Move Console to the end of Main group.
* Move Pause (/Resume) to the right of Spooler - or with the big Stop / Home buttons at the bottom (not showing in the pic above)
* Settings and Preferences should have different icons.
* Usb button - see previous discussion about debug - don't you get the same info in an optional panel of the controller window? Can we dispense with this?

tatarize, 2021-05-24:
[image]

Fetching a new icon took the most time. Though I did find a nice ram chip icon. Not sure what it'll be used for but it's clearly perfect for something.

tatarize, 2021-05-24:
Also, of interest. This is a lot of real estate for 17 buttons.

tatarize, 2021-05-24:
[image]

I lean towards this a bit. Mostly because Devices are a real thing now. And they basically setup the Spooler, Config, and Controller. And Pause looked dumb left in windows. Maybe Navigator would want to move too leaving camera all alone in Window.

Sophist-UK, 2021-05-24:
I still think pause is like the emergency cancel or home panels. And Devices and Config definitely belong in Settings. And Spooler and Controller seem to fit with Navigation and Camera into a set of laser controls.

So I preferred the previous design plus my changes to this.

Sophist-UK, 2021-05-24:
My thinking is more about what buttons get used when i.e. user centric.

tatarize, 2021-05-24:
[image]

Well I couldn't get the ribbon bar to minimize but I made the little arrow there let you get the taskbar thing for it and consequently close it. I added a couple panes like navigation for show. These could be disabled or put where-ever, though highly customizable some consideration needs to be given towards initial arrangements.

tatarize, 2021-05-24:
https://user-images.githubusercontent.com/3302478/119372953-c014a380-bc6c-11eb-8370-198b0b092cd1.mp4

tatarize, 2021-05-24:
The problem I have is that config and controller change in other devices. They might even be absent. That's because classwise those really are devices.

We could throw away pause and do that in a pane if folks wanted such a thing. It's a pretty easy change. In fact, done.

Sophist-UK, 2021-05-24:
I quite like the idea that you can move the buttonbar anywhere or close it - can we have that as standard?

And I see we now have View/Panes/stuff which is great.

So dispense with the up/down arrow, and if possible move the help button to the end of the menu bar - or if not add a Help menu which has a variety of places to go for help.

tatarize, 2021-05-24:
[image]

Sophist-UK, 2021-05-24:
I like that more than a button on the button bar. Does it switch to a resume icon when you click it?

If Config and Controller change with the device, (understandable) and if the other device doesn't have these, then we simply grey out the buttons or remove them.

tatarize, 2021-05-24:
Help is a on the menu bar, language goes last because it's added dynamically if there are translations. But, help is always there and launches the wiki stuff etc. 

It didn't... but now it does.

Doesn't update though if something else changes it.

tatarize, 2021-05-24:
Currently the buttons launch a weird thing that says there's no gui for this thing. But, others like File or TCP might well be better off with a different icon and name. Or a different button that is turned on and the other turned off as needed.

tatarize, 2021-05-24:
https://user-images.githubusercontent.com/3302478/119376254-965d7b80-bc70-11eb-83c0-f53efbfd06db.mp4

tatarize, 2021-05-24:
I'll get to whatever ideas you have in due course, heading to bed. I'm pretty close to having optimization working. I need a few more tweaks but the simulation is generally kinda reasonable. I did set passes to 3 and it went a bit crazy. But, that might be a side-effect bug. And the rotary scaling effects the main scene, and a couple more known bugs. But, it's almost to big old bug-hunt time.

closed 2021-06-20 by tatarize