meerk40t/meerk40t #433: Feature request: Do Nothing Color - registration marks?

inspectionsbybob, 2021-05-26:
I would like a option to have a color that does nothing... to be used for **Registration marks**.

I know I can Set PPI to "0" but I think the head will still move and act on the elements thus taking up time and resources, and complicating the Operations tab and re-rendering...
Would like an "Mode Type" of "Registration Marks" that is mostly ignored by the system so that you can tightly register a job on an object

Sophist-UK, 2021-05-26:
In 0.7 you will be able to disable an operation if that helps.

inspectionsbybob, 2021-05-26:
I guess the question is, if I am working with Inkscape for example, and
want to layout something, and I make my corner marks in "Green" for example
and everything else is Red and Black, Cut and engrave, will the Green move
with everything so I can use it for positioning?

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

tatarize, 2021-05-27:
I believe they would if they were in the same Group object, in 0.7.0. Moving things apart from each other was a stumbling block in 0.6.x but required groups and some work to fix it, and that work was in 0.7.0.

tatarize, 2021-05-27:
If you put green color within the group, and flagged green to be a disabled operation for stroke color, it would let you do that.

tatarize, 2021-05-27:
Though adding this directly in rather than having a disabled operation would have an operation that does nothing.

inspectionsbybob, 2021-05-27:
I will need to play with 0.70 to make sure it does what I think others
might want...
Been working mostly with the 3D printers recently, time to switch back....

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

Sophist-UK, 2021-08-12:
@inspectionsbybob I am working on another classification PR. Amongst other things, it will classify white strokes with no fill into an Engrave operations which will be disabled by default. Would this meet your needs?

Alternatively you can define your own disabled operation with a special colour for registration marks (other than white so that they are visible) and save it to your Operations profiles.

The classification part is done - and working. there are some minor GUI changes that are needed and I will push it - and then you can give it a try (if you don't mind running from source) by downloading the Zip file for the PR branch.

tatarize, 2021-08-12:
You can also not see white. So it might be a problem in itself. You might be able to use dots for this but they'd be a little hard. You might also add the element but then do nothing with the op. So you set a color and then disable the operation that matches that color. This seems like the best workaround there.

inspectionsbybob, 2021-08-13:
Yes it would. Go ahead and close the file

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

closed 2022-07-20 by tatarize