meerk40t/meerk40t #435: Prerelease 1: Testing... Report any issues found here.

tatarize, 2021-05-27:
https://github.com/meerk40t/meerk40t/releases/tag/0.7.0-beta26

The latest beta should be the end of needed features, from here it's just bugs and reporting them and turning off features that do not work. UI Tweaks and requests are also quite welcome.

Sophist-UK, 2021-05-27:
Amazingly fast and stable for a beta!! Going to start to use this for production. 

# UI
1. Load a file. Tree/Elements is closed - open it to show the files loaded.
2. Tools menu - should we have Windows in a submenu like Panes?
3. Tools menu - move Reset Panes to the Panes sub menu (or take out the separator between Reset Windows and Reset Panes).
4. Paths etc. have the friendly Inkscape names, but groups don't.
5. Panes do not yet have titles
6. What is the purpose of the top-level "Project" trunk in the tree? Do we need this for technical reasons or can we dispense with it? It does have a context menu "Expand all Children" which makes it potentially useful - especially if we can have "Collapse All Children" too.
7. Background colour for selected objects is garish yellow - can we have something a bit tamer like light grey?
8. Can we dispense with tabs on the button bar and have each tab in a separate pane?
9. Pause is available as a pane - can we lose it from the button bar or does that make it too difficult to find?
10. Can we add the emergency stop pane as a button also.
11. I like the Project button group. Can we rename Windows group to "Control" and include Spooler, Controller, Pause and Stop on it. Devices, Config, Settings, Keymap and Rotary should be in a "Settings" group. Can we rename the Settings window and button to "Preferences"?
12. Need a way to remove the camera image from background.
13. Execute Job window has incorrect caption Preview Job.
14. When I click Execute Job, why do I have to click Copy to populate the Window rather than have it pre-populated?
15. Execute Job window - Tooltip for Copy is "Start the Laser Job" rather than an explanation of Copy. Ditto for later buttons.
16. Cursor shape needs to change when MK is doing something like Execute Job / Validate.
17. Help/About - needs a margin. Text needs to flow i.e. between "continuing" and "participation". Text after "This is the brainchild of" cannot be seen. "Thanks" should be a header plus "Big thank yous to:" and then a bullet list of people with explanations. You might want to separate out significant contributors into a Contributions thank you list.
18. Can we make windows into panes i.e. so we can dock e.g. the Notes window? Can we make buttons for e.g. Notes toggles so that you can click the button again to close the window? Can we have ticks in Tools menu against open Windows / Pains like in the View menu?
19. Console Help needs to be split as previously discussed.
20. Help/Webpage should be Help/Github. Also add Help//Releases to point to the GH Release page. Plus Help/Facebook and Help/IRC and Help/Report bug. Plus separator before Help/About.
21. File/New should be File/Clear (IMO).
22. Order of Tools menu should match button bar. Names should match button bar (JobSpooler vs, Spooler)
23. Camera window - white drag circles VERY laggy indeed (on an I7) - very noticeable considering how fast the rest of MK is.
24. Settings window needs indicator if stuff is off the bottom of the window (like the Disable Tooltips option). Scroll bars? Or perhaps make it fixed size so everything shows. Also check boxes need to start a few px lower.
25. File/Recent/Clear Recent needs a separator before it.
26. Can we move Tools windows into a submenu Tools/Windows.

# Bugs
1. Adjust horizontal window size to the right amount and the Windows button groups minimises to just a drop down arrow. Click on the arrow and MK crashes.
2.  SVG file with image without a user friendly name in a group with a user friendly name. Group friendly name was used instead of imagexxxxxx. 
3. Execute Job / Blob with no laser connected gave a crash. Details sent 22:50 BST 27/5.

joerlane, 2021-05-28:
Object selection in the scene is not reflected in the Tree's current selection. 
Clicking on an object in the scene does not allow individual selection.
Drag selecting bottom up does not select objects who's corner point is not in the selection area. For example; trying to select a rectangle from a selection in its middle section and not across a corner point.

tatarize, 2021-05-28:
- Load a file. Tree/Elements is closed - open it to show the files loaded.
Ok. Done.

- Tools menu - should we have Windows in a submenu like Panes?
Maybe.

- Tools menu - move Reset Panes to the Panes sub menu (or take out the separator between Reset Windows and Reset Panes).
Ok. Done

- Panes do not yet have titles
You haven't reset your panes. Your view is just from a version of MeerK40t before those were installed.

- What is the purpose of the top-level "Project" trunk in the tree? Do we need this for technical reasons or can we dispense with it? It does have a context menu "Expand all Children" which makes it potentially useful - especially if we can have "Collapse All Children" too.

I was considering adding some functions there, but yeah, unneeded.
Ok. Done.

- Background colour for selected objects is garish yellow - can we have something a bit tamer like light grey?
Done. Though now the angry fruit salad Magenta doesn't look right.
https://github.com/meerk40t/meerk40t/blob/f5877f3ccf6a0f0b20f9b3d77435a48c15b36d94/meerk40t/gui/wxmeerk40t.py#L3041

Has all those colors, feel free to find a theme that works. Also, keep in mind that Tiger and Frog and some other run Dark Mode (OSX/Linux Only) and it might be best if it doesn't go weird against that theme either.

- Can we dispense with tabs on the button bar and have each tab in a separate pane?
Probably. Though I tried and the whole thing got weird. I turned off toolbox and I'll make a note to make Position into a Pane.

- Pause is available as a pane - can we lose it from the button bar or does that make it too difficult to find?
I don't know. I am unaware of which parts get used and which doesn't. 

- Can we add the emergency stop pane as a button also.
We certainly could but if we lost it from the ribbon bar it would seem a bit strange.
Ok. Done.

- I like the Project button group. Can we rename Windows group to "Control" and include Spooler, Controller, Pause and Stop on it. Devices, Config, Settings, Keymap and Rotary should be in a "Settings" group. Can we rename the Settings window and button to "Preferences"?
Ok. Done.

- Execute Job window has incorrect caption Preview Job.
Ok. Done.

- When I click Execute Job, why do I have to click Copy to populate the Window rather than have it pre-populated?
Yes. Or was that rhetorical?

- Execute Job window - Tooltip for Copy is "Start the Laser Job" rather than an explanation of Copy. Ditto for later buttons.
Ok. Done.

- Cursor shape needs to change when MK is doing something like Execute Job / Validate.
Changing the cursor has had a lot of problem at the OS level. (Maybe Review)

- Help/About - needs a margin. Text needs to flow i.e. between "continuing" and "participation". Text after "This is the brainchild of" cannot be seen. "Thanks" should be a header plus "Big thank yous to:" and then a bullet list of people with explanations. You might want to separate out significant contributors into a Contributions thank you list.
Ok. Done.

- Can we make windows into panes i.e. so we can dock e.g. the Notes window?
Some of them yes. Other might be best subdivided in several panes.

- Can we make buttons for e.g. Notes toggles so that you can click the button again to close the window?
Sure. Added `window toggle` to permit just that. Only applied it to Notes.

- Can we have ticks in Tools menu against open Windows / Pains like in the View menu?
I'm not sure I understand this.

- Console Help needs to be split as previously discussed.
#430

- Help/Webpage should be Help/Github. Also add Help//Releases to point to the GH Release page. Plus Help/Facebook and Help/IRC and Help/Report bug. Plus separator before Help/About.
Ok. Done.

- File/New should be File/Clear (IMO).
New is common, clear is not. While new clears everything. It wouldn't fit with the regular control+N sorts of things etc.

- Order of Tools menu should match button bar. Names should match button bar (JobSpooler vs, Spooler)
Ok. Done.

- Camera window - white drag circles VERY laggy indeed (on an I7) - very noticeable considering how fast the rest of MK is.
This may be due to using an older version of the Scene routine. I switched it for Simulate and used it for main. It might be over-redrawing on each mouse move, rather than permitting frame skipping. Also those handles shouldn't be white. (fixed, somebody on Facebook mentioned this).

- Settings window needs indicator if stuff is off the bottom of the window (like the Disable Tooltips option). Scroll bars? Or perhaps make it fixed size so everything shows.
Maybe. 

- Also check boxes need to start a few px lower.
Ok. Done.

- File/Recent/Clear Recent needs a separator before it.
Ok. Done.

- Can we move Tools windows into a submenu Tools/Windows.
Repeat of 2. But, I didn't do it then so I guess it feels important to you.
Ok. Done.

---

The following were requested but would take more than like 5 minutes each.

- [ ] Paths etc. have the friendly Inkscape names, but groups don't.
- [ ] Need a way to remove the camera image from background.

tatarize, 2021-05-28:
* Adjust horizontal window size to the right amount and the Windows button groups minimises to just a drop down arrow. Click on the arrow and MK crashes.

Cannot reproduce.

* SVG file with image without a user friendly name in a group with a user friendly name. Group friendly name was used instead of imagexxxxxx.

Well, yeah, svg files cascade their info. It might be stored.

* Execute Job / Blob with no laser connected gave a crash. Details sent 22:50 BST 27/5.

I will check.

tatarize, 2021-05-28:
```
meerk40t-0.7.0-beta26/meerk40t/core/planner.py", line 434, in plan
    plan[i-1].extend(plan[i])
TypeError: 'NoneType' object is not iterable
MeerK40t crash log. Version: 0.7.0 Beta-26 on darwin
```

Found and fixed.

```
MeerK40t crash log. Version: 0.7.0 Beta-26 on win32
Traceback (most recent call last):
  File "wx\core.py", line 3407, in <lambda>
  File "meerk40t\gui\widget.py", line 311, in refresh_scene
  File "meerk40t\gui\widget.py", line 339, in update_buffer_ui_thread
  File "meerk40t\gui\widget.py", line 374, in draw
  File "meerk40t\gui\widget.py", line 557, in draw
  File "meerk40t\gui\widget.py", line 557, in draw
  File "meerk40t\gui\widget.py", line 554, in draw
  File "meerk40t\gui\simulation.py", line 361, in process_draw
  File "meerk40t\gui\laserrender.py", line 208, in draw_cutcode
AttributeError: 'NoneType' object has no attribute 'MoveToPoint'
```
Found and fixed.
x3
```
MeerK40t crash log. Version: 0.7.0 Beta-26 on win32
Traceback (most recent call last):
  File "wx\core.py", line 3407, in <lambda>
  File "meerk40t\gui\widget.py", line 311, in refresh_scene
  File "meerk40t\gui\widget.py", line 339, in update_buffer_ui_thread
  File "meerk40t\gui\widget.py", line 374, in draw
  File "meerk40t\gui\widget.py", line 557, in draw
  File "meerk40t\gui\widget.py", line 557, in draw
  File "meerk40t\gui\widget.py", line 554, in draw
  File "meerk40t\gui\simulation.py", line 384, in process_draw
IndexError: list index out of range
```
Tiger found this one, too. Fixed.

----

x2
```
MeerK40t crash log. Version: 0.7.0 Beta-26 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1444, in run
  File "camera\camera.py", line 440, in threaded_image_fetcher
cv2.error: Unknown C++ exception from OpenCV code
```
Unknown Crash Bug. https://github.com/meerk40t/meerk40t-camera/blob/465d3bf4e3ba3c7e07e7ca7b83a353ebc7b2d435/camera/camera.py#L440 In capture.retrieve(). Likely best to catch it.

```
MeerK40t crash log. Version: 0.7.0 Beta-26 on win32
Traceback (most recent call last):
  File "meerk40t\gui\jobpreview.py", line 549, in on_button_start
  File "meerk40t\kernel.py", line 138, in __call__
  File "meerk40t\kernel.py", line 1738, in console
  File "meerk40t\kernel.py", line 1824, in _console_parse
  File "meerk40t\kernel.py", line 1193, in inner
  File "meerk40t\core\planner.py", line 414, in plan
  File "meerk40t\core\planner.py", line 564, in execute
  File "meerk40t\core\planner.py", line 747, in actualize
  File "meerk40t\core\planner.py", line 858, in make_actual
TypeError: 'NoneType' object is not subscriptable
```
Fixed from Log.

```
MeerK40t crash log. Version: 0.7.0 Beta-26 on win32
Traceback (most recent call last):
  File "meerk40t\gui\jobpreview.py", line 551, in on_button_start
  File "meerk40t\kernel.py", line 138, in __call__
  File "meerk40t\kernel.py", line 1738, in console
  File "meerk40t\kernel.py", line 1824, in _console_parse
  File "meerk40t\kernel.py", line 1193, in inner
  File "meerk40t\core\planner.py", line 434, in plan
TypeError: 'NoneType' object is not iterable
```
Fixed In Sneaky Extra Release.

tatarize, 2021-05-28:
* Object selection in the scene is not reflected in the Tree's current selection.

Cannot duplicate. It's light since the tree is not in focus but it highlighted.

* Clicking on an object in the scene does not allow individual selection.

This is by design it was annoying selecting one single thing in the scene.

* Drag selecting bottom up does not select objects who's corner point is not in the selection area. For example; trying to select a rectangle from a selection in its middle section and not across a corner point.

This is true, but I'd some verification this is how this should work. Dragging left, doesn't select either but most things don't seem to cover fully contained. Also you might not be able to select something inside something else with that. EG. Draw a rectangle, then a rectangle inside the rectangle. You would not be able to select the inner rectangle since the outer one would always match too.

Sophist-UK, 2021-05-28:
# UI
1. Raster operations missing from context menu on tree or layout panes.
2. Can we open raster wizard with the resample step activated? I think people want to review it from the top, and IME the resample step is the one I normally want to change.
3. Can we have a default resample level of 2 rather than 3 - a resample of 3 is effectively a PPI of 333.33 and that is too big a reduction in my experience. This may be just me though and it is a significant change from 0.6.
4. Can we have 2 buttons in the Raster Wizard to save preferences for that script and reset?
5. Raster wizard - when you change the resample level, can we please automatically adjust zoom and position of the image so that it stays the same?
6. Layers - when we have load in reverse order set as a preference, can we also reverse the order that layers are displayed (the display should stay the same regardless of which order they are loaded i.e. when you have white backgrounds to images that hide other paths, they should stay the same regardless of this setting.
7. Right click on Operation and menu has both Passes and Add Passes. I believe Add Passes should be dropped.
8. Simulate window takes a while to open - UI needs some indicator to show we should wait,
9. When I played the simulation, it was a bit jerky and when I tried to change the playback speed and / or pause it, it hung. I could move the Simulate window, but rest of UI was unresponsive. Eventually came back to life (having waited and switched to GH and typed the preceding parts of this paragraph.
10. Simulation suggest that optimisation optimises across Operations i.e. with a raster with 6 separate images, and a cut with 12 paths, it does one area first, a cut, a raster, a cut, then moves to the next area. I think this is probably good, but it needs an option to turn it off and do operations individually.

Sophist-UK, 2021-05-28:
# Future UI improvements
1. Undo / Redo
2. Save needed indicator (any changes after last save)
3. Autosave (to a different file)

Sophist-UK, 2021-05-28:
Wow - amazing responsiveness.

>> Panes do not yet have titles
>   You haven't reset your panes. Your view is just from a version of MeerK40t before those were installed.

This is not good - there is nothing to say you need to reset panes to get more functionality. Can we detect e.g. panes are not showing titles and fix it without a reset, or do an automatic reset when user runs a higher version for the first time?

> * Background colour for selected objects is garish yellow - can we have something a bit tamer like light grey?
>   Done. Though now the angry fruit salad Magenta doesn't look right. [...]

Ah - I forgot about Dark Mode - (though I run Dark mode on my PC and I have never had Dark Mode in MK 0.7 and nor can I find a setting for Dark Mode in MK.

> * When I click Execute Job, why do I have to click Copy to populate the Window rather than have it pre-populated?
>   Yes. Or was that rhetorical?

Not rhetorical. Execute Job requires way, way, way, way too many clicks on the Copy button - is there any reason why we cannot have 1 click to do everything or 2 clicks - 1 to prepare everything and 1 to send to the laser?

> * Execute Job window - Tooltip for Copy is "Start the Laser Job" rather than an explanation of Copy. Ditto for later buttons.
>   Ok. Done.

> * Cursor shape needs to change when MK is doing something like Execute Job / Validate.
>   Changing the cursor has had a lot of problem at the OS level. (Maybe Review)

I am really not bothered what the indicator is, but if the user has to wait more than (say) 0.5s, there needs to be some sort of UI indicator that shows that MK is working and not missed the click or noticed it and hung.

> * Can we make windows into panes i.e. so we can dock e.g. the Notes window?
>   Some of them yes. Other might be best subdivided in several panes.

Notes, Spooler, Console are the ones that need to be panes I think (because they are the ones you might want to keep permanently open). Rest left as Windows.

> * Can we make buttons for e.g. Notes toggles so that you can click the button again to close the window?
>   Sure. Added `window toggle` to permit just that. Only applied it to Notes.

That was an example. All Windows that are windows and not panes (i.e. all the Settings stuff) should work this way.

> * Can we have ticks in Tools menu against open Windows / Pains like in the View menu?
>   I'm not sure I understand this.

UI consistency - the View menu has ticks on  the things that are selected. I am suggesting in the Tools menu there are tick marks against the Windows that are open to be consistent with the View menu and with the button bat being toggle buttons for Windows.

> * File/New should be File/Clear (IMO).
>   New is common, clear is not. While new clears everything. It wouldn't fit with the regular control+N sorts of things etc.

Yes - I wasn't sure about this myself. File Open does not work like other tools - because it opens a new file in the same project. So the issue is that we do not distinguish between Open Project and Add File - Open Project should clear the old project, Add file should keep the old project and add an extra file.

> The following were requested but would take more than like 5 minutes each.
> 
> * [ ]  Paths etc. have the friendly Inkscape names, but groups don't.
> * [ ]  Need a way to remove the camera image from background.

Camera background image disappeared after closing and opening MK. Not sure if this is what is supposed to happen. (I was using my laptop webcam rather than a camera in the K40.) 

Not having a camera in the K40 I can only guess how workflow is supposed to work:
1. You calibrate the camera.
2. You put the material on the bed and take a snapshot in MK - then you can see the burn design over the material.
3. You move the burn design where you want it in MK.
4. You execute the burn.

In which case, background disappearing between MK executions is correct. But we do need a way to remove it manually without closing and opening MK.

Sophist-UK, 2021-05-28:
# Bugs
1. Crash on emergency stop from pane.
2. Home does not work immediately after emergency stop - even if you close and reopen MK. Emergency stop leaves K40 in odd state. New load of MK and reconnect does not reset the K40 so it works. Should we issue a home on a USB connect in order to sync (and so user knows that commands are being sent)?

tatarize, 2021-05-28:
1. Raster operations missing from context menu on tree or layout panes.

I do not understand.

2. Can we open raster wizard with the resample step activated? I think people want to review it from the top, and IME the resample step is the one I normally want to change.

Maybe. It does tend to be the most important.

3. Can we have a default resample level of 2 rather than 3 - a resample of 3 is effectively a PPI of 333.33 and that is too big a reduction in my experience. This may be just me though and it is a significant change from 0.6.

It's actually set in the script. 

4. Can we have 2 buttons in the Raster Wizard to save preferences for that script and reset?

This is a somewhat reasonable idea but the scripts are predefined and really you could have any items in any order and you aren't given that option. A fully modifiable Raster Wizard would be kinda needed for that and mostly the idea was to just add scripts as people either told me about them or figured out some good set of commands.

6. Layers - when we have load in reverse order set as a preference, can we also reverse the order that layers are displayed (the display should stay the same regardless of which order they are loaded i.e. when you have white backgrounds to images that hide other paths, they should stay the same regardless of this setting.

No. You can have a different option to view them in a reversed order. But, after the data was loaded everything's done. I'm not keeping around flags as to how that data was loaded.

7. Right click on Operation and menu has both Passes and Add Passes. I believe Add Passes should be dropped.

It also has duplicate which is also fairly similar.  But these all do different things.

8. Simulate window takes a while to open - UI needs some indicator to show we should wait,

Ok. Done.

9. When I played the simulation, it was a bit jerky and when I tried to change the playback speed and / or pause it, it hung. I could move the Simulate window, but rest of UI was unresponsive. Eventually came back to life (having waited and switched to GH and typed the preceding parts of this paragraph.

This might be greatly improved. I hit a couple bugs and rewrote it a bit and it turned out much better. In wxPython 4.0.7 which is supported due to legacy issues with a weird Locale bug it could even fully stop forever, so it had to be worked around.

10. Simulation suggest that optimisation optimises across Operations i.e. with a raster with 6 separate images, and a cut with 12 paths, it does one area first, a cut, a raster, a cut, then moves to the next area. I think this is probably good, but it needs an option to turn it off and do operations individually.

This is probably accurate. It needs a number of toggles to disable things and coax it into running the correct ways.

11. Reset pane - we need to set all panes so that when they are undocked they fit the contents reasonably closely. Can we put the Move on the bottom bar after home?

Move could be moved to whatever position. The current idea is to have you arrange the UI however you think best then we go ahead and hardcode that as the perspective that everybody gets the first time they load up. It's either that or they will get *every* single pane when they start the first time. Which might be rigged into looking fairly okay, but isn't optimal. 

---

- [ ] Raster wizard - when you change the resample level, can we please automatically adjust zoom and position of the image so that it stays the same?

tatarize, 2021-05-28:
1. Crash on emergency stop from pane.

Doesn't matter where you set it from if the crash is not multiple of something something -3, -4 or something, it's a bug I want to kill it. I cannot reproduce it. Weirdly you can't reproduce it either. It's a sync error, the hardest of all bugs to kill.

2. Home does not work immediately after emergency stop - even if you close and reopen MK. Emergency stop leaves K40 in odd state. New load of MK and reconnect does not reset the K40 so it works. Should we issue a home on a USB connect in order to sync (and so user knows that commands are being sent)?

Emergency state could well leave K40 in an odd state. It's buffering clearly it doesn't necessarily kill it all the way. This is a sort of odd protocol thing. Issuing a home might not do it, it might take a couple. I'm not sure the reason it finally clears out. But, I've done two homes or whatever until it responds.

tatarize, 2021-05-28:
Dark Mode is not an MK option, it's a special option of wxPython that can't be accessed by MSW. A lot of the other people use it, so it has some support. But, there's no way to provide it to MSW without code to recolor every wx.Control everywhere.

tatarize, 2021-05-28:
```
MeerK40t crash log. Version: 0.7.0 Beta-26 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1444, in run
  File "meerk40t\gui\rasterwizard.py", line 192, in wiz_img
  File "meerk40t\gui\rasterwizard.py", line 406, in focus_viewport_scene
TypeError: 'NoneType' object is not subscriptable
```

```MeerK40t crash log. Version: 0.7.0 RC-2 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2438, in on_click_new
  File "meerk40t\core\elements.py", line 4056, in clear_all
  File "meerk40t\core\elements.py", line 4046, in clear_elements
  File "meerk40t\core\elements.py", line 616, in remove_all_children
  File "meerk40t\core\elements.py", line 616, in remove_all_children
  File "meerk40t\core\elements.py", line 617, in remove_all_children
  File "meerk40t\core\elements.py", line 604, in remove_node
  File "meerk40t\core\elements.py", line 602, in remove_node
  File "meerk40t\core\elements.py", line 651, in notify_destroyed
AttributeError: 'Path' object has no attribute 'node'
```

...and 2x more stop-while running crashes.

Sophist-UK, 2021-05-28:
> 1. Raster operations missing from context menu on tree or layout panes.
> 
> I do not understand.

In 0.6 if you select an image and right click on it, you get two context menu items with submenus -  Raster Wizard and Applyh Raster Script. These do not appear in 0.7

> 1. Can we open raster wizard with the resample step activated? I think people want to review it from the top, and IME the resample step is the one I normally want to change.
> 
> Maybe. It does tend to be the most important.

Currently opens with the Dither options shown.

> 1. Can we have a default resample level of 2 rather than 3 - a resample of 3 is effectively a PPI of 333.33 and that is too big a reduction in my experience. This may be just me though and it is a significant change from 0.6.
> 
> It's actually set in the script.

Not sure what you mean. The script uses fixed values, the raster wizard uses the same values as defaults.
 
> 1. Can we have 2 buttons in the Raster Wizard to save preferences for that script and reset?
> 
> This is a somewhat reasonable idea but the scripts are predefined and really you could have any items in any order and you aren't given that option. A fully modifiable Raster Wizard would be kinda needed for that and mostly the idea was to just add scripts as people either told me about them or figured out some good set of commands.

I understand how we got here - these are "recipes" other people have recommended on t'internet and you have replicated those recipes in MK. And I think that makes sense for the Apply Raster Recipe.

But yes - in the future perhaps we can have a Raster Cookbook which contains by default a set of Recipes (the current set), but where you can add your own recipes.

> 1. Layers - when we have load in reverse order set as a preference, can we also reverse the order that layers are displayed (the display should stay the same regardless of which order they are loaded i.e. when you have white backgrounds to images that hide other paths, they should stay the same regardless of this setting.
> 
> No. You can have a different option to view them in a reversed order. But, after the data was loaded everything's done. I'm not keeping around flags as to how that data was loaded.

How about always loading them in the order in the file (i.e. under entities) so they always display in the correct way, but the setting reverses how they are added to operations (whenever they are added i.e. loading the file, regenerating the operations).

> 1. Right click on Operation and menu has both Passes and Add Passes. I believe Add Passes should be dropped.
> 
> It also has duplicate which is also fairly similar. But these all do different things.

Well - sort of. Duplicate duplicates the Operation. Add passes duplicates the objects in the same operation. Passes sets a parameter for burn time. And I don't think we need 3 different ways to do it - and it will confuse newbies.
> 
> 1. Simulate window takes a while to open - UI needs some indicator to show we should wait,
> 
> Ok. Done.
> 
> 1. When I played the simulation, it was a bit jerky and when I tried to change the playback speed and / or pause it, it hung. I could move the Simulate window, but rest of UI was unresponsive. Eventually came back to life (having waited and switched to GH and typed the preceding parts of this paragraph.
> 
> This might be greatly improved. I hit a couple bugs and rewrote it a bit and it turned out much better. In wxPython 4.0.7 which is supported due to legacy issues with a weird Locale bug it could even fully stop forever, so it had to be worked around.
> 
> 1. Simulation suggest that optimisation optimises across Operations i.e. with a raster with 6 separate images, and a cut with 12 paths, it does one area first, a cut, a raster, a cut, then moves to the next area. I think this is probably good, but it needs an option to turn it off and do operations individually.
> 
> This is probably accurate. It needs a number of toggles to disable things and coax it into running the correct ways.
> 
> 1. Reset pane - we need to set all panes so that when they are undocked they fit the contents reasonably closely. Can we put the Move on the bottom bar after home?
> 
> Move could be moved to whatever position. The current idea is to have you arrange the UI however you think best then we go ahead and hardcode that as the perspective that everybody gets the first time they load up. It's either that or they will get _every_ single pane when they start the first time. Which might be rigged into looking fairly okay, but isn't optimal.

You have to have some layout as default for new users and Reset - and to avoid confusion I think these should be the same. But the Move pane would fit nicely on the bottom and is the sort of thing you might want to keep open. The Navigate-Jog is similar but too big to go on the bottom pane.

NEW: Panes in the same area are equally sized - so if I put Navigate-Jog on the top area with the button bar, they each get 50% - and that looks weird. Can we give them a % propotional to the size of the contents?

> * [ ]  Raster wizard - when you change the resample level, can we please automatically adjust zoom and position of the image so that it stays the same?

Sophist-UK, 2021-05-28:
> Dark Mode is not an MK option, it's a special option of wxPython that can't be accessed by MSW. A lot of the other people use it, so it has some support. But, there's no way to provide it to MSW without code to recolor every wx.Control everywhere.

Confused. You mentioned that light grey might be an issue with Dark Mode. I am not advocating for Dark Mode as any sort of priority esp. since I do understand just how difficulty it is to get right.

Sophist-UK, 2021-05-28:
> 1. Crash on emergency stop from pane.
> 
> Doesn't matter where you set it from if the crash is not multiple of something something -3, -4 or something, it's a bug I want to kill it. I cannot reproduce it. Weirdly you can't reproduce it either. It's a sync error, the hardest of all bugs to kill.
> 
> 1. Home does not work immediately after emergency stop - even if you close and reopen MK. Emergency stop leaves K40 in odd state. New load of MK and reconnect does not reset the K40 so it works. Should we issue a home on a USB connect in order to sync (and so user knows that commands are being sent)?
> 
> Emergency state could well leave K40 in an odd state. It's buffering clearly it doesn't necessarily kill it all the way. This is a sort of odd protocol thing. Issuing a home might not do it, it might take a couple. I'm not sure the reason it finally clears out. But, I've done two homes or whatever until it responds.

This works differently in 0.6 - using the button in the Controller it just stops. And Home works first time. 

In 0.7 different button (may work differently under the covers) leaves K40 in a state where it takes 2x Home to get it to work. My jobs always start with a Home, but only one, so starting a job after an emergency stop without a manual Home makes it burn in the wrong place.

tatarize, 2021-05-28:
Certainly worth checking out...

Sophist-UK, 2021-05-28:
> Certainly worth checking out...

I get the crash pretty much every time I click Emergency Stop. How can I help you diagnose this (without putting a dev environment on my production PC I use for lasering? Can you provide me an executable that has some logging / diagnostic code?

Sophist-UK, 2021-05-28:
# UI
1. Menu items do not have normal letters as shortcuts. Alt, F, N is normally File/New. Alt, F, A is normally File Save As. You normally specify the letter by preceding it with an ampersand.

Sophist-UK, 2021-05-28:
Here are further recommendations for UI esp. missing tooltips or tweaks for existing ones based on beta 27:

### Settings
* General 1: Make Settings window fixed size i.e. user cannot change size or maximise.
* General 2: Checkbox strings might look better as Sentence case, not Title Case.

* Save Window Positions – Open Windows at the same place they were last closed
* Print Shutdown – Close MeerK40t when the K40 job completes
* SVG Uniform Save - ???? No idea what this does 	
* Image DPI Scaling – Set: Use the DPI setting saved in the image to scale the image to the correct size. Unset: Use the image as if it were 1000 pixels per inch.
* DXF Centering – Fit (scale down if necessary) and centre a DXF file within the bed.
* Show Negative Guide – Extend the Guide rulers with negative values to assist lining up objects partially outside the left/top of the bed.
* Launch Spooler JobStart – Open the Spooler window automatically when you Execute a Job.
(P.S. Should we call this Open Spooler on Execute?)
* MouseWheel Pan – Default normal MouseWheel is Zoom (and horizontal pan with Shift key) – this changes normal MouseWheel to vertical pan, and zoom with Ctrl key. 
* Invert MouseWheel Pan – Reverses the direction of the MouseWheel for horizontal & vertical pan.
* Invert MouseWheel Zoom – Reverses the direction of the MouseWheen for zoom.
* Default Operation Empty - ???? No idea what this does
* Reverse SVG Element Load – The Inkscape Object List is shown in reverse order. This setting allows you to burn objects in the same order as shown in the Inkscape Object List.
* Disable Tool Tips – If you do not want to see tooltips like this one, check this box. Particularly useful if you have a touch screen.

* SVG Pixel Per Inch – ??? I have no idea what this does.
(P.S. Should be Pixels not Pixel)

### Execute Job
* Automatic/Before: Home, Physical Home – should they be mutually exclusive as doing both does not seem to make sense

* Automatic/After: As above plus Return to Origin should be mutually exclusive, and also:
  * Dropdown list order: Home, Physical Home, Return to Origin, Beep, Interrupt, Unlock
  * Execution order: Beep, Home, Physical Home, Origin, Unlock - Interrupt was not added to the execution list.
Order should be the same in dropdown and execution. IMO Beep should come last in both lists.

* *Tools menu is non-functional.*

* Rapid Moves between Objects – Default move speed is xxxxx. This setting moves at the faster of this speed and the current vector speed.
* Reduce Travel Time – Resequences the objects in order to minimise the time spent travelling
* Cut Inner First – Ensure that inside burns are done before an outside cut which might result in the cut piece dropping out of the material or moving.
(Note: If Optimisations are done across all Operations rather than doing operations sequentially, then this should be “Burn Inner first” as ALL operations (regardless of operation type inc. rasters and images and regardless of operation sequence) inside a closed path that is in a Cut operation need to be done before the cut.)

* *Reduce Direction Changes / Remove Overlap Cuts – these should be hidden completely rather than greyed out (because greyed out makes it look unfinished).*

### Notes
Can we make the edit box rich text so we can have Italics, Bold etc.?

### Navigate
* Translate buttons – Make the bounding box bigger than original rather than moving the bounding box – so move right moves the right side, but not the left side etc. Ditto for scale and rotate buttons.
* Reset Matrix Button – HOLY SHIT - the point of this button is what? (aside from totally screwing up the layout) – can we replace this with a reload from file button?
* Not sure what the arrow buttons in the middle block do – they seem to operate like the similar buttons in the right block, but are different and have no tooltips. Can we replace these with centres – left, right, top, bottom centres of selected objects?
* Short pulse – can we have that as a pane also (like the Move widget)?

ampostata, 2021-05-28:
-Burning order wrong: I have to operations, and in the simulation it is done as first element first, then second. But in the real burn, the last element is done first.

-Document size. My documents are almost as big as the bed dimensions, and the text showing the size is cut/painted outside, so I can not see it. Can we have the document size somewhere else (like in the botton little bar, where status is).

-Big icons. Can we have smaller icons for computers without a big screen? In my laptop they are almost 1/4 of the full screen, with little space for the rest. Even the windows (navigation one, for examle) is so big as almost 1/4 of my entire screen. I'm at 1366x768 pixels.

-Center/reset bed. Can we have a button to reset zoom/position of the bed, to fit the window?

-Center document: can we have a button to center the document in the bed? (If my bed is 300x200, and the document is 200x100, to have it centered, with 50 at each border)

-Ghost lines. After a clear all, blue ghost lines are still painted in the bed (not in the burn!).

-Clear all in elements: after choosing clear all, the element is cleared but still shown under elements tree.

inspectionsbybob, 2021-05-29:
Sorry, I put some Comments in another issue... I will put the rest of them Here.

- (Duplicate) Perspective Alignment dots cover targets, see issue for pictures
- (Duplicate) Fisheye Correction doesn't seem to do anything anymore...
- Too Many Keypresses to "Start Job"
        Preprocess Operations or  Copy ?? , Blob, Optimize, (BIG Pause) Spool
        (After Optimize Preview Jobs disappears for a few seconds, and the Spool show up, turned off Optimize travel and pause went away)
- When you load an SVG, it no longer shows in Operations at all
   (found it, under the (+) after a few seconds, default should be to show it pls, hide as option)
- LIKE the STOP button !
- Like the SAVE OPERATIONS
-      (But Need way to NAME the saves so they make sense beyond #1, #2, etc...)
- How do you invoke the tools?  Love the idea of Align or Space
- How about a "Clear LaserPath" to clean up a screen from multiple jobs
-

tatarize, 2021-05-29:
>(Duplicate) Perspective Alignment dots cover targets, see issue for pictures

RC2 fixed.

> Too Many Keypresses to "Start Job"

I might get this down by 1, getting rid of copy.

>Preprocess Operations or Copy ?? , Blob, Optimize, (BIG Pause) Spool

Weak names to be sure. Optimize took a bit.

> (After Optimize Preview Jobs disappears for a few seconds, and the Spool show up, turned off Optimize travel and pause went away)

Because it is no longer optimizing there.

 >When you load an SVG, it no longer shows in Operations at all
> (found it, under the (+) after a few seconds, default should be to show it pls, hide as option)

It fails to show right away. That should be addressed.

> (But Need way to NAME the saves so they make sense beyond #1, #2, etc...)

You also could name it after a material. But also proper names can be provided in console. Not that that is much of a solution.

> How do you invoke the tools? Love the idea of Align or Space

I hadn't finished them, so they were disabled in RC2.

>How about a "Clear LaserPath" to clean up a screen from multiple jobs

I think it might do this if you zoom in a little bit.

---

- [ ] Fisheye Correction doesn't seem to do anything anymore...
- [ ] New tree children do not show right away.

tatarize, 2021-05-29:
> In 0.6 if you select an image and right click on it, you get two context menu items with submenus - Raster Wizard and Applyh Raster Script. These do not appear in 0.7

They do. But, only in the first time you right click. Bug. Fixed.

> Currently opens with the Dither options shown.

Ok. Done.

> I understand how we got here - these are "recipes" other people have recommended on t'internet and you have replicated those recipes in MK. And I think that makes sense for the Apply Raster Recipe.

Mostly stressing that the set of general image procedures and the order they occur in is not copyrightable content (you may perhaps patent it). But, also, it would be really hard to turn that into a full on properly configurable script maker. Not too impossible to be sure. And if you serialized the data it could be absolutely spat out as a bit of text and passed around, and new things recommended. But, that's outside the scope of a feature frozen new version.

> How about always loading them in the order in the file (i.e. under entities) so they always display in the correct way, but the setting reverses how they are added to operations (whenever they are added i.e. loading the file, regenerating the operations).

Like classify backwards? That would be downright trivial to add, but as an option it might be gawdy. I could give you a checkbox in settings though. "Perform Classification B2T"
 
> Yes - in 0.7 we have Operation Passes, Duplicate Operation, Passes/Add Passes, Duplicate/Duplicate elements. Duplicate Operation and Operation Passes do unique things. Add Passes / Duplicate (Elements) do pretty much the same thing - can we lose Passes/Add Passes and rename Duplicate to Duplicate Elements?

- [ ] Note, check on this. 

This might still be correct. I'll put it on the bottom of stuff to be checked for a bit now.

> **NEW:** Panes in the same area are equally sized - so if I put Navigate-Jog on the top area with the button bar, they each get 50% - and that looks weird. Can we give them a % propotional to the size of the contents?

Ok. Done.

tatarize, 2021-05-29:
>Similate icon needs to have thicker lines to be consistent with other icons.
 Ok. Done.

> Confused. You mentioned that light grey might be an issue with Dark Mode. I am not advocating for Dark Mode as any sort of priority esp. since I do understand just how difficulty it is to get right.

No. My point is some folks already use it because they run OSX or Pi, and are wicked clever at things. Color scheme are fine, my point was things like pure-white can cause them issues so it might be best to choose something a bit more centrist.

tatarize, 2021-05-29:
Settings

Print Shutdown – Close MeerK40t when the K40 job completes
MouseWheel Pan – Default MouseWheel is Zoom (and horizontal pan with Shift key) – this changes MouseWheel to vertical pan, and zoom with Ctrl key

> (P.S. Should be Pixels not Pixel)
Ok. Done.

>Automatic/Before: Home, Physical Home – should they be mutually exclusive as doing both does not seem to make sense

But, they can do both. It's just dumb. I don't stop dumb harmless things.

> Automatic/After: As above plus Return to Origin should be mutually exclusive, and also:

Ditto.

>Dropdown list order: Home, Physical Home, Return to Origin, Beep, Interrupt, Unlock
>Execution order: Beep, Home, Physical Home, Origin, Unlock - Interrupt was not added to the execution list.

Ok. Done.

>Order should be the same in dropdown and execution. IMO Beep should come last in both lists.

Ok. Done.

> Tools menu is non-functional.

Bzzzt! Wrong. It works fine... it's just empty and doesn't need to be there. In, 0.6.xx this has step repeat.

>Rapid Moves between Objects – Default move speed is xxxxx. This setting moves at the faster of this speed and the current vector speed.

We don't know how fast rapid move actually goes. Or where the cutoff should be. And no, it will actually move slower if that's what your speed is slow. It must because it could be a rotary job or something. Or somebody is dealing with very loose belts.

> Reduce Direction Changes / Remove Overlap Cuts – these should be hidden completely rather than greyed out (because greyed out makes it look unfinished).

Ok. Done.

>Can we make the edit box rich text so we can have Italics, Bold etc.?

It has to save to the notes, on the current document. I'm not actually sure what rich-text saves like. Odds are fairly good it shouldn't be too bad. not sure how to add in the bolds and italics and such though. Could also make it display urls and make those clickable but, not sure how many people use notes to make such features worthwhile.

I'll add the flag to make it rich text. Not really sure if it'll save or anything.

> Translate buttons – Make the bounding box bigger than original rather than moving the bounding box – so move right moves the right side, but not the left side etc. Ditto for scale and rotate buttons.

I don't follow. But, big GUI changes are hard.

>Reset Matrix Button – HOLY SHIT - the point of this button is what? (aside from totally screwing up the layout) – can we replace this with a reload from file button?

It resets the matrix.

>Not sure what the arrow buttons in the middle block do – they seem to operate like the similar buttons in the right block, but are different and have no tooltips. Can we replace these with centres – left, right, top, bottom centres of selected objects?

You select the object in scene and it moves the laser head and object so you can keep them together to find the right spot. It should actually get it's own pane too. It's pretty useful. Whisperer has something like this for most of their movement. But, that also moves origin too.

> Short pulse – can we have that as a pane also (like the Move widget)?

Probably. Not sure how often it'd get used. But, might as well. Added to List.

>Some are Title case, some lower case. (If we are getting to the point of listing stuff like this, we must be close!!!!)

This is actually by design.
```python
    ("en", u"English", wx.LANGUAGE_ENGLISH),
    ("it", u"italiano", wx.LANGUAGE_ITALIAN),
    ("fr", u"français", wx.LANGUAGE_FRENCH),
    ("de", u"Deutsch", wx.LANGUAGE_GERMAN),
    ("es", u"español", wx.LANGUAGE_SPANISH),
    ("zh", u"Chinese", wx.LANGUAGE_CHINESE),
```

English requires that English be capitalized as a proper name of a language. That is not the rule in Spanish or Italian. In German it's the rule that every noun is capitalized which would include languages. Chinese should properly be 中文 however, it gets weird and wants a different encoding to do that so it's capitalized because it's just the word "Chinese" written in English. The Chinese translation is really just Google Translated since I sort of wanted this translated back into Chinese since it's running the Chinese board. And bad Chinese might be perfectly fine too, though I wouldn't turn down translation help for that.

> Position tab on Button Bar

This got its own pane. I tried to remove it but it was a jerk.

Sophist-UK, 2021-05-29:
> > Automatic/Before: Home, Physical Home – should they be mutually exclusive as doing both does not seem to make sense
> 
> But, they can do both. It's just dumb. I don't stop dumb harmless things.

Unfortunately not everyone is as sensible as us. My maxim is to do as much as possible to prevent dumb users doing dumb things even if they are harmless.

> > Tools menu is non-functional.
> 
> Bzzzt! Wrong. It works fine... it's just empty and doesn't need to be there. In, 0.6.xx this has step repeat.

How do I do Step Repeat in 0.7 then?

> We don't know how fast rapid move actually goes. Or where the cutoff should be. And no, it will actually move slower if that's what your speed is slow. It must because it could be a rotary job or [...]

Travel at Cut Speed - `Travel between objects (laser off) at the Cut/Engrave speed rather than at the K40 default travel speed`

> It has to save to the notes, on the current document. I'm not actually sure what rich-text saves like. Odds are fairly good it shouldn't be too bad. not sure how to add in the bolds and italics and [...]

I'll test it soon. We will probably eventually want buttons for Bold, Italic etc.

> > Translate buttons – Make the bounding box bigger than original rather than moving the bounding box – so move right moves the right side, but not the left side etc. Ditto for scale and rotate buttons.
> 
> I don't follow. But, big GUI changes are hard.

When you select an object, it displays the bounding box. When do anything to the object (enlarge, translate) it only partially adjusts the bounding box by moving sides which enlarge the box but it doesn't move sides that would reduce the size of the bounding box.

> > Reset Matrix Button – HOLY SHIT - the point of this button is what? (aside from totally screwing up the layout) – can we replace this with a reload from file button?
> 
> It resets the matrix.

Does anyone ever want to do this?

Whereas reload object(s) from file is probably much more useful?

> You select the object in scene and it moves the laser head and object so you can keep them together to find the right spot. It should actually get it's own pane too. It's pretty useful. Whisperer [...]

Don't get me started about Whisperers movement on the bed - I switched to MK because of this.

I never use it, but I accept that it is useful. But being able to move to centres of the bounding box sides would also be useful for when e.g. you have rounded corners.

> This is actually by design.
>
> ```python
>     ("en", u"English", wx.LANGUAGE_ENGLISH),
>     ("it", u"italiano", wx.LANGUAGE_ITALIAN),
>     ("fr", u"français", wx.LANGUAGE_FRENCH),
>     ("de", u"Deutsch", [...]

OK

tatarize, 2021-05-29:
@ampostata, first off thank you for taking the time to help.

>-Burning order wrong: I have two operations, and in the simulation it is done as first element first, then second. But in the real burn, the last element is done first.

This was a serious flaw in RC1.

> -Document size. My documents are almost as big as the bed dimensions, and the text showing the size is cut/painted outside, so I can not see it. Can we have the document size somewhere else (like in the botton little bar, where status is).

I added Position as a frame due to this comment, and sort of needing to otherwise. You'd have to reset your panes to see it if your UI is already set but it should tell you what your document size is and let you change those.

>-Big icons. Can we have smaller icons for computers without a big screen? In my laptop they are almost 1/4 of the full screen, with little space for the rest. Even the windows (navigation one, for examle) is so big as almost 1/4 of my entire screen. I'm at 1366x768 pixels.

The RibbonBar is a bit hard to mess with. I tried making it smaller but it's doesn't seem to allow that. It's art is pretty set the size it is. You can modernly close it down. The little //\\ thing in the corner will give you the bar and let you drag it around and close it if needed. All the commands on the main bar are in the menu bar under tools->window and the things like emergency stop can have their own buttons floating or otherwise.

>-Center/reset bed. Can we have a button to reset zoom/position of the bed, to fit the window?

View, Zoom to Scale. Will do that.

>-Center document: can we have a button to center the document in the bed? (If my bed is 300x200, and the document is 200x100, to have it centered, with 50 at each border)

I added in a console command `align bedcenter`, this also lead to me figuring out the bug that was breaking my other alignment tools, and fixing those too. Which is kind of a shame since they got ripped out in RC2, and will still be gone in RC3. I also added in all the `preserveAspectRatio` values from SVG into being align commands. So `align xmidymid` will scale your project up to the size of the bed and align it to the center. This is basically what oversized DXFs do.

But, in console `align bedcenter` will do that for you, also you can bind it to a key: `bind control+b align bedcenter` or just bind it to something in your keymap. 

> -Move document to 0,0. Can we have a button to move document to coordiates 0,0?

Just added `align topleft` though this can also be done with positions.

>-Ghost lines. After a clear all, blue ghost lines are still painted in the bed (not in the burn!).

Clearing all clears the blue lines. That or something else is here and I cannot duplicate this. wxmeerk40t.py2536 should do exactly that.

-Clear all in elements: after choosing clear all, the element is cleared but still shown under elements tree.
Ok. Done.

>-If I click in "elements", all navigation buttons are selectable. If i click in "operations", only the first cross is available. I think they have to be all of them also.

They don't. They have links to the elements. It might be possible to make you able to move these independently though and get the selected `elems` (elements in the element tree) and `opnode` (links to elements under operations). But, it may be the case that if the same item exists 2x times it will move 2x as far. I suppose I could try to work around that.

>-Simulation: a full line is instant draw, instead of drawn "at burn speed"

Taken under advisement. I didn't think of it doing such things at this speed but apparently other people did.

>-The burns are done in a very strange behaviour, look at this videos:

Corrected in RC2.

tatarize, 2021-05-29:
@Sophist-UK 

> Unfortunately not everyone is as sensible as us. My maxim is to do as much as possible to prevent dumb users doing dumb things even if they are harmless.

My maxim is to let users do everything if it's very bad to do, hide it better.

> When you select an object, it displays the bounding box. When do anything to the object (enlarge, translate) it only partially adjusts the bounding box by moving sides which enlarge the box but it doesn't move sides that would reduce the size of the bounding box.

So like making it smaller?

> Travel at Cut Speed - Travel between objects (laser off) at the Cut/Engrave speed rather than at the K40 default travel speed

Ok. Done. (With slight rewording because it's travel at rapid speed rather cut)

> Does anyone ever want to do this?

I do sometimes. Though I certainly don't use that button to do it.

> Whereas reload object(s) from file is probably much more useful?

Sure, but I dunno which object to reload since there could be a lot of files open etc.

tatarize, 2021-05-29:
Moving to releasing RC3. Still somethings in the list of tweaks (See Project 1, column).

Sophist-UK, 2021-05-29:
> > When you select an object, it displays the bounding box. When do anything to the object (enlarge, translate) it only partially adjusts the bounding box by moving sides which enlarge the box but it doesn't move sides that would reduce the size of the bounding box.
> 
> So like making it smaller?

Like keeping it an exact X/Y fit to the selection.

> > Travel at Cut Speed - Travel between objects (laser off) at the Cut/Engrave speed rather than at the K40 default travel speed
> 
> Ok. Done. (With slight rewording because it's travel at rapid speed rather cut)

So, what is the definition of "rapid speed" How does MK determine what speed to use?

> > Whereas reload object(s) from file is probably much more useful?
> 
> Sure, but I dunno which object to reload since there could be a lot of files open etc.

Selected objects. But at present we only have reload file in totality and so reload selected objects is new functionality.

tatarize, 2021-05-29:
> So, what is the definition of "rapid speed" How does MK determine what speed to use?

The default speed the card goes when you tell it to move without telling it to move at a speed. We never actually measured how fast it goes.

Sophist-UK, 2021-05-29:
Success - using MK RC3 for production burns!!!!

# Bugs
1. With optimise, passes 8: a circle in one continuous burn; an ellipse, 1st burn from bottom, 2nd burn from left, 3rd and subsequent burns continuously from top.

# Usability
1. Optimise currently creates a single spool job for everything. Not sure what happens if you don't optimise. It would be great if we could split the optimised items into separate spool jobs based on outermost outside cuts.
2. In Simulate, can we put arrows on the travel moves as you cannot tell direction otherwise.
3. Load works with friendly names, but save does not preserve these - can we use the friendly names on a save?
4. Optimise with Inside-Out - it should try to do all cuts inside the same outside before moving to other cuts to try to complete each outside in sequence. (See also 1. above.) (This may not be completely TSP optimal, though it should be close because it is still localised - but it does help if something goes wrong (not necessarily with MK but e.g. with the material or burn) to localise the issue and make it easier for the user to recover as much as possible as easily as possible.)
5. "Processing simulation"->"Preparing simulation".

tatarize, 2021-05-29:
> With optimise, passes 8: a circle in one continuous burn; an ellipse, 1st burn from bottom, 2nd burn from left, 3rd and subsequent burns continuously from top.

Yeah, if that's true then it likely violated greedy algorithm. The exception being if it happened to accidently double back on itself then it would force itself into non-optimal situations. In theory every endpoint, is coincident with 15 other points. These are basically going to be kind of random. You go with the wrong one and if you went backwards once, you force yourself to backtrack at least one more time. 

> Optimise currently creates a single spool job for everything. Not sure what happens if you don't optimise. It would be great if we could split the optimised items into separate spool jobs based on outermost outside cuts.

I could see some utility for this but that ordering is mostly highly mixed and could be basically anything after planning.

>In Simulate, can we put arrows on the travel moves as you cannot tell direction otherwise.

Ok. Done.

> Load works with friendly names, but save does not preserve these - can we use the friendly names on a save?

The main load friendly name is in the inkscape namespace. Though I do check normal "label" too so maybe.

> Optimise with Inside-Out - it should try to do all cuts inside the same outside before moving to other cuts to try to complete each outside in sequence. (See also 1. above.) (This may not be [...]

I think it might be a quirk if it goes backwards it screws up the later moves. Which is why greedy has some issues.

> "Processing simulation"->"Preparing simulation".

Ok. Done.

> Good news - based on the impressive nature of Optimise, I think that some of the options I discussed with you before can be ignored - we still need a few tweaks, for usability (not of the software but for practical burning) as above.

Yeah, but I took the shortest path here. They are greedy and brute, they could be speed up and globally better with a few things like two-opt. But, I want some version done more than I want to go about adding that.

Sophist-UK, 2021-05-29:
> > With optimise, passes 8: a circle in one continuous burn; an ellipse, 1st burn from bottom, 2nd burn from left, 3rd and subsequent burns continuously from top.
> 
> Yeah, if that's true then it likely violated greedy algorithm. 

It did. It's a bug.

> The exception being if it happened to accidently double back on itself then it would force itself into non-optimal situations. 

It didn't double back - all burns were in the same anti-clockwise direction.

> In theory every endpoint, is coincident with 15 other points. 

Some points i.e. which are tangential and continue the burn smoothly should have priority - I think this probably translates to giving priority for connecting this path end point to a path start point - true I think for normal cases, though I could explicitly create an SVG with a path that doubled back on itself (in which case it should double back on itself). 

> These are basically going to be kind of random. 

Definitely should NOT be the case for closed paths which are multi-passed. If someone has created a combined path (like the birdhouse) where some segments of a closed path are in one direction and some segments in the other, then hopefully the optimise algorithm will join them optimally by connecting the current end point to the end-point of another segment and reverse that segment to make it work.

> You go with the wrong one and if you went backwards once, you force yourself to backtrack at least one more time.

The algorithm has to avoid doing this.

> I could see some utility for this but that ordering is mostly highly mixed and could be basically anything after planning.

Use case 1: In v0.6 I use the spooler extensively to queue up cut operations, and when a cut operation breaks through I then cancel any remaining cuts. I haven't checked whether I can still do this if I turn off both Optimise and Inside-First, but I hope so - or I can do an Execute Job for all operations with one cut, and then add more cut operations afterwards.

Use case 2: Optimising mixes the Operations - so they are not spooled as separate operations. But **if** we make the planner group the optimised instructions by outermost outside paths (even ones with nothing inside), and then splits the optimised instructions into separate spool jobs by outermost path, then that would give more flexibility to the user to insert an Interrupt operation after MK burns one part of the design, or for the user to cancel specific parts of the burn (similar but different to use case 1).

In essence when MK creates a single spool job for literally everything, it becomes all or nothing with far less flexibility for controlling the burn real-time as you sometimes need to do.

> > Load works with friendly names, but save does not preserve these - can we use the friendly names on a save?
> 
> The main load friendly name is in the inkscape namespace. Though I do check normal "label" too so maybe.

I use inkscape with friendly names. They appear on paths and images, but not yet groups. But when I save and reload, the friendly names have gone. I just want MK to save SVG files using the friendly name, but it can save it as a SVG standard name rather than an Inkscape special.

> I think it might be a quirk if it goes backwards it screws up the later moves. Which is why greedy has some issues.

Not sure I understand this or that it answers the requirement. What I am saying is that if the greedy algorithm selects an outside closed path as the next nearest point, it then needs to burn the inside paths. So instead of selecting from all remaining objects,  it should restrict choosing objects to those that are "inside" including vector engraves, vector rasters, images and cuts until it has exhausted those, and then switch back to using the global list of remaining objects until it gets to another outside closed path.

I know this is **not** going to be TSP optimal, but it is still going to be pretty good and will both feel better for the user **and** actually be better for the user in terms of stopping the burn part way through (and still having usable pieces) and recoverability from issues (i.e. picking up where you left off).

> > "Processing simulation"->"Preparing simulation".
> 
> Ok. Done.

> Yeah, but I took the shortest path here. They are greedy and brute, they could be speed up and globally better with a few things like two-opt. But, I want some version done more than I want to go [...]

The results are already pretty impressive. If the objective is saving burn time, then you have probably already saved 90% of what can be saved. IMO 2-opt is far less important than fixing inside-out to cope with paths that look visually closed to the user, but inside the SVG are not closed.

tatarize, 2021-05-29:
[image]

Unknown Critical Bug. Likely in plot planner. Likely duplicated by @joerlane's project there. I'll get to it when I wake up. Running out of minor things. Might even relate to the crash on estop (that shouldn't crash) but does, so there's some bug there.

Sophist-UK, 2021-05-29:
> Unknown Critical Bug. Likely in plot planner. Likely duplicated by @joerlane's project there. I'll get to it when I wake up. Running out of minor things. Might even relate to the crash on estop (that shouldn't crash) but does, so there's some bug there.

Is this the same bug as causing this one?
[image]

And could it be associated in any way with the SVG Matrix one we discussed on IRC?

Sophist-UK, 2021-05-29:
# UI - Execute Jobs
1. Can we have a Preparing Simulation... style pop-up when You click the big button in Execute Jobs?
2. Can we add an Optimization option "Optimize Across Operations" to allow the user to decide whether to do Inside-Out and TSP separately for each Operation in the tree or (as at present) to merge all operations before optimising, (Obviously if this is too disruptive to code, it should wait until 0.7.1.)

Sophist-UK, 2021-05-29:
# PLEASE DO ALL FURTHER TESTING ON v0.7.0b27 RC3
The latest beta release is v0.7.0beta27 Release Candidate 3.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases.

inspectionsbybob, 2021-05-30:
BIG SCARY PAUSE after "Validate" before it goes to the spooler.
Strongly suggest a Countdown bar or something so users don't think it is frozen.
Still don't like BLOB" 
Depending on WHERE you invoke the job, you may see "PreProcess" Then "Validate, Blob, Optimize...."
Should be consistent....

When I had "Optimize" turned on, it seemed to go back and "touchup" the cut... Off it just made the cut as one pass, I suspect the artwork was multiple segments...   Also "fast move between cuts" should be on by default, otherwise it may travel at cut speeds, which is 10mm/sec vs 300mm/sec

tatarize, 2021-05-30:
The scary pause is optimize. If you hit Simulate it'll give you a busy info. I guess I could do the same for that step in Execute Job.

Blob is a bad term and likely should be switched. It's the point where it changes path/image filled operations into individual cuts each with their own settings object. Which is internally called cutcode. But, is actually a tree of CutObjects and CutGroup. This is then able to be effectively furthered processed by the optimization routines. A process which I feel is still a bit too complex, but could be simplified later.

Sophist-UK, 2021-05-30:
> The scary pause is optimize. If you hit Simulate it'll give you a busy info. I guess I could do the same for that step in Execute Job.

Yes.

> Blob is a bad term and likely should be switched. It's the point where it changes path/image filled operations into individual cuts each with their own settings object. Which is internally called [...]

You know my view - whilst all these different clicks may be useful to you as a developer, I cannot see how they are helpful to a user. My view is that as many of these should be combined as possible. The Copy step should be omitted and the window prepopulated. The user then needs to have the opportunity to set optimisation options. Ideally you would just get a `Plan` button after that - which would pop up a Processing... box whilst it is doing the Blob, Optimize etc. User then gets `Preview` and `Execute` buttons - or they can change the optimisation settings and get the `Plan` button back.

tatarize, 2021-05-30:
I think Validate, Blob, and Optimize , Spool are the required ones. Might be what you get currently.

There's something to merging preview and execute job, but I dunno what.

tatarize, 2021-05-30:
Your workflow there might be fairly reasonable.

ampostata, 2021-05-30:
I also don't see the sense for an average user to press the same button 4 times to burn. If they are not skipable, why then? The button should be "burn", and when you click on it the text can change automatic from validating... Blobing... Optimizing... Spooling... Burning!, with perhaps a progress bar.

Currently is just like pressing for times a button that says "I-want -to -burn -now" 😅

ampostata, 2021-05-30:
- If you move the document, "draw burn box check button" ( I don't know the word, I'm not in my computer), it is not updated. I have to save the document and reload it again. 
- The reset matrix button (the center one with a cross) just mess my proyect.

ampostata, 2021-05-30:
-the burns / optimization should be done by colors/order, or at least an option or the cut one being always the last : I have not (yet) a real bed, my mdf sheet is over pot glasses ("floating") and sometimes the pieces fall before the inner engrave is engraved.

Sophist-UK, 2021-05-30:
> * If you move the document, "draw burn box check button" ( I don't know the word, I'm not in my computer), it is not updated. I have to save the document and reload it again.
> * The reset matrix button (the center one with a cross) just mess my proyect.

Any chance of a screen shot?

Sophist-UK, 2021-05-30:
# Bug
1. Save of SVG does not preserve group structure. This is likely to be necessary for e.g. optimisation.

inspectionsbybob, 2021-05-30:
I would like to see a way to name the operations "saves".  I saw something about it, but it didn't clarify the way to name something other than the predefined "1" "2" etc...

tatarize, 2021-05-30:
@inspectionsbybob The predefined list are just names I quickly made up. They aren't hardcoded but for the menu. If you load up console and type `operation save Bob` you will save the current operations under the name Bob. You can then resave them over that name or load them up accordingly (so long as there is at least one operation in the group). The names are entirely dynamic. But, Tiger had reasked about that feature so I wrote it up. I didn't want to prompt for a name each time so I gave a basic list of default names.

ampostata, 2021-05-30:
- Reset matrix mess my document. Notice that my SVGs are exported from LightBurn. If I load a SVG in Meerk and save it, the mess is no more. ALso you can see that the grey text with size (at right of document) is not seen.

[image]
[image]

-The "perform a simple trace of the selection" does not check moved document, it stays with the loaded one.

tatarize, 2021-05-30:
Reset matrix takes your work and destroys the matrix that is currently being used to situate the particular pieces. These apply transformations on each to scale, move, and rotate them into the correct order, this goes out the window with reset.

The save routine goes ahead and save them in reified form. It doesn't mess with the matrix and just modifies the actual shapes themselves.

inspectionsbybob, 2021-05-30:
Uhm... Ok... but the console is carry to some...
And where is that profile stored?  Current Directory?

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

Sophist-UK, 2021-05-30:
> @inspectionsbybob The predefined list are just names I quickly made up. They aren't hardcoded but for the menu. If you load up console and type `operation save Bob` you will save the current [...]

I doubt it will be that difficult to improve the UI for this in the future, but I imagine it is not a priority at present.

Sophist-UK, 2021-05-30:
> * Reset matrix mess my document. Notice that my SVGs are exported from LightBurn. If I load a SVG in Meerk and save it, the mess is no more. 

See my comments higher up about this. I cannot see a use case for this button at present, though if MK does a Reify on every element on load (resetting these matrixes but without moving anything), then I think this button would essentially put the elements back to where they were when the file was loaded - which is a pretty useful function.

ampostata, 2021-05-30:
@Sophist-UK I don't know if it is working as intended or a bug. But I know that in 0.60 when I click "reset matrix", the/my document is moved to 0,0 coordinate, and in 0.70, it just mess it

Here you have a "messable" file if you want to test it.

[www.frikland.com/sb/01.svg](url)

- If you move the document, "perform a simple trace of the selection" it is not updated. I have to save the document and reload it again to "update" starting position of the laser.
- can we have an option to burn the layers in order with optimization (per layer)?

Sophist-UK, 2021-05-30:
I think we need to wait for the next beta and see what the button now does.

ampostata, 2021-05-30:
In "operations", if you right moue button on any layer, you can "execute job"; on the main "operations" text, there is no "execute job" to execute all the jobs :)

[105 more replies]

closed 2021-06-05 by tatarize