meerk40t/meerk40t #436: Loosing registration between power cycles (v 6.22)

inspectionsbybob, 2021-05-27:
I noticed that when I left the K40 alone (off) for a few days, when I went back, the registration was off by about 1/8" of an inch.

Not much, but enough if you are used to "hairline registration" and are putting things close to the edge to get the most out of your materials.

I had spent extra time last time getting it "spot on" so I could group things TIGHT and use every bit of my material.

When I ran today, I was "JUST" over the edge, no one will notice, but I was surprised...

Again, this is 6.22...

Sophist-UK, 2021-05-27:
Odd - I never had that. I had an X offset in my device config. And it has been consistently the same. Could it be that your bed has moved slightly?

inspectionsbybob, 2021-05-27:
nope, everything is literally glued down...
thought it was me, until it happened more than once, typically front to
back, and only by a bit...

I will re-set after this job and see how much it was off, or if corners
moved.

I have a stiff cardboard template I can drop in, line up with some
registration marks on the cartridge and find the corners really quickly.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

tatarize, 2021-05-28:
The homing sequence in MeerK40t involves sending an "IPP$" packet and is run directly on the board. Other than running your laser into the endstops manually I'm not sure there's an easy method to correct for this. Earlier in development I had written a "Home and Dot" function to see how much slop there might be for head and it was usually really close.

I am unaware of any routine that could help with this, other than manually unlocking the rail and putting it in the corner yourself. Which I would provide consistent location. But, *usually* the laser head is within a pretty tight location after homing.

tatarize, 2021-05-28:
* Running the head into the endstops might tighten those locations because I'd assume the gantry would win against the belts and motor at roughly the same physical spot regardless where it thinks it is. But, it might mess up your belt tightness after a bit. You could maybe do a Dots Operation (0.7.0 feature) if you could have some way to see the marks always or something. But, hardware issues, without a clear way of assisting in software doesn't seem possible to address...

closed 2021-05-29 by tatarize