meerk40t/meerk40t #443: Release Candidate 0.7.0: Testing... Report any issues found here.

tatarize, 2021-06-05:
# PLEASE DO ALL FURTHER TESTING ON v0.7.0b30 RC-6
The latest beta release is v0.7.0beta31 Release Candidate 6.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

https://github.com/meerk40t/meerk40t/releases/tag/0.7.0-beta31

There are at time of release, no known bugs, most UI changes are finished.

# CAUTION
**CAUTION:** Please note that (depending on your design) the path optimisation in combination with multiple passes of the same path can result in the laser beam burning the same path several times in quick succession without allowing time for the material to cool down between. This can have unintended consequences. **Please do not use this feature of the beta whilst your K40 is unattended.**

tatarize, 2021-06-05:
first!

Sophist-UK, 2021-06-05:
We need to warn users that 0.7.0beta31 no longer shares preferences / settings with 0.6.x or earlier 0.7.,0betas.

Sophist-UK, 2021-06-05:
# UI
1. Bottom margin on default /minimum size of Position and Distances panes are larger than needed. Slight cut off of bottom of Jog pane. Pulse icon box can shrink if you make the pane smaller.
2. Managed to get the Pause pane to go very narrow in bottom bar and cannot get it back again under any circumstances. The minimum height / width when undocked need also to be the case when docked.
3. Adjusting width height of panes when docked is still very odd - should not have sticky points but should be subject to 3. above.
4. Simulation travel arrow heads still huge on some travels and miniscule on others. Ideally should be fixed size, but if not the scaling factor needs to be turned way down. Still need small circles showing start and end of travels.
5. Simulation speeds still unrelated to actual burn speeds - small circles slow straight lines very fast. Simulate still sometimes hangs for tens of seconds part way through play - but close and reopen Simulation and it plays just fine.
6. YUK!!! We got rid of flourescent yellow, but now we have flourescent pink instead. Duplicate a Cut, open both cuts and the file. Select a Cut - background of objects in that Operation mid-grey, in File a bright Turquoise, in other Cut bright pink - and red letters on bright pink background are not exactly readable. Can we just use mid-grey everywhere?
7. Would prefer to see "Lock Panes" rather than "Unlock Panes".

# Other
1. I think meerk40t.ini should be `%AppData%\MeerK40t\meerk40t.ini` rather than `%AppData%\meerk40t.ini`

ampostata, 2021-06-06:
As the new version has reset my settings, my bind to ctrl+b to autocenter document in the bed as gone... also it was written down on notes, that has gone, and i can not find it. ANd I can not find it... any one can help?

Sophist-UK, 2021-06-06:
See above comment about settings being moved and the need to reconfigure MK RC6.

ampostata, 2021-06-06:
- I have 2x K40s, and using a single monitor computer with 2 additional desktops. Machine 1+MK1 in desktop 2, and machine 2+MK2 in desktop 3. When optimizing, the "optimizing" window is shown on all the 3 desktops over all the windows, Don't know it is it wrong or correct. Just to note it.

- Can we have some feedback while optimizing? I have some optimizations taken more than 5 minutes (even a rotating \ line will do), and don't know if it has crashed or it is working

ampostata, 2021-06-06:
- Here you have the file with "ghost" movements.
[e2.zip](https://github.com/meerk40t/meerk40t/files/6604116/e2.zip)

ampostata, 2021-06-06:
@Sophist-UK yeah, I know, but someone told me that created a command from the console to center document in the bed, and has been deleted on my MK, and can not find the text / command or the bind secuence. Sorry.

ampostata, 2021-06-06:
found!

bind control+b align bedcenter

tatarize, 2021-06-06:
* We need to warn users that 0.7.0beta31 no longer shares preferences / settings with 0.6.x or earlier 0.7.,0betas.

This isn't strictly true. This holds true for MSW only. OSX and Linux will still share those settings. There is some relevance to this because of some mild incompatibilities between the versions. Mostly dealing with the binds where some commands in the console have been changed "element* delete" rather than "element * delete" for example. Which means the delete key, if taken from 0.6.x might actually not have compatibility out of the gate.

tatarize, 2021-06-06:
* found!
* bind control+b align bedcenter

That might be broken. There was some code that brought that back into working that broke some other stuff and rather than go through and solve it completely I reverted the code. So bedcenter might send to the wildly wrong place. I tend to prioritize non-console only features.

ampostata, 2021-06-06:
@tatarize I have used this afternoon and it seems to be working...

tatarize, 2021-06-06:
>- Bottom margin on default /minimum size of Position and Distances panes are larger than needed. Slight cut off of bottom of Jog pane. Pulse icon box can shrink if you make the pane smaller.

Okay. Done.

> - Managed to get the Pause pane to go very narrow in bottom bar and cannot get it back again under any circumstances. The minimum height / width when undocked need also to be the case when docked.

Okay. Done. (I also made home look a bit blue tinted, for no particular reason).

> - Adjusting width height of panes when docked is still very odd - should not have sticky points but should be subject to 3. above.

I don't really know what exactly causes that.  It manages the layout and does DoFrameLayout and DoEndResizeAction, and has some stuff with CheckMovableSizer(). I'm turning on live resize which should maybe give a better idea of what's going on. I think it might be possible to figure out why the panes do that, which ones do that and maybe make them all size adjust properly.

>- Simulation travel arrow heads still huge on some travels and miniscule on others. Ideally should be fixed size, but if not the scaling factor needs to be turned way down. Still need small circles showing start and end of travels.

Would the small circle be at the start or the end or both? I've added additional arrows at 10% and 90%. That's like the same thing right? https://github.com/meerk40t/meerk40t/commit/7975a777c6c82dbfa9699f1f7e5819a08961d271

>- Simulation speeds still unrelated to actual burn speeds - small circles slow straight lines very fast. Simulate still sometimes hangs for tens of seconds part way through play - but close and reopen Simulation and it plays just fine.

That's a bit of a tall order. Not fixing the hang, clearly I should fix the hang. It used to be worse, but somehow it's getting stuck behind and isn't dropping the frames correctly to fix it or something. 

> - YUK!!! We got rid of flourescent yellow, but now we have flourescent pink instead. Duplicate a Cut, open both cuts and the file. Select a Cut - background of objects in that Operation mid-grey, in [...]

```python
    def set_enhancements(self, node):
        tree = self.wxtree
        node_item = node.item
        tree.SetItemBackgroundColour(node_item, None)
        try:
            if node.highlighted:
                tree.SetItemBackgroundColour(node_item, wx.LIGHT_GREY)
            elif node.emphasized:
                tree.SetItemBackgroundColour(node_item, wx.CYAN)
            elif node.targeted:
                tree.SetItemBackgroundColour(node_item, wx.Colour(0xFF00FF))
        except AttributeError:
            pass
```
(Line 3408 in wxmeerk40t.py)

They mean other things than the mid-grey. The reddish background means that this is a cloned reference copy of the selected node. How will ordinary users know this is a referenced linked object of a selected object without a garish pink color? I could put it to mid-grey but then it would get confused for the children of a directly selected node. Ones children within the selected group and therefore part of the larger group, the other ones are references to the original node. Both usually end up deleted if you delete the selected object.  The cyan are indirectly selected nodes.

>- Would prefer to see "Lock Panes" rather than "Unlock Panes".

Ok. Done.

Sophist-UK, 2021-06-06:
> I don't really know what exactly causes that. It manages the layout and does DoFrameLayout and DoEndResizeAction, and has some stuff with CheckMovableSizer(). I'm turning on live resize which should [...]

I have a feeling (which means that I haven't actually tested this) that this happens on some panel bars and not others if this helps.
 
> > * Simulation travel arrow heads still huge on some travels and miniscule on others. Ideally should be fixed size, but if not the scaling factor needs to be turned way down. Still need small circles showing start and end of travels.
> 
> Would the small circle be at the start or the end or both? I've added additional arrows at 10% and 90%. That's like the same thing right? [7975a77](https://github.com/meerk40t/meerk40t/commit/7975a777c6c82dbfa9699f1f7e5819a08961d271)

Start and end. And no, arrowheads close to start and end is NOT the same thing. User needs to know precisely which path the travel starts and stops at. I havent run from real-time source to look at this but please revert the additional arrowheads.

> ```python
>     def set_enhancements(self, node):
>         tree = self.wxtree
>         node_item = node.item
>         tree.SetItemBackgroundColour(node_item, None)
>         try:
>             if [...]

Does the user care? Since the copy objects are highlighted that they are cloned reference copies (or in user speak "the same object") is inferred. The colour is not needed to make this inference.

Sophist-UK, 2021-06-06:
> * We need to warn users that 0.7.0beta31 no longer shares preferences / settings with 0.6.x or earlier 0.7.,0betas.
>
> This isn't strictly true. This holds true for MSW only. OSX and Linux will still [...]

Yes - true only for Windows.

P.S. Not that I am that bothered about it, but this ini file does **not** follow ini file standards.

P.P.S. Not that I am that bothered about it but panel_lock=0 means that the panels are locked - WTF?

Sophist-UK, 2021-06-06:
# UI
1. Convert Operation menu should be in same order as default Operation i.e. Image, Raster, Engrave, Cut.
2. Preparing simulation... box is "on top of all windows" rather than being part of the MK window - so it stays on top if you switch to another window.

tatarize, 2021-06-06:
> P.P.S. Not that I am that bothered about it but panel_lock=0 means that the panels are locked - WTF?

I should have that fixed. It was weird when I flipped the functionality I went through and made it consistent that lock means there's a caption.

Sophist-UK, 2021-06-06:
Lock should mean NO caption.

tatarize, 2021-06-07:
> I have a feeling (which means that I haven't actually tested this) that this happens on some panel bars and not others if this helps.

It seems like it uses the dock property stuff and I put that stuff up to a higher value and it seems less sketchy throughout.

>Start and end. And no, arrowheads close to start and end is NOT the same thing. User needs to know precisely which path the travel starts and stops at. I havent run from real-time source to look at this but please revert the additional arrowheads.

Okay, reverted. But, I made the lone arrowhead 75% of the way through the path. 

> Does the user care? Since the copy objects are highlighted that they are cloned reference copies (or in user speak "the same object") is inferred. The colour is not needed to make this inference.

I dunno, but this is somewhat useful information and does imply things about what would happen if some actions were taken.  I'll make this mostly gray with a cyan tint and mostly gray with a magenta tint. and leave yellow just pure gray. Though there is some requirements that "Color should not be the only method to tell one thing from another." This is due to colorblind people not being able to discern differences. So really if I want users to tell things apart I should maybe shift the font or fontstyle or something like that in addition.

Sophist-UK, 2021-06-07:
Accessibility is more complicated than just colour blindness. Not sure that blind people will be using lasers and so unsure whether MK should be screen-reader friendly, but perhaps BEEP should have a visual cue. And we should probably have a high contrast mode (for those people who have managed to partially blind themselves with their laser?).

tatarize, 2021-06-07:
Yeah, proper accessibility stuff is complex. But, it mostly comes from Whisperer that had a bad habit of allowing vector cuts to be determined from rasters by the mere ability to tell grey from blue. Which is exactly what isn't possible for some number of people. I don't know what it would take to be screen reader friendly. But, I'm not too concerned about that. It does seem like in addition to beep there might be some form of non-audio cue that something has happened. There is interrupt but that functionally stops the ability to continue until you hit the button.

https://github.com/meerk40t/meerk40t/wiki/Tech:-UI-and-UX-Design-Guide

tatarize, 2021-06-07:
4x
```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\executejob.py", line 568, in on_button_start
  File "meerk40t\kernel.py", line 138, in __call__
  File "meerk40t\kernel.py", line 1757, in console
  File "meerk40t\kernel.py", line 1843, in _console_parse
  File "meerk40t\kernel.py", line 1199, in inner
  File "meerk40t\core\planner.py", line 422, in plan
  File "meerk40t\core\planner.py", line 574, in execute
  File "meerk40t\core\planner.py", line 664, in make_image
  File "meerk40t\core\planner.py", line 651, in make_image_for_op
  File "meerk40t\gui\laserrender.py", line 453, in make_raster
wx._core.wxAssertionError: C++ assertion "bmp.IsOk()" failed at ..\..\src\msw\graphics.cpp(2575) in wxGDIPlusRenderer::CreateContext(): Should select a bitmap before creating wxGCDC
```
Apparently for some rasters during make image it crashes.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\executejob.py", line 568, in on_button_start
  File "meerk40t\kernel.py", line 138, in __call__
  File "meerk40t\kernel.py", line 1757, in console
  File "meerk40t\kernel.py", line 1843, in _console_parse
  File "meerk40t\kernel.py", line 1199, in inner
  File "meerk40t\core\planner.py", line 422, in plan
  File "meerk40t\core\planner.py", line 574, in execute
  File "meerk40t\core\planner.py", line 664, in make_image
  File "meerk40t\core\planner.py", line 651, in make_image_for_op
  File "meerk40t\gui\laserrender.py", line 464, in make_raster
MemoryError
```
This is likely a *giant* image.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3709, in specific
  File "meerk40t\core\elements.py", line 1416, in inner
  File "meerk40t\core\elements.py", line 3871, in break_subpath_elem
  File "meerk40t\kernel.py", line 138, in __call__
  File "meerk40t\kernel.py", line 1757, in console
  File "meerk40t\kernel.py", line 1843, in _console_parse
  File "meerk40t\kernel.py", line 1199, in inner
  File "meerk40t\core\elements.py", line 1954, in subpath
AttributeError: 'Polygon' object has no attribute 'as_subpaths'
```
Seems legit. Fixed.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1450, in run
  File "meerk40t\core\drivers.py", line 122, in _driver_threaded
  File "meerk40t\core\drivers.py", line 133, in _process_spooled_item
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 812, in plotplanner_process
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 989, in goto_octent_abs
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 1006, in goto_octent
ValueError: Not a valid diagonal or orthogonal movement. (dx=-1, dy=-2)
```
Crud. I was hoping this was fixed. Slightly different, still a thing.

3x
```
MeerK40t crash log. Version: 0.7.0 RC-6 on darwin
Traceback (most recent call last):
  File "meerk40t/gui/wxmeerk40t.pyc", line 2732, in on_click_open
  File "meerk40t/core/elements.pyc", line 4503, in load_types
NameError: name '_' is not defined
```
Might have existed but should be fine. This seems like Tiger since the timing doesn't match up for actual RC-6 release.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
SystemError: deallocated bytearray object has exported buffers
```
This provides nearly no information at all.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\pathproperty.py", line 187, in on_button_color
AttributeError: 'Group' object has no attribute 'node'
```
Group shouldn't be modified by path-property though it might be possible.

tatarize, 2021-06-11:
4x
```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2797, in on_click_open
  File "meerk40t\gui\wxmeerk40t.py", line 2504, in load
  File "meerk40t\core\elements.py", line 4524, in load
  File "meerk40t\image\imagetools.py", line 1686, in load
  File "meerk40t\core\elements.py", line 418, in add
  File "meerk40t\core\elements.py", line 250, in notify_attached
  File "meerk40t\core\elements.py", line 1165, in notify_attached
  File "meerk40t\gui\wxmeerk40t.py", line 3219, in node_attached
  File "meerk40t\gui\wxmeerk40t.py", line 3380, in node_register
AttributeError: 'Node' object has no attribute 'item'
```

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 1163, in open_simulator
  File "meerk40t\kernel.py", line 138, in __call__
  File "meerk40t\kernel.py", line 1757, in console
  File "meerk40t\kernel.py", line 1843, in _console_parse
  File "meerk40t\kernel.py", line 1199, in inner
  File "meerk40t\core\planner.py", line 422, in plan
  File "meerk40t\core\planner.py", line 574, in execute
  File "meerk40t\core\planner.py", line 664, in make_image
  File "meerk40t\core\planner.py", line 651, in make_image_for_op
  File "meerk40t\gui\laserrender.py", line 464, in make_raster
MemoryError
```
Expected Giant Image.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3709, in specific
  File "meerk40t\core\elements.py", line 1416, in inner
  File "meerk40t\core\elements.py", line 3871, in break_subpath_elem
  File "meerk40t\kernel.py", line 138, in __call__
  File "meerk40t\kernel.py", line 1757, in console
  File "meerk40t\kernel.py", line 1843, in _console_parse
  File "meerk40t\kernel.py", line 1199, in inner
  File "meerk40t\core\elements.py", line 1954, in subpath
AttributeError: 'Rect' object has no attribute 'as_subpaths'
```
Already fixed.

x2
```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1450, in run
  File "meerk40t\core\drivers.py", line 122, in _driver_threaded
  File "meerk40t\core\drivers.py", line 133, in _process_spooled_item
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 812, in plotplanner_process
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 989, in goto_octent_abs
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 1006, in goto_octent
ValueError: Not a valid diagonal or orthogonal movement. (dx=-2, dy=1)
```
Variant of plot after a stop.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\lhystudios\lhystudiosdrivergui.py", line 311, in on_button_set_home_current
  File "meerk40t\gui\lhystudios\lhystudiosdrivergui.py", line 272, in calc_home_position
AttributeError: 'LhystudiosDriverGui' object has no attribute 'bed_dim'
```
Home Right home left calls self.bed_dim without defining it. Corrected.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\scene\scene.py", line 175, in on_left_mouse_up
  File "meerk40t\gui\scene\scene.py", line 541, in event
  File "meerk40t\gui\scene\scenewidgets.py", line 473, in event
TypeError: 'NoneType' object is not subscriptable
```
Some click order allows end_location be None rather valid during left-up.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on darwin
Traceback (most recent call last):
  File "meerk40t/gui/wxmeerk40t.pyc", line 2732, in on_click_open
  File "meerk40t/core/elements.pyc", line 4503, in load_types
NameError: name '_' is not defined
```

"All Valid Types" as a text should have _ defined. Only on Darwin/OSX

ampostata, 2021-06-13:
This takes ages on my computer to optimize.

The green one, as my other "wood" patterns, I have to burn without optimization: I waited for 20 minutes and have to close it, as I didnt know if it was working or not.

But the blue one also takes too long, around 10 minutes, and it should not, I guess.

Just check it. Thank you.
[via.zip](https://github.com/meerk40t/meerk40t/files/6644625/via.zip)

ampostata, 2021-06-13:
- Can we have an option to save the compiled/optimized/you-name-it generated in the spooler? (and load it also in the spooler window), so my so long "optimizations" can be saved optimizated :) something like a "raw" file.

Sophist-UK, 2021-06-15:
More burning today. Some feedback:
# Bugs & not bugs
1. I ran some vector engraves of lines of small circles with and without optimisation (but both with Rapid moves on) - without optimisation no twitches, with optimisation big twitches.
2. Emergency stop is now working great - home works first time afterwards.
3. Launch spooler Job Start isn't working.
4. Reverse SVG element load needs to be changed from reversing SVG as it is loaded to reversing SVG as it is classified into operations. Option should now be "Reverse default element order in operations" with a tooltip "Classify elements into operations in reverse order to match Inkscape's Object List".
5. Bed dimensions should be part of the device configuration IMO not a global Setting.
6. Disable tool tips tooltip needs a tweak "Note: You will need to restart MeerK40t for any change to take effect."
7.

ampostata, 2021-06-16:
Question: if I load a SVG file, spool it with optimize, and save it back as SVG, is it saved "optimized"?

tatarize, 2021-06-16:
> Question: if I load a SVG file, spool it with optimize, and save it back as SVG, is it saved "optimized"? (or "compile & sent to simulator")

There's no method of doing that. There's a couple methods of getting cutcode in the Operations. For example `ruidadesign` and sending in data from RDWorks would create cutcode objects. But, you can't, currently, then turn around and save those. There's also methods to use the GRBL server to get lasercode which could then be converted into cutcode, but you could not save that either. And there's an oversight not be permitted to save various command nodes, for example home and interrupt, that might also need methods of saving.

ampostata, 2021-06-16:
@tatarize it could be great to save "raw" data or SVG optimized files... I have one that takes 20 minutes to optimize in my son's gaming comoputer (a beast). On my burning laptop could take hours.

At least save the "bits" in the SVG file in the optimized order... currently, without optimizing, the head jumps from one side of the sheet to the other, like making the very worst movement available to waste time. Trolling me.

ampostata, 2021-06-16:
- The blue horizontal lines: some are burned twice (go and back), others just once. Should be only one. Also many ghost moves.
[Casa_Rota-3.zip](https://github.com/meerk40t/meerk40t/files/6665925/Casa_Rota-3.zip)

tatarize, 2021-06-16:
I'd have to assume that the burned twice stuff are actually there twice.

While it might be nice to get rid of the double cuts it's not a problem I'm planning on solving in the near future.

ampostata, 2021-06-16:
No, it is not. There is a single line, and it makes two pases.

tatarize, 2021-06-17:
I'll check. If it's unilaterally adding an extra cut that isn't supposed to be there that's a critical bug.

ampostata, 2021-06-17:
Here you have a much smaller file with the same error.

[error.zip](https://github.com/meerk40t/meerk40t/files/6668278/error.zip)

Sophist-UK, 2021-06-17:
I think it is likely that the SVG has two paths, even if there is one object that shows on screen. It could be one of the following:

a. A combined path, consisting of two identical paths combined.
b. e.g. rather than a single circular path it could be a hoop consisting of an outside path and an inside path.

Both of these will result in double cuts in MK.

Sophist-UK, 2021-06-17:
I took a look at error.zip and it looks fine. So sorry @tatarize, it does look like an optimisation bug.

ampostata, 2021-06-17:
As some files takes ages on my computer to optimize (and on a beast computer too), in my head there is always flying solutions to improve this. As I have been told before, the SVG file can not be saved optimized, but... orderer?

Correct me if I am wrong: The SVG file is done with a chunk of XML entries, each a "line", or a circle...

Can we have an option to "order" the chunks of the SVG files? I mean: the "orderer" choose the first element in the list, then search the closest one and place it 2nd place, then search the next one closest to the 2nd. So the (saved) SVG file will have the chunks half-optimized.

Here you have a wood pattern I am adding to most of my design, and it is the one that make (my) optimizations taking so long.

If you see the file in actual burning (can simulate be done without optimizing?) you will see the laser jumping from one side to the other like crazy, with no real logic :)

I don't know if optimizing speed can be improved, but with this "ordered" sequence, the time will be improved without a real optimization

Anyway, forget all about this if the problem is with my SVG file that can be fixed somehow (I am now searching for SVG optimizers, perhaps there is a solution out there)

[wood.zip](https://github.com/meerk40t/meerk40t/files/6672045/wood.zip)

Sophist-UK, 2021-06-17:
In inkscape you can use the Object List to order the objects - then MK without optimisation (and assuming that you haven't turned on the reverse Inkscape order setting) it will burn the bottom most object first.

In other words, if you don't want to wait for MK to optimise, you can manually optimise the burn sequence yourself using Inkscape.

tatarize, 2021-06-20:
Report came in saying crosshatch didn't work.

Sophist-UK, 2021-06-23:
# Bugs
1. Loaded a file saved by MK to get the operations (6x duplicate cut operations) - then went to the file object and removed it and added a new file. 3 of the 6 operations retained the cuts from the previous file.

Sophist-UK, 2021-06-24:
# Bug
1. Cloned objects are loaded incorrectly - in the wrong place(s) at the wring size.

Sophist-UK, 2021-06-26:
# Buggettes
1. Laser apparently not communicating - turned out that MK was paused but there was not **any** indication in the UI that this was the case.

Sophist-UK, 2021-06-29:
# Bug
1. I had an SVG with text. I duplicated the text object, hid the original, converted the copy to paths and combined the paths into a single object. When I loaded the SVG into MK, the hidden original text was loaded / shown. In the XML it is hidden by setting css `display:none` rather than attribute `visibility="hidden"` - so it appears that there are two ways of hiding SVG objects. (As an aside, the single text object in Inkscape was also converted into two text objects in MK - looking at the XML it seems that Inkscape stores each line of a text object as a `<tspan>` sub markup . Not sure whether this is connected with it showing up..)
[propriedade privada.zip](https://github.com/meerk40t/meerk40t/files/6734331/propriedade.privada.zip)
[image]

tatarize, 2021-06-29:
>Laser apparently not communicating - turned out that MK was paused but there was not any indication in the UI that this was the case.

Hm.  This is entirely accurate. Not sure where the fact that it's paused should go. There's not even a simple pane for it.

> In the XML it is hidden by setting css display:none rather than attribute visibility="hidden" - so it appears that there are two ways of hiding SVG objects. 

In the spec these do slightly different things. In that `display: none` is not parsed at all and visibility hidden, is parsed but then omitted. Yes, the tspan markup is most of the issue there, they are a bit weird and get treated somewhat like text objects but they are really just text span objects that have their own sub-properties.

> Selecting corners or sides of objects on scene is VERY tricky because the mouse has to be almost exactly over the corner. Can we make this less precise please.

Yes, this needs some correcting I think I tried a bit but its somehow related to some other bugs. I think it might be a bit wider but doesn't appear to be, because it's a mouse changing bug.

>(Very minor) Rectangles with curved corners appear OK on scene but do not have rounded corners in tree.

Hm. That might be minor, but it's really weird. It should render both of these in the same way. 

>Disabled operation - you can right click and select execute but it tries to execute everything else instead of the disabled operation. IMO disabling an operation should stop it being included when you click the Execute Job button in the ribbon but should still allow you explicitly execute it with a right click on the operation.

Yes, this is absolutely the case.

>Move To pane not updated with position. Move To box in Navigation Window not updated to current position when the window is opened.

This sounds entirely accurate. I'm not sure if this is the intended results since move to is to move to that location and if it constantly updated to the current position it would by default move nowhere. I'm not sure these should have the current position and I think that might be why they were omitted.

tatarize, 2021-06-29:
>In the second image, the tiny white blob in the bottom RH corner is the bed with thee original object (left hand). And the huge disaggregated object is the clone.

I might need to get sent a copy of the file to check what happened there. Clearly it's a bug though.

Sophist-UK, 2021-06-29:
> > Laser apparently not communicating - turned out that MK was paused but there was not any indication in the UI that this was the case.
> 
> Hm. This is entirely accurate. Not sure where the fact that it's paused should go. There's not even a simple pane for it.

There is a Pause button in the ribbon and a Pause pane. Both of these should IMO either switch background colour when paused, or change the pause icon to a play icon.

> In the spec these do slightly different things. In that `display: none` is not parsed at all and visibility hidden, is parsed but then omitted. Yes, the tspan markup is most of the issue there, they [...]

In Inkscape all you do is click the eye-icon in the object list to make the object (group or individual object) invisible. It used to work but I upgraded to Inkscape 1.1 and now it uses CSS to hide it rather than the visibility attribute. Same action in Inkscape, different result in MK. :-( 

> > (Very minor) Rectangles with curved corners appear OK on scene but do not have rounded corners in tree.
> 
> Hm. That might be minor, but it's really weird. It should render both of these in the same way.

It is possible that the corners are just so small they cannot be seen.

> This sounds entirely accurate. I'm not sure if this is the intended results since move to is to move to that location and if it constantly updated to the current position it would by default move [...]

Sorry - but I obviously didn't explain myself very well. The X / Y boxes should show the existing position after a move. You can then change the values and click the button to move to a different position. The issue is that the boxes don't show the current position as starting values.

Sophist-UK, 2021-06-29:
hmmm - I hid the text object (xml: css `display:none`) and yet it still displayed in MK. I hid a rectangle object (xml: css `display:none`)  and MK did **not** display it.

Sophist-UK, 2021-06-30:
# Issue (not really a bug)
1. Was rastering at quite a slow speed - and I had a x-shift. I wasn't watching but I suspect that there was a USB error and when it recovered it was out of position. Is it possible to have a Setting / Device Config for "Home after USB error" - with code that recovers from a USB error by homing and then (cleverly) recovering from where it failed?

gsboylan, 2021-07-03:
Usability issue - not sure if this constitutes a bug or just unexpected behavior

1. Import a DXF object (here: https://drive.google.com/file/d/1zgtKqZbLzWcO2NqEQn8_Z4q3ZQmHFDTO/view?usp=sharing)
2. Select the object, right click, "duplicate"
3. In the tree view, duplicates are ungrouped into individual cuts rather than grouped by objects
[image]

Would be nice if duplicating a dxf preserved groupings

tatarize, 2021-07-04:
Got optimization at least as good as before and added in some merge options. So moving on to the backlog of fixes.

@ampostata
> Can we have an option to save the compiled/optimized/you-name-it generated in the spooler? (and load it also in the spooler window), so my so long "optimizations" can be saved optimizated :) something like a "raw" file.

Extremely cursory work done towards this, put into developer mode. Might fix it if there's time. `plan return` as a command is intended to do this work.

@Sophist-UK 

- I ran some vector engraves of lines of small circles with and without optimisation (but both with Rapid moves on) - without optimisation no twitches, with optimisation big twitches.
     - Maybe there's some other issues with directionality or something.
- Launch Spooler JobStart isn't working (and should probably be "Launch Spooler on Job Start").
     - Changed name. 
- Reverse SVG element load needs to be changed from reversing SVG as it is loaded to reversing SVG as it is classified into operations. Option should now be "Reverse default operations element order" with a tooltip "Classify elements into operations in reverse order e.g. to match Inkscape's Object List".
     -  OK done.
- Bed dimensions should be part of the device configuration IMO not a global Setting.
     - Joe agrees with you. And I also tend to agree I'm planning on switching it after the full release. It needs some slight related changes and switch to steps from mm.
- Disable tool tips tooltip needs a tweak "Note: You will need to restart MeerK40t for any change to take effect."
     - Ok. Done.
- SVG files are still not saved with groups and friendly names.
     - Yeah, not sure if that

Sophist-UK, 2021-07-04:
>> * I ran some vector engraves of lines of small circles with and without optimisation (but both with Rapid moves on) - without optimisation no twitches, with optimisation big twitches.
>   
>   * Maybe there's some other issues with directionality or something.

Not that sort of twitch. These are bigger than the normal twitch and look like they are an explicit artefact of optimisation.

>> * Launch Spooler JobStart isn't working (and should probably be "Launch Spooler on Job Start").
>   
>   * Changed name.

Did you fix it too?

> * OK done.

Brilliant!

>> * Bed dimensions should be part of the device configuration IMO not a global Setting.
>   
>   * Joe agrees with you. And I also tend to agree I'm planning on switching it after the full release. It needs some slight related changes and switch to steps from mm.

Not urgent.

>> * SVG files are still not saved with groups and friendly names.
>   
>   * Yeah, not sure if that

Not sure if that....???

gsboylan, 2021-07-09:
Bug - inside cuts are not being performed first even with the setting checked.

Sophist-UK, 2021-07-09:
# Bugettes
1. Spooler - Right click to remove special operation removes first such operation in list not selected operation.
2. Spooler - Right click - Remove operation shows python object address not Name.
3.

tatarize, 2021-07-09:
- Loaded a file saved by MK to get the operations (6x duplicate cut operations) - then went to the file object and removed it and added a new file. 3 of the 6 operations retained the cuts from the previous file. Reclassify Operations made no difference. Bug hunting hint: It appears to remove from alternate operations.
     -  I'll make a note. I think the duplicate creates a kind of cline for the sub-items which wouldn't remove correctly. 

- Enhancement suggestion: burn operations seems to fill the laser buffer regardless of whether instructions are very short (e.g. rastering) or very long (long straight vector cuts). Can we estimate the time for a vector operation and delay sending the instruction until (say) a second or two before the laser catches up - so that the screen rendered laser position is not more than a second or two ahead of the actual laser position? (Or perhaps we send the instructions AFAP as at present, and simply delay the movement of the red circle.
     -  Pre-estimation is likely going to be hard. But, the lhyemulator would do this for the most part. If you type `lhyemulator` it will read the sent command to the laser and be basically without buffer and should be closer to real time. It will add an additional reticle for that.

- We really need a better overall progress indicator than the little red circle which gives no idea about overall progress % or even which duplicate operation or Passes it is doing. What about some sort of indicator in the tree (if it is possible to relate cut paths back to operations and possibly elements in the operation?
     -  This is pretty accurate. But likely deferred.

Sophist-UK, 2021-07-10:
# Buggette
1. After using raster wizard you cannot use the Position Pane / Window to move the image. I appreciate that you wouldn't want to allow e.g. rotations or scaling, but surely simple X/Y translations are OK?

tatarize, 2021-07-10:
That window calls a particular routine that can scale it so the do not permit that command applied, but you are right there are certain unlocked operations that would be no concern within that. It should move, but then refuse to increase in size with other things.

tatarize, 2021-07-11:
- Spooler - Right click to remove special operation removes first such operation in list not selected operation.
- Spooler - Right click - Remove operation shows python object address not Name.

The fixes for these might actually be to remove them. Spooler might need a bit of an overhaul, and it's likely mothballed for a version or two.  Lemme see how easy it would be to fix.

Sophist-UK, 2021-07-11:
> * Spooler - Right click to remove special operation removes first such operation in list not selected operation.
> * Spooler - Right click - Remove operation shows python object address not Name.
>
> The [...]

I think that "buggettes" by being minor are by definition fix now if easy otherwise leave until a later version. 😃 

> Appears to only show the python object name if that's the underlying name. The other thing certainly is correct.

I am guessing that the default `__name__` method is being used because the Python objects for special operations do not currently have an overriding `__name__` method.

tatarize, 2021-07-11:
```
  File "<frozen importlib._bootstrap_external>", line 1109, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "_ufuncs.pyx", line 1, in init scipy.special._ufuncs
ModuleNotFoundError: No module named 'scipy.special._ufuncs_cxx'
```

2x. Mac Compile fails. The expectation there was to use the fallback in svgelements length_scipy to fall back on *any* routine. Ah, version RC-3 for Darwin. Version is old.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "mk40t.py", line 13, in <module>
  File "meerk40t\main.py", line 412, in run
  File "meerk40t\kernel.py", line 860, in bootstrap
  File "meerk40t\gui\wxmeerk40t.py", line 206, in plugin
  File "meerk40t\kernel.py", line 1757, in console
  File "meerk40t\kernel.py", line 1843, in _console_parse
  File "meerk40t\kernel.py", line 1199, in inner
  File "meerk40t\gui\wxmeerk40t.py", line 4162, in window
  File "meerk40t\kernel.py", line 701, in <lambda>
  File "meerk40t\gui\wxmeerk40t.py", line 4155, in window_open
  File "meerk40t\kernel.py", line 472, in open
  File "meerk40t\kernel.py", line 506, in open_as
  File "meerk40t\gui\wxmeerk40t.py", line 395, in __init__
  File "meerk40t\gui\wxmeerk40t.py", line 1103, in __set_ribbonbar
  File "meerk40t\gui\icons.py", line 16, in GetBitmap
  File "wx\lib\embeddedimage.py", line 64, in GetImage
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1694) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!
```

3x. This is the locale bug it's the reason there's two different versions of the windows version.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2797, in on_click_open
  File "meerk40t\gui\wxmeerk40t.py", line 2504, in load
  File "meerk40t\core\elements.py", line 4524, in load
  File "meerk40t\core\svg_io.py", line 289, in load
  File "meerk40t\core\svg_io.py", line 349, in parse
  File "meerk40t\core\svg_io.py", line 349, in parse
  File "meerk40t\core\svg_io.py", line 349, in parse
  File "meerk40t\core\svg_io.py", line 314, in parse
  File "meerk40t\core\elements.py", line 418, in add
  File "meerk40t\core\elements.py", line 250, in notify_attached
  File "meerk40t\core\elements.py", line 1165, in notify_attached
  File "meerk40t\gui\wxmeerk40t.py", line 3219, in node_attached
  File "meerk40t\gui\wxmeerk40t.py", line 3398, in node_register
  File "meerk40t\gui\wxmeerk40t.py", line 3440, in set_icon
  File "meerk40t\svgelements.py", line 3650, in bbox
  File "meerk40t\svgelements.py", line 3651, in <listcomp>
  File "meerk40t\svgelements.py", line 4408, in bbox
  File "meerk40t\svgelements.py", line 4421, in _real_minmax
OverflowError: (34, 'Result too large')
```

The offending line appears to be `delta = a[1] ** 2 - (a[0] + a[1]) * a[2] + a[2] ** 2 + (a[0] - a[1]) * a[3]` and it appears to have outstripped the size of a floating point number. So like 1e255 or so.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2797, in on_click_open
  File "meerk40t\gui\wxmeerk40t.py", line 2504, in load
  File "meerk40t\core\elements.py", line 4524, in load
  File "meerk40t\dxf\dxf_io.py", line 52, in load
  File "ezdxf\filemanagement.py", line 154, in readfile
  File "ezdxf\filemanagement.py", line 94, in read
  File "ezdxf\document.py", line 282, in read
[... 346 lines cut ...]
```
MeerK40t crash log. Version: 0.7.0 RC-6 on darwin
Traceback (most recent call last):
  File "meerk40t/gui/wxmeerk40t.pyc", line 2732, in on_click_open
  File "meerk40t/core/elements.pyc", line 4503, in load_types
NameError: name '_' is not defined
```

20X - Does not appear to exist. Only Mac

```
MeerK40t crash log. Version: 0.7.0 Beta-17 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2988, in on_item_selection_changed
AttributeError: 'Rect' object has no attribute 'node'
```
1X - Rect object somehow not registered.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\pathproperty.py", line 203, in on_button_color
AttributeError: 'Group' object has no attribute 'node'
```
12X - Change Group Color, fails.

Sophist-UK, 2021-07-14:
# UI
1. Don't include empty operations in list on Execute Job.
2. In the tree, dim operations that have no elements to burn.

[1 more reply]

closed 2021-07-16 by Sophist-UK