meerk40t/meerk40t #445: Raster alignment is off and tree plane becomes weird.

bergie5737, 2021-06-08:
Hi.
I have a few puzzles to laser cut and decided to see how this fair compared to K40Whisperer. I'm impressed with the software and sure it is better! Thanks for the hard work. One issue I found, is where I make a small name puzzle, the text is not inside the puzzle anymore. See screenshot.

[image]

I tried to enlarge the "tree" pane, to show the font detail, but then the screen becomes weird, See below.
[image]

Hope it helps. I am trying v0.7.0 RC6 on Windows 10.

Sophist-UK, 2021-06-08:
You need to convert the text to paths inside Inkscape to make this work.

tatarize, 2021-06-09:
The text there is dynamic. The idea was to do the harder thing and keep it editable so you could do things like make flowable text or a series of monogrammed basesballs or something. And not need to make a different svg file for each. The arranging of the text, however, is a bit hard and is off. Text2Path would bypass this and just make the text a normal shape. It's a clear defect in the program currently and methods of fixing this, thus far have been a bit lackluster.

Abizarl, 2021-08-09:
Love what is being done with the software. Have the same issue as bergie5737. Text (rasters) are off but also having an issue with imports of non-raster data. Have not tried the text to paths yet for the raster data
[image]
. The attached image shows a few files after import into Whisperer and into MeerK40t.

Sophist-UK, 2021-08-09:
As I said before, currently you need to convert the text to paths inside Inkscape to make this work.

Abizarl, 2021-08-09:
Thanks for the quick response!

closed 2022-02-20 by Sophist-UK