meerk40t/meerk40t #446: Super speed on laser, causing missing steps etc on v 0.7

bergie5737, 2021-06-13:
Hi
I've tried cutting and engraving before with no problems. Today I converted a grid pattern to vector engrave to save time. I set to engrave at 50mm/s (settings I used from K40Whisperer) and hit start. The first thing to happen, was the vector engrave, even though I set it to do the raster engrave first. It did not matter at all in this case, but the speed is just crazy. I flung the head so fast that the adjustable focusser lost alignment and it skips steps like crazy. 

I did the obvious and reduced speed to like 10mm/s, just a bit faster than cutting speed, but the result is the same. I thought my laptop or controller has gone mad, and reset both with power cycle, same result. I then tried v0.6 and the file seems to work fine.

And maybe as a feature request, please add deeper "Undo" levels. I stuff up too frequently as a newbie!
Many thanks to all the developers for this project!

Sophist-UK, 2021-06-13:
Please confirm the exact beta version of 0.7.0 you were using and upload a version of the SVG file saved from that version.

tatarize, 2021-06-14:
50 is fairly quick. The optimization can certainly move those things out of order. My best guess is that it's not doing jogging. So what you get is it moving to the next location at the speed of the object it's going to engrave in this case a fairly fast raster. So it goes to the raster at raster speed, which it really shouldn't do.

Sophist-UK, 2021-06-14:
Is this a bug in optimiser?

bergie5737, 2021-06-14:
Hi. I'm using v0.7.0 RC-6.
I had to upload as a zip as svg is not supported. The blue lines is my problem. 

As mentioned I used speeds that worked for me from K40Whisperer. Raster engrave 250mm/s,  vector engrave (blue lines) as 50mm/s and cut 8mm/s. 
The only difference is I used the 'ppi function in Meerk40t to have the "same" power, i.e. 400ppi to get 40% power etc. When the blue lines were "rastered" it worked fine, even though I did not manage to cut right through the plywood as expected, but that is a small issue, probably more to do with my laser's power curve than with the software.

[calendar backing.zip](https://github.com/meerk40t/meerk40t/files/6647916/calendar.backing.zip)

bergie5737, 2021-06-14:
Hi. Just as further information. In Optimizations, I turned off "Rapid moves between objects." But it made no difference. 

I just now figured out how to simulate just the blue grid and it basically seems to be as erratic as the actual cut. (At least it did not bang my laptop around! LOL!) But I did expect it to draw the lines one full length at a time in stead of mostly single short lines.

tatarize, 2021-06-14:
Well, if you turned off optimizations altogether you could likely see if it was caused by optimizations. There's some reasonable chance that it's doing things you generally would not want. I'm not sure this is a bug in optimizer but since that is a major change and something is weird with the ordering it's the number one suspect.

bergie5737, 2021-06-20:
Hi. I am busy with upgrades, so spare laser time is for that! Will try again when it is up and running again. No that the week end is gone, it becomes unlikely this week.

bergie5737, 2021-06-21:
Hi tried tonight with the same file. Turning optimizations off does not do that much better. It stops the "ramming" but I can see how dots looks like it "spreads out" on the work. It seem that after a direction change the dots are spaced further apart, and then gets closer as it appears to break before the next direction change. I don't think dots should be separate trying to vector engrave. I did 50mm/s, then 30mm/s, the issue is the same, only less pronounced.

Sophist-UK, 2021-08-13:
@bergie5737 

1. Please confirm the problem still exists in RC8.
2. Please upload your original SVG file and a version of the SVG saved by MeerK40t (which has the operations defined).

MiBu-dev, 2022-01-19:
Hi
Mine k40 cannot vector ebgrave faster than 30 -35mm/s - most usual it will loose steps on Y axis -and engraving is misplaced :(
I tried acceleration setting reduced to 1 - but it didn't help
also if I take metric ruler to engrave  as vector - at 35mm/s - vertical lines  are wavy - it is vibration from axis  - reducing to 30mm/s helps  keep things straight 
and even at 35mm/s   sometimes there is Y axis skip - for example such ruler - it starts engraving scale from 0  to 4 then skips few millimetres and engraves up to 10cm  and then go back  to engrave missing ones  but they are not aligned  - I can provide photo of such result.

I tried to set acceleration reduced to 1 but it doesn't help much.
I tried to  tighten belts, loosing them - no effect, gantry feels rigid - not much play  between ends (cannot be skewed manually  - I cannot feel any play by hand - if enough force is applied - it will skew - always  ;) just a matter of force :P
I had  cleaned and greased  rollers, checked for play on stepper <>axis coupler 
I upgraded power supply - nano board with motors is powered from external 24V PSU
what I was able to read somewhere -  k40 parts are on the budget  - cheapest  found in all China ;)
so most likely there is a culprit.
Funny thing is - it can raster engrave even at 400mm/s (not crosshatch) although power has to be increased

I am in the process of converting gantry to V-slots  and maximizing work area to 600x300mm - I will most likely change motors  - and then I will test

Best regards
Mike

Sophist-UK, 2022-01-19:
@MiBu-dev @tatarize Here is my take on this:

Clearly the steppers are capable of moving the head in both directions at faster than 35mm/s because we raster in those two specific directions at 10x+ that speed.

Whatever speed you choose, the head needs to reach that speed and maintain it for the entirety of the burn - which means that it needs to be able to achieve the acceleration needed on even the smallest / narrowest of curves. It is possible that when you get to sharp corners you stop burning, do a travel and approach from a different direction, and I have seen this happen in practice, though not sure whether this is something that happens automatically in the m2 nano or something that MK explicitly does.

So... it seems to me that:

Suppose you set 100mm/s for a vector burn at 1000 ppi.

It would then be possible to do some paths at 100mm/s at 1000ppi if they were straight or only slightly curved.

And then we could be clever, and burn tighter curves at (say) 30mm/s at 300ppi and get the same strength of burn.

**In other words, you MeerK40t could do variable speed vector operations depending on the curvature of the path, and thus reduce the time taken.**

Sophist-UK, 2022-01-19:
@MiBu-dev 0.7.4 / 0.7.5 beta 1 are the current versions.

Default speed for new Engrave operations is 35mm/s but you can change the speed either by double clicking the operation node to open properties or by using a right click menu.

I have also considered changing the 24v supply for the steppers to a separate PSU as my steppers don't seem to have the power I think they should have - but I will not bother given your experiences.

You should also consider setting `steps` of 2 or 3 on your raster operations to speed them up. In essence IMO you should almost always be burning at 12mA, and using speed and steps (or PPI on rare occasions) to reduce the burn depth where needed. So rather than doing raster engraves every 0.25mm (which is approximately 10 mil) doing 60 sweeps at 35mm/s you can do e.g. steps 5 at 200mm/s and be even faster.

MiBu-dev, 2022-01-19:
@tatarize  - I will check tension again - although belts are not loose and make sound - not like guitar strings but are tensioned as good  as in my 3d printer - although my k40 worked before in transformer workshop - cutting a lot of parts 
I made a plan to test  some speeds  and shapes  - so I will keep you posted.
I will test also Whisper with same files  to be sure
I have checked axle coupler  on the motor - all screws are tight and there is no free play on that axle - gantry with lock enabled doesn't appear to have it also - but those are  minimal resonance  disruptions - same effect is seen with 3D printing - when head is turning 90deg on print appears pattern - because line is not straight  but waving  because of  vibrations 

I thought that accel parameter was just like in Marlin firmware for 3D Printers - determining how fast  stepper driver will spin  the motor to achieve full speed  and how fast  to decelerate before direction change - it has quite an impact    called "ringing"  when on object's wall is some detail and print head is making  movement changes  - sometimes it is just  loose belt, sometimes  the mass of the whole bed moving ... not the topic here ;)

@Sophist-UK  - I am using 0.7.4  on Ubuntu 20.04( so I updated wiki  because manual was not up to date  - URL missing )

PSU upgrade  I made  after https://k40.se/k40-laser-electronics/connect-a-24v-external-psu/
and the fact  that previous owner  blow the original one , I had replaced HV transformer but .. PSU exploded after few test (damaged mosfets or something else)  so I replaced whole unit from Cloudray and it is OK.

thanks for the tip about steps - somehow  I did not consider this (maybe as beginner playing with image engraving - this option is grayed out (set to pass through) - that is exactly what I was thinking - instead of glacial line by line - few lines with more power although slower - in such case material is not vaporized but carbonized :) so it is darker  (on some plywood types  engraving  leaves beige   (somewhat darker) marks. Heck  - on stained  plywood i managed even to... engrave in  "white"  - paint was removed and  clear wood  appeared (it was not  as dark as engraving clean plywood (it was stained one side)

I will test steps and do some timings.
I was thinking about automatic hatch infill depending on color  - it can be achieved manually ofc

MiBu-dev, 2022-01-19:
OK - i recorded video, made some samples, tighten the belt from one side some more,
when running  simple 20mm circle with rectangle inside - almost everything went OK up to 85mm/s (although some waves appear 
[image]
at 95mm/s it went nuts - at 105 rectangle was misplaced
as gantry  was making some stuttering noises - that's was the reason i suspected  motors
 I opened the lid  and started poking around - and... voila - roller in aluminum profile  has worn bearing :(  it has   unexpected play up/down (gantry is not moving  in that plane - it would be Z axis :)   
the play is minimal - about 0,5 mm - but wheel is skewing and most likely not turning  and making gantry to catch on one side
as  I am going to gut out everything  inside to make room for new  gantry  from 2020 and 2040 V-slot profiles - there is no point to reanimate this bearing - replacement from Aliexpress would take too long to arrive - and  default work area is ridiculous 

Thank you very much for patience and helpful hints.

As for  infill with hatch infill - hmm  - remarks from @Sophist-UK  were quite helpful - I will experiment some more with steps and speed  - hatch infill can be still be useful in some cases (except  that after applying them in Inkscape plugin project become non scalable somewhat

tatarize, 2022-01-19:
@joerlane What is that? I swear you mentioned something almost exactly like that.

I'm pretty sure there's a hardware issue that exactly matches your results I forget what exactly is though. Something like belts are too tight or something. Joe might know assuming he's not neck deep in his new toy.

tatarize, 2022-01-19:
Also, most folks stop cutting speeds before going that high, it's sort of hard to keep the machines from getting very iffy above 50mm/s just because of the amount of weight being thrown around.

Sophist-UK, 2022-01-20:
> Also, most folks stop cutting speeds before going that high, it's sort of hard to keep the machines from getting very iffy above 50mm/s just because of the amount of weight being thrown around.

This is true for current MK and @MiBu-dev 's current problem.

But as discussed above, the issue is acceleration on curves - hence my suggestion for a future enhancement which does variable speed cuts, speed of a path-segment varying upon the curvature of the path-segment, and using PPI to compensate for burn depth when the speed is reduced from the Operation speed.

tatarize, 2022-01-20:
I think if you switch to jog reset it might actually work better at higher speeds. The core idea is that the twitching during a full reset between two different jumps should actually twitch in the correct direction. Then if you're mostly going in an orthogonal or diagonal direction you should decelerate correctly directioned. So rather than bank a turn at speed you'd cut the turn into two different sections where one side would jerk the slowing position and the other would jerk up to speed the other way. I don't know what the gain/loss there might be where you'd basically need to decelerate at every rapid speed switch. For a rectangle as an example you'd go to the right, decel, acel down, go down, decel, acel left, go left etc. So you would avoid any corners if you could avoid them.

tatarize, 2022-01-20:
The NSE jogs twitch according to the originally set raster direction. Whereas @NSE jogs would terminate and should jog in the direction the stepper motors were currently set. So you should get the 8 different directions of jogging. You'd likely need some code to calculate the current acceleration at any particular point and stop when it gets too high. Acceleration during curves that is as you'd go the same speed but the vector of movement would be changing direction, if you would be jogging entirely x-parallel or y-parallel you that happened to match the major axis for the rastering you could also just do an NSE jog in that direction. The end result is that you could actually go at raster speeds for non-rasters so long as you happened to be going raster directions. But, you'd still need code that could calculate where you'd need to throw in a break especially if that was a corner or corner like part.

The same code thing I'm doing with ruida rasters would sort of apply there. If you just happen to be going a direction where you could apply an accel or decel in a coherent fashion, you could do that selectively. This might be best in cases where you switch from going right to going left. And in those cases you could even call an NSE jog to switch your directionality. In the case where you're going along the would-be raster axis this would correctly switch your directionality. And in cases where your ending directionality isn't along this axis you could just apply an @NSE jog which would twitch in the current direction.

There is also some potential to cut plan this job so that you use a metric where you don't go to the closest point but the point that happens to be along the same directionality. In the cases when your jogging but the jog location is basically in the same line you could continue through there at speed and just start firing the laser on the otherside switching the jogging altogether. If code that did that was a thing optimizing for that would also be reasonable.

So if your next cut requires a turn around and are going the wrong direction you terminate with @NSE and either rapid move or start up at speed going the correct direction (since the rapid might actually be slower than too fast laser head). If you need to turn around and you're going the direction of your major axis you set during the startup you could apply an NSE raster to turn around. You wouldn't even need to do anything you'd just say you're going right, NSE, now you're going left and the turn around would apply a raster twitch turn around which is actually pretty reasonable at high speeds. And if you're going the same direction as the next segment you'd actually just skip that gap completely and never jump out of program mode. You'd maintain your speed and reach the other side of the gap which would need to be more-or-less in the same direction. Then anywhere where you'd apply a rapid change in direction of unknown thresholds or calculations you'd break that turn with some @NSE jogs.

closed 2022-07-20 by tatarize