meerk40t/meerk40t #449: USB Rejected

kingspidor, 2021-06-24:
I keep getting a USB connection rejected message, anyone have this before?

Sophist-UK, 2021-06-24:
Your MeerK40t version and operating system

tatarize, 2021-06-26:
RPi can reject some stuff based on permissions. You can try running it with sudo, there's some instructions in the wiki to help some stuff.

kingspidor, 2021-06-27:
0.7.0rc & 0.6.6; win10,  & Linux mint 19

kingspidor, 2021-06-27:
It seems to be working now, but only from physical console (not over VNC). I have to open Meerk40t, then run K40 Whisperer & initialize laser, then exit Whisperer for it to work.

tatarize, 2021-06-28:
@inspectionsbybob reported having something similar. I'm unsure of anything Whisperer does during an initialization that MeerK40t wouldn't. I guess it does a home command but I don't see how that could change much. I guess I could map them out in some more detail.

Sophist-UK, 2021-06-28:
> 0.7.0rc & 0.6.6; win10, & Linux mint 19

That is a pretty comprehensive quadrant of version and O/S. Either an issue in MK or an issue with your h/w.

tatarize, 2021-06-28:
There's also the permissions issues. Though it might not be VNC. You really could just run the linux system or whatever in CLI mode and running the Lhyserver and connecting to it from a local windows 10 machine. Though I didn't think folks beyond inpain or Tiger ran in that fashion, but I wouldn't be surprised to see it get more popular especially since an PI-zero can basically do all the processing needed on the host machine.

kingspidor, 2021-07-06:
Works great on my Pi400, though only when physically at it, not over VNC nor SSH. So far I cannot even get meerk40t to even run on my Mint 17.3 (possibly something to do with a Python & Python3 issue?), not sure.
Can Lhyserver be run standalone, like as a service/module or something? I have fallen years behind in in my computer sciences, learning to fall behind farming sustaianly, lol. Meerk40t is worth struggling over to put on all my systems though. 😊

kingspidor, 2021-07-06:
Linux Mint 17.3 (will be 20x soon) is the only one left that doesn’t work at console; K40 whisperer runs on it, but won’t communicate with laser due to an issue with pyusb and python3.  I’ve failed to run meerk40t via SSH or VNC yet, on any system. Runs near perfect on the Pi400 though.

tatarize, 2021-07-07:
Rather than SSH or VNC you could also run it networked. You run the pi on the hooked to the laser, and run the `lhyserver` on it in command line mode. Then you can have a little PI-zero control the laser and send the PI-zero the data from a copy of MeerK40t with a networked device.

When 0.7.0 gets finished up and we do some documentation stuff I sort of need to add that info into guide somewhere. I think it's actually pretty great, it's how Klipper works and the M2 Nano is a lot like Klipper.

kingspidor, 2021-07-09:
I’m goint to try and figure this out.

Could it work with Docker? I have the Makermade M2 cnc router (it has Arduino due) set up with docker so it can be controlled via web browser from any device on network.
https://forums.maslowcnc.com/t/webcontrol-setup-on-raspberry-pi/12473
https://github.com/webcontrolcnc/webcontrol
If Meerk40t could do something like that, it would be.. just wow..

tatarize, 2021-08-12:
@tiger12506 Did do a bit of a Flask plugin. Not sure if the original issue got resolved here.

kingspidor, 2021-08-13:
Somewhat. I use a pi4b(8gb) now, but I still have to open whisperer and initialize with meerk40t open, then close whisperer in order for meerk40t to connect to the controller.

Is there a way to integrate with docker like Makerverse does? Meerk40t could be controlled via webpage then, that would be awesome.