meerk40t/meerk40t #450: SVG curves bend wrong when opening in Meerk40t

bergie5737, 2021-06-27:
Meerk40t 0.7.0 RC-6

Hi. I have a file that was separate blocks that shares a common border. Meerk40t cut those borders twice and the gap then becomes too big and not a tight fit anymore. So I used Inkscape to  "Union" it. This what it looks like. (The red shapes used to be separate areas and I removed the raster to give a clearer picture here.)
[image]

Once I import this into Meerk40t, it looks like this.
[image]
As you can see the curves now bend in a weird way.  When I open the file in third party apps, it looks the same as in Inkscape. This issue looks the same in Meerk40t 0.6##

I cannot resolve this problem and also tried "Object to path" and "Combine path" with no answer.
[calendar ouma one object to path.zip](https://github.com/meerk40t/meerk40t/files/6721587/calendar.ouma.one.object.to.path.zip)

Sophist-UK, 2021-06-27:
This svg is full of empty paths - and groups consisting of a single path - but it seems unlikely that this tis the cause.

The red cuts are in a single SVG combined path (layer 2 copy/g1175/path1134). So since this is already a combined path, object to path will not do anything, and nor will combine paths. 

When I went into Inkscape and resaved the file as a different filename, the same errors occurred, but when I removed all groups and other objects, and left just  path1134 and saved it, the issues went away.

So it appears to me that this is due to something odd happening in the SVG file rather than something in MeerK40t. It appears that the Bezier curve parameters for the path segments for some of the corners are somehow incorrect.

Here is the resaved SVG file with just the one object. Comparing the XML for path 1134 in both files might highlight something.
[calendar ouma one object to path resaved.zip](https://github.com/meerk40t/meerk40t/files/6721633/calendar.ouma.one.object.to.path.resaved.zip)

Sophist-UK, 2021-06-27:
Ok - I removed all but the path and the group it was in in order to keep the group transformation matrix, and had two XML files which were identical for both group and path - yet the original displayed incorrectly, and the simplified one displayed fine.

So contrary to my previous view, it does seem to be an issue in MK. Here is the new version so you can compare the XML for g1175 and path1134.

[calendar ouma one object to path resaved.zip](https://github.com/meerk40t/meerk40t/files/6721654/calendar.ouma.one.object.to.path.resaved.zip)

bergie5737, 2021-06-28:
Thank you for the responses and the modified file. I understand very little of how Inkscape work and is completely unsure how a good XML file should look like. I reckon your post is more for the developers. The SVG I attached was after the combined path, i.e. as far as I know how to fix it. The other option I have is to ungroup and remove "double" borders manually. The initial file was drawn as single lines, but I wanted the completed puzzle for a specific day to show a birthday wish, so I had to move puzzle pieces around. I thus made separate pieces and put them where I want them for the completed file. And that is where things went South for me! After I had multiple passes cutting the same line, I tried to combine it as one piece.

bergie5737, 2021-06-28:
Just an update. I used this file to cut, but it still does "double passes" on the lines. I guess I will have to manually edit the file again  so that there is only one line to cross like the original. :-(

tatarize, 2021-06-28:
This looks like it broken. Like the shapes are bad at the exact same shape which tends to strongly imply bug in either MeerK40t or svgelements.

Sophist-UK, 2021-06-28:
Double passes is a whole different issue. Break Apart path1134 and examine the separate segments to see if there are duplicates.

Also, you may need to check that what looks like a line isn't really a very thin closed loop with fill as these also cut double.

tatarize, 2021-06-28:
Yeah, but this displays differently in Meerk40t than it does in Inkscape and is prima facie a bug.

Sophist-UK, 2021-06-28:
@tatarize Yes - I agree. As I said previously:

> So contrary to my previous view, it does seem to be an issue in MK. Here is the new version so you can compare the XML for g1175 and path1134.

I was responding to @bergie5737 comment about double cuts - which is a separate issue entirely, with a separate cause.

Sophist-UK, 2021-06-28:
> Thank you for the responses and the modified file. I understand very little of how Inkscape work and is completely unsure how a good XML file should look like. I reckon your post is more for the developers. 

Yes - users should not need to look in the XML file. 😃 But to prove it was an issue in MK and not a quirk inside the XML created by Inkscape, I looked at the XML and then used Inkscape to simplify the file as far as possible to having the same XML for the g1175 & path1134. So we have your original file and the second version of my modified file which have **identical** XML for these objects, and yet one displays correctly in MK and the other doesn't - so it is something in the rest of the XML which is triggering this, and hopefully this will help @tatarize narrow down the cause and fix it.

> The SVG I attached was after the combined path, i.e. as far as I know how to fix it. The other option I have is to ungroup and remove "double" borders manually. The initial file was drawn as single [...]

I understand where you are coming from.

Combining multiple paths into a single path just concatenates the segments together. A path segment string starts with an M, so a combined path simply has several Ms rather than just one. It does **not** remove double paths. You need to break apart the original shapes into the constituent line segments, and then remove duplicates and then (for neatness and / or performance) combine them into a single path again. (Or go into the paths where the segments are duplicated and delete the individual line-segments. I had to do this myself yesterday with two rectangles that shared a common side - had to use Inkscape functionality to break apart the path at one of the ends of the line segment I wanted to remove before removing it so that Inkscape didn't move the two ends that resulted to a new common (pseudo-random) point in order to preserve the path as closed.)

And as previously stated, double cuts can be because you have multiple line paths that are superimposed **OR** paths that appear to be lines but are actually very very thin closed loops.

I have previously suggested to @tatarize that MK should have an optimisation to deduplicate superimposed line segments, and reduce thin closed loops that appear to be lines so that they actually are lines, but this is still a discussion topic rather than actual functionality. (Also to have options to close paths that look closed but actually aren't - closed paths end with a Z in XML - and to deal both visually and during cuts with SVG end-caps that make paths look closed when they aren't. But these functions are probably less important than deduplicating cuts.)

Sophist-UK, 2021-06-28:
As an aside, https://deepnest.io/ has software that does pretty much exactly what you have tried to do manually - and which deduplicates cuts. Worth a look.

tatarize, 2021-06-28:
Deduping cuts is somewhat planned but there's literally no work done on it and I'm pretty sure it would require some tolerances, etc. It certainly wouldn't get added in the next version in either case.

Sophist-UK, 2021-06-28:
I am absolutely certain it would require some tolerances. And I am not personally expecting it in v0.7.0.

tatarize, 2021-07-15:
I finally did the check on this and it's a real life example of negative-negative radii for arcs. Which draws them as swoops rather than normal. Since lightburn fixed their bug, I'll remove strict and go back to treating the arcs as regarded in the spec.

tatarize, 2021-07-15:
Turned out to be a bit more basic than that. Fixed in next version:

https://github.com/meerk40t/meerk40t/commit/45bb6bcacdcf90ca9006dfbef32be0f0c24fc48c

closed 2021-07-17 by tatarize