meerk40t/meerk40t #452: Run from source RC6/7 testing as at commit 83fe922a977e6519866612b235dce6e644d81850

Sophist-UK, 2021-07-13:
# Bugs
1. Operation titles in tree are not updated when e.g. you Disable, change Operation Passes or speeds.
2. Crash in Cut Inner First with all optimize options set: TypeError: object of type RasterCut has no len() in planner.py at line 1087.
3. (Fix > 0.7.0) Simulate still shows spurious travel moves when Optimising is on - starts cut from a different segment, then moves to the original start point before moving to the next cut.
4. (Fix > 0.7.0) Simulate should start from Home.
5. Raster operation with steps > 1 shows correctly in main scene, but reduced in simulation scene - and reduction in distances and time is scaled by step ^ 2 rather than by step.
6. Right-click Operation Disable does not work on Raster Operation with steps > 1 though you can uncheck the enabled button in the Operation properties window.
7. (Minor) Optimisation - Assuming that it makes no difference to time, preference should always be given to cutting in original rather than reversed direction.
8. (Minor) Optimisation - Assuming that it makes no difference to time, preference should always be given to cutting paths continuously rather than going elsewhere and returning to the path later.
9. Still cannot right click and execute a disabled Operation.
10. We probably need an ability to interrupt / cancel a long running optimisation (either for Execute or Simulation).

# UI
1. (Fix > 0.7.0) Simulate distances make sense for vector operations, but probably not for raster operations. Raster operations should perhaps be measured by area. If we stick with distances, large distances resulting from raster ops e.g. 995242.51mm are difficult to make sense of and need to be more user friendly. 
2. (Fix > 0.7.0) Need to be able to change Execution Optimisation etc. options from the simulate window and replan/show new simulation. At present you have to exit simulation, open Execute, change options, then reopen simulate.
3. (Fix > 0.7.0) Simulation needs to be closer to real-time.

tatarize, 2021-07-16:
1. Operation titles in tree are not updated when e.g. you Disable, change Operation Passes or speeds.
- Fixed

2. Crash in Cut Inner First with all optimize options set: TypeError: object of type RasterCut has no len() in planner.py at line 1087.
- Fixed.

3. (Fix > 0.7.0) Simulate still shows spurious travel moves when Optimising is on - starts cut from a different segment, then moves to the original start point before moving to the next cut.
- Can't find examples of this.

4. (Fix > 0.7.0) Simulate should start from Home (at least if auto Before/Home is set - and perhaps always).
- Ok. Fixed.

5. Raster operation with steps > 1 shows correctly in main scene, but reduced in simulation scene - and reduction in distances and time is scaled by step ^ 2 rather than by step.
- Interesting. Can't repeat.

6. Right-click Operation Disable does not work on Raster Operation with steps > 1 though you can uncheck the enabled button in the Operation properties window.
- Shouldn't have worked on any operation, since it would never show up as having worked.

7. (Minor) Optimisation - Assuming that it makes no difference to time, preference should always be given to cutting in original rather than reversed direction.

I think this is the preference. It might fall to minor rounding errors but the forward should win over the reverse all things being equal.

8. (Minor) Optimisation - Assuming that it makes no difference to time, preference should always be given to cutting paths continuously rather than going elsewhere and returning to the path later. (Though not sure whether I have experienced this not happening.)
- Here the preference is to the earliest value rather than maintaining the original order. Not sure there's an easy way to prefer this, or rather not sure if there's a way to do this without giving it a little bit of a metric. Like you get a couple bonus points if you opt for things in the original order.

9. Still cannot right click and execute a disabled Operation.
- Fixed.
 
10. We probably need an ability to interrupt / cancel a long running optimisation (either for Execute or Simulation).
- Likely push to later version.

11. XML entities with CSS display:none are still being loaded.
- This didn't happen in my test unless display none was instantly overwritten by a different more direct value for display.

tatarize, 2021-07-16:
1. (Fix > 0.7.0) Simulate distances make sense for vector operations, but probably not for raster operations. Raster operations should perhaps be measured by area. If we stick with distances, large distances resulting from raster ops e.g. 995242.51mm are difficult to make sense of and need to be more user friendly.
- I could format the value more, but there's other internationalization problems.

2. (Fix > 0.7.0) Need to be able to change Execution Optimisation etc. options from the simulate window and replan/show new simulation. At present you have to exit simulation, open Execute, change options, then reopen simulate.
- Entirely accurate but likely should be pushed for a bit. The two things for execution and simulation themselves likey need to be combined.

3. (Fix > 0.7.0) Simulation needs to be closer to real-time.
- I would need better data here. Joe has some good data on the timing of things, but also realtime simulation would require moving smaller amounts based on the distance being traveled. It's a weird can of worms. This likely could be done converting to raw mode or something, but I dunno the best ways to do that stuff.

4. (Fix > 0.7.0) Simulation - Can we show Second and subsequent Operation Passes using slightly thicker lines to make it clearer where cuts have been repeated?
- I'll try setting the simulation alpha lines to 127. So they will show up darker on repeated passes.

5. Settings - Default Operation Empty should be renamed Default Operation Red/Blue/Other and Tooltip should perhaps refer to K40 Whisperer compatibility.
- Fixed. No Whisperer reference though.

6. (Fix > 0.7.0) Device Config - Can we have settings for maximum raster X and Y speeds and maximum vector speeds?
- This seems like it would be so rarely used it's best as a hidden feature. Also, I didn't test it that much. `rapid_override`? There's also a couple settings.  Unless you mean in general in which case `window toggle -d AccelerationChart` is much closer to the right place to put that.

7. (Fix > 0.7.0) Can we show disabled operations in the main scene as e.g. muted colours or finely dotted lines or similar?
- Operations are operations they don't exist in the scene. Some elements exist in operations but that's little to do with the operations themselves.

Sophist-UK, 2021-07-16:
# UI
1. Execute Job - Merge options need overheating warnings in tooltips e.g. 

For Merge Operations: "Operations will be merged before optimisation so that e.g. raster burns may be interspersed with vector burns. If you have Duplicated Operations (with identical paths), please be aware that these paths may be burned in rapid succession and this may cause much higher material temperatures than if operations were executed sequentially."  

For Merge Passes: "Rather than executing one pass followed by a second separate pass, this allows e.g. the same path to be burned repeatedly in rapid succession, and this may cause much higher material temperatures than if operations were executed sequentially."

closed 2021-08-12 by tatarize