meerk40t/meerk40t #453: v0.7.0 Release Candidate 7: Testing... Report any issues found here

Sophist-UK, 2021-07-16:
# PLEASE DO ALL FURTHER TESTING ON v0.7.0b32 RC-7
The latest beta release is v0.7.0beta32 Release Candidate 7. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

https://github.com/meerk40t/meerk40t/releases/tag/0.7.0-beta32

# CAUTION
**CAUTION:** Please note that (depending on your design) if you use Merge Operations after you have used Duplicate Operation, or you use Merge Passes when you have set Operation Passes > 1, these can result in the laser beam burning the same path several times in quick succession without allowing time for the material to cool down in between. This can have unintended consequences. **Please do not use this feature of the beta whilst your K40 is unattended.**

Sophist-UK, 2021-07-16:
# UI (minor)
1. Simulation - starts at home but travel move to first burn point is not shown. (e.g. Propriedade Privada)
2. Simulation - if you move the playback point slider back to the very beginning (fully to the left) then you get all travel moves shown (but no burns) - an edge case? (e.g. Propriedade Privada)
3. Simulation - raster operation steps > 1 object is shown correct size in main scene but reduced size in simulation scene - and distance / time estimates are e.g. reduced by n^2 (fewer lines, shorter lines) rather than n (just fewer lines of same length). (e.g. Propriedade Privada)

gsboylan, 2021-07-21:
Bug - Crash dialog due to a NPE when doing any operation on path 864 or 866 in the attached svg. Optimizing in the execute workflow also hits this issue, so I can't cut this file
[open.zip](https://github.com/meerk40t/meerk40t/files/6859206/open.zip)

tatarize, 2021-07-22:
It's possible because after the hasAttr it gets replaced with it with the subobject which might be None. Just because it has that attribute doesn't mean the attribute is correct.

inner_path = inner_path.path

is a fine line of code, unless inner_path.path is None.

```python
if hasattr(inner_path, "path") and inner_path.path is not None:
        inner_path = inner_path.path
```

https://github.com/meerk40t/meerk40t/commit/74924a56859804b0c106a81be0ef43904f9df466

I guess the crash is less rare than I thought. I'll try to deal with the known stuff and do a faster RC8.

gsboylan, 2021-07-22:
Aha - helps if I'm looking at the right branch :)

Thanks for the quick turnaround.

Sophist-UK, 2021-07-25:
I had the same error on two different SVGs today. I send @tatarize the simpler one by email.

Sophist-UK, 2021-07-25:
# UI
Couldn't understand why laser seemed very short on power yesterday - water was a little warmer than it should be but it seemed to have very little power at all. Turned out that I had set 100ppi instead of 1000ppi. UI is not helpful here!! Can we:
1. Do something to flag up when someone sets PPI of 100 or less - usually only for dotted lines - like an "Are you sure?"; and / or
2. Format the 1,000 with a comma so that it is immediately different at a glance from 100; and / or
3. Show the PPI as a % in the Operations Tree (not sure about this one - I think it might need to be in addition to the PPI and space is limited); and / or
4. Show a different colour background for the operation in the tree if PPI is 100 or less.

gsboylan, 2021-07-25:
> 
> 
> I had the same error on two different SVGs today. I sent @tatarize the simpler one by email.

74924a5 does fix that bug.

gsboylan, 2021-07-26:
> Couldn't understand why laser seemed very short on power yesterday - water was a little warmer than it should be but it seemed to have very little power at all. Turned out that I had set 100ppi instead of 1000ppi. UI is not helpful here!! Can we:  
> 1. Do something to flag up when someone sets PPI of 100 or less - usually only for dotted lines - like an "Are you sure?"; and / or
> 1. Format the 1,000 with a comma so that it is immediately different at a glance from 100; and / or
> 1. Show the PPI as a % in the Operations Tree (not sure about this one - I think it might need to be in addition to the PPI and space is limited); and / or
> 1. Show a different colour background for the operation in the tree if PPI is 100 or less.

What about a tooltip update and a warning symbol next to the field label? #461

Sophist-UK, 2021-07-26:
> 100 ppi is a fine ppi to have it at. 

I agree - providing that your choice was a deliberate one rather than a typo.

> It needs something more obvious while also not being a flashing red warning as if it's somehow bad or wrong.

Agreed - but if it is a mistake, it needs things to draw attention to it missing a digit.

tatarize, 2021-07-28:
I added gsboylan's update. It might be effective. Not sure how universal the ⚠️ symbol is though. But, the change is generally harmless.

Sophist-UK, 2021-07-28:
# Bugs
1. IMO every element loaded from a file should be put into an operation - I have no idea why the paths in slightly different red colours are correctly put into operations, but paths in black aren't put into an operation when they are loaded. I think this is a (non-urgent) bug that should probably be addressed at some point soonish.
2. Widening the left pane with the tree in it beyond a certain point - the tree stays same width and pane gets wider.
3. Default path descriptions (when XML does not contain an id or name) - can we shorten the string by reducing the displayed resolution to e.g. 2 decimal places?
4. Default group descriptions (when XML does not contain an id or name) - I am really not sure what the default group names are trying to tell me - could we e.g. just put the coordinates of the top left corner and the width/height (matching the size shown in the scene when you select the group). To avoid confusion if you resized the group, perhaps it could say `Initial: Position (x,y), Size (w,h)`?

gsboylan, 2021-07-28:
I'm no longer able to import DXF files - is that intentional? Loading DXFs directly out of Fusion 360 was often part of my workflow in spite of the lack of color coding

tatarize, 2021-07-31:
DXF breaking is not intentional. Might be the ezdxf install failing... I'll check.

tatarize, 2021-07-31:
```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\laserrender.py", line 93, in render
AttributeError: 'Node' object has no attribute 'draw'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3693, in specific
  File "meerk40t\core\elements.py", line 1459, in inner
  File "meerk40t\core\elements.py", line 3504, in compile_and_simulate
  File "meerk40t\kernel.py", line 134, in __call__
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\core\planner.py", line 710, in plan_validate
  File "meerk40t\core\planner.py", line 70, in execute
  File "meerk40t\core\planner.py", line 205, in make_image
  File "meerk40t\core\planner.py", line 194, in make_image_for_op
  File "meerk40t\gui\laserrender.py", line 471, in make_raster
  File "meerk40t\gui\laserrender.py", line 108, in render
  File "meerk40t\gui\laserrender.py", line 404, in draw_image_node
  File "meerk40t\gui\laserrender.py", line 516, in make_thumbnail
RuntimeError: Failed to gain raw access to bitmap data.
```
No idea.

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\laserrender.py", line 93, in render
AttributeError: 'Node' object has no attribute 'draw'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 1149, in open_simulator
  File "meerk40t\kernel.py", line 134, in __call__
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\core\planner.py", line 710, in plan_validate
  File "meerk40t\core\planner.py", line 70, in execute
  File "meerk40t\core\planner.py", line 205, in make_image
  File "meerk40t\core\planner.py", line 194, in make_image_for_op
  File "meerk40t\gui\laserrender.py", line 471, in make_raster
  File "meerk40t\gui\laserrender.py", line 108, in render
  File "meerk40t\gui\laserrender.py", line 404, in draw_image_node
  File "meerk40t\gui\laserrender.py", line 515, in make_thumbnail
  File "PIL\Image.py", line 744, in tobytes
MemoryError
```
X3
Pretty sure this is *giant*.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 4170, in window
KeyError: 'window/ExecuteJob'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 1131, in <lambda>
  File "meerk40t\kernel.py", line 138, in __call__
  File "meerk40t\kernel.py", line 1757, in console
  File "meerk40t\kernel.py", line 1843, in _console_parse
  File "meerk40t\kernel.py", line 1199, in inner
[... 210 lines cut ...]

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1450, in run
  File "meerk40t\core\drivers.py", line 122, in _driver_threaded
  File "meerk40t\core\drivers.py", line 133, in _process_spooled_item
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 812, in plotplanner_process
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 989, in goto_octent_abs
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 1006, in goto_octent
ValueError: Not a valid diagonal or orthogonal movement. (dx=-1, dy=2)
```
X2 - Bane of my existence. From hell's heart I stab at thee!

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "C:\Users\...\git\meerk40t\meerk40t\gui\operationproperty.py", line 622, in on_text_power
    self.__do_layout()
  File "C:\Users\...\git\meerk40t\meerk40t\gui\operationproperty.py", line 370, in __do_layout
    layer_sizer.Add(self.button_layer_color, 0, 0, 0)
wx._core.wxAssertionError: C++ assertion ""!m_containingSizer"" failed at ..\..\src\common\wincmn.cpp(2471) in wxWindowBase::SetContainingSizer(): Adding a window already in a sizer, detach it first!
```

Ran from git. Might be altered dunno.

Sophist-UK, 2021-07-31:
How about a warning message if you are about to load a large image saying that you may experience out-of-memory errors?

Sophist-UK, 2021-07-31:
# UI
1. Are the following no longer valid menu items: View/Hide Tree

Sophist-UK, 2021-08-01:
# Optimisation
1. As previously reported direction of cuts/engraves is random when it should prefer direction in the SVG all other things being equal. In addition, we either need an option to force closed paths to be cut in one continuous cut or do so automatically. If you have multiple passes where cuts are made in different directions, then any sloppiness in the belts etc. can easily cause cuts to be slightly out.
2.  An optimisation optimisation - on multi-pass operations with Merge Passes **not** set, once you have optimised pass 1, if pass 2 is the same as pass 1 (i.e. merge operations off or has same merged operations, then we can repeat the first optimisation. (Remember, to avoid lasering the same path in quick succession e.g. at the end of pass 1 and the beginning of pass 2, we should do a virtual home after each pass anyway, in which case the optimisation in the above case should be identical anyway).

Sophist-UK, 2021-08-01:
# Bug
1. Execute Job - set all optimise boxes, clicked until it said Optimise in the big button, then cleared all optimise boxes and clicked Optimise and I got the Cut-Inside crash. These should be greyed out once their values have been used and you cannot change them.

tatarize, 2021-08-01:
```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3693, in specific
  File "meerk40t\core\elements.py", line 1459, in inner
  File "meerk40t\core\elements.py", line 3784, in make_raster_image
  File "meerk40t\gui\laserrender.py", line 462, in make_raster
wx._core.wxAssertionError: C++ assertion "bmp.IsOk()" failed at ..\..\src\msw\graphics.cpp(2575) in wxGDIPlusRenderer::CreateContext(): Should select a bitmap before creating wxGCDC
```

I dunno.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1450, in run
  File "meerk40t\core\drivers.py", line 122, in _driver_threaded
  File "meerk40t\core\drivers.py", line 146, in _process_spooled_item
  File "meerk40t\core\drivers.py", line 212, in command
  File "meerk40t\device\grbl\grbldevice.py", line 178, in move
  File "meerk40t\device\moshi\moshidevice.py", line 1001, in write
TypeError: can't concat str to bytearray
```

100% this is a total bug, or something with python 2.7 which has some compatibility stuff.

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1460, in run
  File "meerk40t\core\output.py", line 121, in _sending
  File "meerk40t\core\output.py", line 87, in connect
TimeoutError: [WinError 10060] Ein Verbindungsversuch ist fehlgeschlagen, da die Gegenstelle nach einer bestimmten Zeitspanne nicht richtig reagiert hat, oder die hergestellte Verbindung war fehlerhaft, da der verbundene Host nicht reagiert hat
```

`A connection attempt failed because the remote station did not react correctly after a certain period of time, or the connection established was faulty because the connected host did not react` Dunno. Might need catching correctly, not sure why it messed up here. Seems like timeouts should be dealt with internally.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "wx\core.py", line 3407, in <lambda>
  File "meerk40t\kernel.py", line 1688, in process_queue
  File "meerk40t\gui\wxmeerk40t.py", line 2200, in on_element_update
  File "meerk40t\gui\wxmeerk40t.py", line 3311, in on_element_update
  File "meerk40t\gui\wxmeerk40t.py", line 3501, in update_name
TypeError: TreeCtrl.SetItemText(): argument 1 has unexpected type 'NoneType
```

I think this was fixed in RC7 already.

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3693, in specific
  File "meerk40t\core\elements.py", line 1459, in inner
  File "meerk40t\core\elements.py", line 3424, in group_elements
  File "meerk40t\core\elements.py", line 405, in add
  File "meerk40t\core\elements.py", line 703, in __init__
AttributeError: 'NoneType' object has no attribute 'node'
```

X11. This shows up a lot.  Trying to use Group. I fixed this already.

Sophist-UK, 2021-08-01:
We should IMO probably cease support for Py2.

tatarize, 2021-08-01:
We never supported it.

It's just that you could maybe try running the code in py2 and it would do somethings that would generally error out. We might actually want to put something in the CLI that says Python2 is not supported if it's attempted.

joerlane, 2021-08-02:
_We should IMO probably cease support for Py2._

I asked for this a few times in the 0.6.xx era. Plan 9 build would happen with a 2.7 python version. Who cares? Not toooooo many; but MANY old operating systems support 2.7 out of the box.. Almost all users have a default python for pip and python commands that point to 2.7 still. Legacy for sure.

tatarize, 2021-08-02:
End of life for py2 was Jan 1. 2020.  I wrote some python 2/3 code before... not a fan.

joerlane, 2021-08-02:
Yeah; but 2020 was a bitch. Think about it...

Sophist-UK, 2021-08-03:
> We never supported it.
> 
> It's just that you could maybe try running the code in py2 and it would do somethings that would generally error out. We might actually want to put something in the CLI that says Python2 is not supported if it's attempted.

If you ever spend any time on Python2 specific stuff, you are de-facto supporting it. And users will gripe if something doesn't work under Py2.

Most platforms will run executables anyway. Anyone running from source on a non-executable platform will have the skills to install Python 3 and pip dependency management should be able to handle everything else.

I think it would be a good idea to limit v0.7.0 to Py3, to make changes to prevent it running on Py2, and make further changes to requirements files to ensure that all prerequisites for all functions are installed as standard - if someone doesn't want these because disk space is so limited and e.g. they don't have a camera, then they can edit the requirements file or manually install pre-requisites.

tatarize, 2021-08-03:
Nah, the point would be to catch that they are trying to run under python 2 in the code. And say that it doesn't run under python 2.

tatarize, 2021-08-03:
Here done.

https://github.com/meerk40t/meerk40t/commit/31dc2357fed80dfc1597406ca0a89f5d72e8e6aa

If you try to run in python versions less than 3.5 it'll just reject it.

tatarize, 2021-08-03:
I dunno if 3.5 even works. We might be 3.6+

tatarize, 2021-08-04:
>How about a warning message if you are about to load a large image saying that you may experience out-of-memory errors?

I might be able to catch them and do avoid some of the fatter image modifications. If you load the image in black and white, which is 2 bits I could avoid conversions to RGBA which is 16 times the size or making two copies of the image-data. Or at least allow for a workflow dealing with them. Apparently people are trying to do giant images.

> Are the following no longer valid menu items: View/Hide Tree

This only kicks in when refreshed. Toggle it then hit F5 ("refresh" in console). It's not really needed anymore since the tree upgrade makes things faster and you can literally just close the tree down if you want.

> Panes menu has "Navigate" when use elsewhere is "Navigation"

Ok. Fixed.

> As previously reported direction of cuts/engraves is random when it should prefer direction in the SVG all other things being equal.

I checked the code and the original direction *is* preferred all things being equal. Though it might be some very small rounding error that messes the stuff up.

> In addition, we need A) an option to enforce this, and B) either an option to force closed paths to be cut in one continuous cut or do so automatically. If you have multiple passes where cuts are made in different directions, then any sloppiness in the belts etc. can easily cause cuts to be slightly out.

I think this might be best to turn off optimization in that case. Though it does seem like it it would be possible to do this, even if the start location can vary the rest of a closed shape could just be run normally.

> An optimisation optimisation - on multi-pass operations with Merge Passes not set, once you have optimised pass 1, if pass 2 is the same as pass 1 (i.e. merge operations off or has same merged [...]

The first path is sometimes different than the second. But the second is the same as the third. Since it goes nearest location first. In theory you could have some oddness to the stuff.

> Reticule is currently a small red circle. Any chance we can change the colour to something other than the Red/Blue/Black that are commonly used in SVG files? Mid-Brown perhaps?

The reticules actually go through a sequence of colors depending on how many of them there are. If you are running two different lasers you get two of them and they are different colors. If you run in console "lhyemulator" you get a reticle that tracks the data sent information which will be at a different location than the red one, but also importantly a different color.

> Execute Job - set all optimise boxes, clicked until it said Optimise in the big button, then cleared all optimise boxes and clicked Optimise and I got the Cut-Inside crash. These should be greyed out once their values have been used and you cannot change them.

Failed to replicate.

> Blue Laserpath lines not being drawn except from end-position back to Home.

Actually they are just being drawn for the very last item and nothing else. Which you'd see mostly on a re-home.

Ok. Fixed.

> Getting mock USB errors (maybe) when using Mock USB Connection Mode in controller. Disconnect button in controller window says "Force Continue".

Yeah, the mock USB does not automatically exit the wait loop at the end. It should throw out some 237 codes and get out of there here and there. But, it's an issue with the mock controller not actually doing any work. It doesn't have any *real* job to complete at that place. Force continue lets you skip the wait. It used to be more helpful when there was a hang-job bug. But, it's mostly fine now. You should only push it if the laser is somehow done but the controller didn't see it. This could happen if you unplugged the USB at the exact right time I think.

> When using Mock USB, a LOT of "Using Mock USB" messages are sent to the Controller / USB logging channels. Any chance we can do this only when the Controller starts or when Mock mode is turned on or off?

Nah. People shouldn't be using mock anyway. I should almost hide it unless you have `developer_mode` set to True.

Sophist-UK, 2021-08-04:
> I might be able to catch them and do avoid some of the fatter image modifications. If you load the image in black and white, which is 2 bits I could avoid conversions to RGBA which is 16 times the [...]

There is little point in loading any image which has more pixels in X or Y than the bed-size in mils. But that is still a big image.

Is there actually any point in doing any image manipulation in RGBA when the laser is monochrome anyway - would it help to convert all images to 16-bit grey-scale on load (which would save some memory)?

> > Are the following no longer valid menu items: View/Hide Tree
> 
> This only kicks in when refreshed. Toggle it then hit F5 ("refresh" in console). It's not really needed anymore since the tree upgrade makes things faster and you can literally just close the tree down if you want.

I suggest we eliminate this menu since Tree is a pane that can be closed anyway.

> > As previously reported direction of cuts/engraves is random when it should prefer direction in the SVG all other things being equal.
> 
> I checked the code and the original direction _is_ preferred all things being equal. Though it might be some very small rounding error that messes the stuff up.

As I understand it, greedy works by selecting the closest next start / end point for a path segment. I am guessing that the current algorithm should choose a segment start point if there are both segment start and end points with identical coordinates. I am also guessing that by the point of optimisation you have already converted SVG coordinates (decimal) to mil coordinates (integer), and that you compare distances using (Pythagorean) sum of squares, rather than SQRT(sum of squares). 

Given a closed path (and in my example it is a closed path consisting of 4 quadrants of a circle) there should be 2 segments one starting and one ending at **exactly** the same coordinates - so I am unclear how rounding errors could creep in, but I will take your word for it. (If you want to point me at the source code file, I would be happy to double check the code on this.)

Obviously for lower quality SVG files, where what appears to the naked eye from a distance to be a closed path is actually a compound path or multiple paths, then this might not be the case.

If we have a start/end point of a next segment which is identical to the finish point of the previous burn segment, then we should definitely burn that next (favouring segments with an identical start point), because that is a continuous burn, whereas if the point is not identical then there is a travel.

Otherwise, if there is no continuous burn and we are going to do a travel, it seems to me that e.g. any point within (say) 20 or 50 mil of the closest next point will take an indistinguishably same amount of time to travel to. So if you can do this without a material increase of optimisation time, then perhaps creating a list of segments whose start or end points are within a short distance (say 20 or 50 mil) of the closest start/end point then I would suggest the following approach to selecting within this shortlist:

1. If there is at least one segment with a start point within this 20-50mil, then for step two only use segments which match this, otherwise use only the segments with an end point within this 20-50mil (which - obviously - is the whole shortlist).

2. Of the remaining shortlisted segments, choose the one with the shortest path (as an attempt to keep the laser local and complete local burns before moving to a different area entirely, the idea being trying to avoid later needing a long travel move to come back to finish this area. (Clearly there will be counter examples, but greedy is already only a best guess, and I have a hunch that this might on average provide a better greedy solution.)

> I think this might be best to turn off optimization in that case. Though it does seem like it it would be possible to do this, even if the start location can vary the rest of a closed shape could [...]

I certainly agree that if the algorithm genuinely favours doing paths in the stated direction anyway, then A. will be far less important. (With A. I am really thinking about people whose K40 has e.g. loose belts and where direction of cut in Pass 1 needs to be the same as in Pass 2.) B. is probably also unnecessary if the algorithm works because it will already favour continuous cuts because the next segment will be zero distance from the current position. So, I agree, probably best to forget these.

> The first path is sometimes different than the second. But the second is the same as the third. Since it goes nearest location first. In theory you could have some oddness to the stuff.

As I have said before, if you have Merge Passes **un**checked, then you need to try to ensure that the laser doesn't start from the end of Pass 1 and then do Pass 2 in the exact opposite way which would result in the same path being burned twice in quick succession which is a fire risk. (If I don't mind that risk, I would check Merge Passes.) So if you don't have Merge Passes set then I think you need to start the optimisation for the next pass using Home as a start point and not the end of the previous pass. (YOu don't physically need to do a Home, or travel to Home, just travel to the first burn point calculated on an assumption that you are starting the next pass from Home.) So **because all passes are calculated as starting from home** when you finish the first optimised set [... cut ...]
> > Blue Laserpath lines not being drawn except from end-position back to Home.
> 
> Actually they are just being drawn for the very last item and nothing else. Which you'd see mostly on a re-home.
> 
> Ok. Fixed.

That was exactly my experience.

> Yeah, the mock USB does not automatically exit the wait loop at the end. It should throw out some 237 codes and get out of there here and there. But, it's an issue with the mock controller not [...]

I wanted to try the Lhysemulator, but couldn't work out how to do this from the GUI. So I used Mock USB mode instead.

tatarize, 2021-08-06:
```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "mk40t.py", line 13, in <module>
  File "meerk40t\main.py", line 413, in run
  File "meerk40t\kernel.py", line 869, in bootstrap
  File "meerk40t\gui\wxmeerk40t.py", line 201, in plugin
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\gui\wxmeerk40t.py", line 4147, in window_open
  File "meerk40t\kernel.py", line 710, in <lambda>
  File "meerk40t\gui\wxmeerk40t.py", line 4140, in window_open
  File "meerk40t\kernel.py", line 481, in open
  File "meerk40t\kernel.py", line 515, in open_as
  File "meerk40t\gui\wxmeerk40t.py", line 390, in __init__
  File "meerk40t\gui\wxmeerk40t.py", line 1089, in __set_ribbonbar
  File "meerk40t\gui\icons.py", line 16, in GetBitmap
  File "wx\lib\embeddedimage.py", line 64, in GetImage
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1694) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!
```
X2 - Unfixable region issue. Should be fixed in next version of wxPython. But, the two versions of  MeerK40t are to address this.

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\laserrender.py", line 93, in render
AttributeError: 'ElemNode' object has no attribute 'draw'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "wx\core.py", line 3407, in <lambda>
  File "meerk40t\gui\scene\scene.py", line 348, in refresh_scene
  File "meerk40t\gui\scene\scene.py", line 376, in update_buffer_ui_thread
  File "meerk40t\gui\scene\scene.py", line 432, in draw
  File "meerk40t\gui\scene\scene.py", line 624, in draw
  File "meerk40t\gui\scene\scene.py", line 624, in draw
  File "meerk40t\gui\scene\scene.py", line 621, in draw
  File "meerk40t\gui\scene\scenewidgets.py", line 66, in process_draw
  File "meerk40t\gui\laserrender.py", line 108, in render
  File "meerk40t\gui\laserrender.py", line 283, in draw_shape_node
  File "meerk40t\svgelements.py", line 5354, in __init__
  File "meerk40t\svgelements.py", line 6265, in segments
TypeError: unsupported operand type(s) for +: 'float' and 'NoneType'
```
Rectangle shape failed the draw in segments because width is equal to None. Made that degenerate.

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\laserrender.py", line 93, in render
AttributeError: 'ElemNode' object has no attribute 'draw'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "wx\core.py", line 3407, in <lambda>
  File "meerk40t\gui\scene\scene.py", line 348, in refresh_scene
  File "meerk40t\gui\scene\scene.py", line 376, in update_buffer_ui_thread
  File "meerk40t\gui\scene\scene.py", line 432, in draw
  File "meerk40t\gui\scene\scene.py", line 624, in draw
  File "meerk40t\gui\scene\scene.py", line 624, in draw
  File "meerk40t\gui\scene\scene.py", line 621, in draw
  File "meerk40t\gui\scene\scenewidgets.py", line 66, in process_draw
  File "meerk40t\gui\laserrender.py", line 108, in render
  File "meerk40t\gui\laserrender.py", line 400, in draw_image_node
  File "meerk40t\gui\laserrender.py", line 516, in make_thumbnail
RuntimeError: Failed to gain raw access to bitmap data.
[... 233 lines cut ...]

Datagrammasocket, hm. Nifty, network errors.

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1460, in run
  File "meerk40t\gui\rasterwizard.py", line 189, in wiz_img
  File "meerk40t\image\imagetools.py", line 1458, in wizard_image
  File "PIL\ImageOps.py", line 152, in autocontrast
  File "PIL\ImageOps.py", line 51, in _lut
NotImplementedError: mode P support coming soon
```
Autocontrast does not work on palette images.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2797, in on_click_open
  File "meerk40t\gui\wxmeerk40t.py", line 2504, in load
  File "meerk40t\core\elements.py", line 4524, in load
  File "meerk40t\image\imagetools.py", line 1678, in load
ZeroDivisionError: float division by zero
```
Not sure.

Sophist-UK, 2021-08-06:
# [v0.7.0beta33 Release Candidate 8](https://github.com/meerk40t/meerk40t/releases/tag/0.7.0-beta33) is now available. If you have a problem to report with 0.7.0 please check whether it is still an issue in the new version and report any difficulties in the new issue #465 opened specifically for this new version.

closed 2021-08-06 by Sophist-UK