meerk40t/meerk40t #454: Trying add "Home" adds "300" instead

inspectionsbybob, 2021-07-17:
I went to add a "Home" after my last operation and the event it added was 

"300"

exited, reloaded and it did the same thing...

was trying to run a simple job, but a job, so I couldn't verify the other specials such as "Beep"

tatarize, 2021-07-17:
Hm. Weird fluke. It's the number for the command in the lasercode. Not sure why it's not plaintext.

Sophist-UK, 2021-07-17:
Which version of MK?

tatarize, 2021-07-17:
RC-7. Yeah, it does it. I changed some of the naming stuff so it went to a default name for it, which is not the .name but 300 which is the value of the `Special Command`

inspectionsbybob, 2021-07-17:
I was able to confirm ALL of the operations come over as numbers on the operations Tree,  
[image]

tatarize, 2021-08-03:
This is now fixed in RC8.

https://github.com/meerk40t/meerk40t/commit/69e8ad88fbb9582233d6a55f98f417a6002f2b2b

Should maybe be due out sometime soon.

tatarize, 2021-08-06:
Now live.

closed 2021-08-06 by tatarize