meerk40t/meerk40t #455: USB pausing lots on Raster/Image printing...

inspectionsbybob, 2021-07-17:
Tried printing a photo last night and my USB kept getting stuck...

From what I could see it was getting stuck on "busy"

No packets were lost!

I tried slowing the rate down to HALF the rate (mm/sec) and was still having issues.  

Had to continually "disconnect" the controller, it would reconnect on its own in a few seconds and just keep going, not missing a thing most of the time...

I had been using other USB devices before (another camera and printer) on the same USB a hub so it is possible that it was still looking for those devices?  I read something about USB reserving bandwidth and occasionally not releasing it...

Will be re-trying the same object today, but will disconnect all the other devices and do a cold-boot beforehand.

Any other ideas, logs to look at or other suggestions?

inspectionsbybob, 2021-07-17:
Confirmed... a full COLD BOOT seemed to fix the USB problem.
Suggest looking into the "reservation" of USB bandwidth and if sufficient isn't available either displaying a warning or forcing a re-allocation of bandwidth from idle devices if possible.  

As I wrote, no packets were lost, it just would get "STUCK" in a "busy" mode.

https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-bandwidth-allocation#:~:text=The%20system%20allocates%20bandwidth%20on%20a%20first-come-first-serve%20basis.,configure%20the%20device%20and%20fails%20to%20enumerate%20it.

tatarize, 2021-07-17:
This sounds somewhat stuttering and I don't know what a cold boot could have to do with anything. I can't get my head around that. The OS might prioritize some usb devices but the M2 Nano shouldn't need that much and should be pretty actively being used.

inspectionsbybob, 2021-07-17:
All i know is after using my computer for some other USB apps, camera and
3d printer, I had major issues.  I did a cold boot and I am running a
larger image raster print (Lithopane??) without a single pause after
10min+...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

closed 2021-08-12 by tatarize