meerk40t/meerk40t #456: Jobs won't Print, Just beeps as if Done!

inspectionsbybob, 2021-07-18:
This morning I wanted to do a "test Fit" before I printed the real thing...

Imported it, shows that it has TWO Cuts to do...
Go through the "Execute Job" stuff
Hit "Spool Job" and IMMEDIATLY get the DONE BEEPS...

Check the USB Log.... NOTHING after the connected message
HOME works... Navigation works

Tried another stock file.... same thing, soon as I hit "Spool" it gives me the DONE beeps

NOTHING PRINTS.  Doesn't even try.  Doesn't send ANYTHING to the controller.  When I use navigation I can see the commands go out... Execute doesn't do anything....

Will do a RE-Install (delete app and copy again) and see report back (have jobs on other devices I need to tend to first)
No error messages or files either....

inspectionsbybob, 2021-07-18:
Ok, I re-installed and re-imported the test file and it ran fine. So said new and brought in the real file.  It rastered the text and totally skipped the cut objects?? When I saw this tried to select the cut job and submit JUST that...
BEEP BEEP BEEP after doing nothing?
It is showing the list of objects under CUT so it isn't a color mix-match.  It is enabled...
I am befuddled and frustrated as I have something to run on the laser... and it rastered the text onto t he material so ai don't want to move it or I loose that piece...

joerlane, 2021-07-19:
Send the file; and note which options you had selected in the execute job window. The more bugs like this we can track down completely the better.

inspectionsbybob, 2021-07-19:
I just built a simple file with a 50x50 box in inkscape.  Same result.... just beeps.
Home Works, navigation works, but it won't send it to the printer...
I even went through the Registry and cleaned up all the stuff from all the previous versions, and there was a LOT of garbage left behind.

I Just Turned Off -ALL- the options that were set when I opened the program....
Reduce Travel Time
Cut Inner First

And It Worked....  Whew... only a half a day lost... Now I need to figure out why the scaling is off

Sophist-UK, 2021-07-19:
Can you please confirm this is on RC7?

Sophist-UK, 2021-07-19:
> Send the file; and note which options you had selected in the execute job window. The more bugs like this we can track down completely the better.

Yes - please save the SVG file from MK (because it saves a lot of your settings in the file), take some screen shots, and zip up the SVG file and screen shots and post it here.

inspectionsbybob, 2021-07-19:
Once I finish the project I will reset and recreate the issue and send
you the files from when it is failing...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-07-19:
I tried to re-create the issue this morning, turned on all the things, used the same file that failed...

And it worked.

Feels weird wanting it to fail, but it isn't co-operating.  Did find some other issues that I will list separately and individually.

closed 2021-07-31 by tatarize