meerk40t/meerk40t #458: Beta 32 Doesn't recognize end-stops

inspectionsbybob, 2021-07-19:
It trying to chase other problems I found another.

Beta-32 Runs into the End-stops and Keeps going... Smashing the Microswitches and making some interesting awful sounds

Tried 6.20... Home Works fine
Tried Beta-32   BRRRRRRR left Micro-switch smushed until I released steppers
Tried 6.20... behaved as expected
Tried Beta-32   BRRRRR

All I was doing was hitting the HOME button when it was ALREADY in the home position....
Homing, watching, closing the app and opening the OTHER version.  Homing and watching... Swapping back, repeat
Enclosed is the BLANK MK file so you can see settings.

[BLANK endstop.zip](https://github.com/meerk40t/meerk40t/files/6842643/BLANK.endstop.zip)

tatarize, 2021-07-22:
The endstops apply for home, the only maybe plausible thing is if you have a set home position which is negative. Then it homes to the upper left corner, then rams the wall. Typically home is just home, but with a set home position, you could have it do home them smash wall.

inspectionsbybob, 2021-07-22:
i could understand that if it didn't behave -very- differently in 6.20...

I switched between the beta and 6.20 and the HOME behavior was very
different.  I tried it several time, just to make sure, but didn't want to
hurt my belts or the limit switches by chasing the issue.   Home is set to
physical home (0,0)in both. Home SHOULD go to the limit switches and STOP,
that's why they are called limit switches.  The limit switches are supposed
to set 0,0....

Again... it worked fine in 6.20... but not in the beta...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-07-27:
Had a chance to play...and explore....
There -WAS- a negative number in the Home location... once I did a physical home, and re-set HOME it behaves....
HOWEVER, I still think the Switches should override everything for safety and longevity of belts and such...

closed 2021-07-27 by inspectionsbybob