meerk40t/meerk40t #459: (Documentation) After running from source on windows, meerk40t cannot connect to usb

gsboylan, 2021-07-22:
[image]

Followed instructions here: https://github.com/meerk40t/meerk40t/wiki/Install:-Running-from-source

pyusb and the libusb k40 driver from ScorchWorks are both installed on my machine.

tatarize, 2021-07-22:
Hm. It really should have aborted that and tried the whisperer driver if the libusb driver failed. Maybe for long term I should give you a force switch for the config thing to let you manually choose your driver. So in the future if this comes up more, it'll be fixable without too much trouble.

gsboylan, 2021-07-22:
This works fine for me with the precompiled executable - so I assumed there's some setup missing, somehow. Not 100% sure. For what it's worth, I've never installed the stock driver that came with the k40, only the scorchworks driver.

tatarize, 2021-07-22:
Yeah, when you plug it in the windows driver should load up. This is the CH341DLL driver which works fine, however, it's not loading. In that case it's *supposed* to try the LibUsb Driver which works with Whisperer. but somehow it's not doing that. It's supposed to switch between drivers depending on what works. The goal is to have it always work, but it looks like the windows driver failed and it failed in a way that prevented it from trying the other driver or simply didn't correctly attempt the fallback.

This needs to be checked before full release, I'll try deleting the dll and doing other things with it to see if I can replicate it.

gsboylan, 2021-07-22:
Let me know if you need me to test out a patch

Sophist-UK, 2021-07-22:
> Followed instructions here: https://github.com/meerk40t/meerk40t/wiki/Install:-Running-from-source

I rewrote these when I installed to run from source on my own Windows laptop. I am not sure anyone other than you has tried them since, so there may well be a problem.

The Windows shortcut that is created does **not** create a console for console errors to appear on. You might want to create a small `.cmd` file as follows which you can run:
```
@Echo On
python meerk40t.py
pause
```

@tatarize The error message `Could not find module 'CH341DLL.dll' (or one of its dependencies). Try using the full path with constructor syntax.` is really not very helpful to the user - though it may be helpful to you because it seems to be finding the dll when running from an executable but not when running from source. You might want to ask @gsboylan to check that this dll is in the expected place when running from source.

tatarize, 2021-07-22:
The error message isn't mine, it's likely python with the call to ctypes.

https://stackoverflow.com/questions/65345213/could-not-find-module-or-one-of-its-dependencies-try-using-the-full-path-with

The real issue is that when hitting that it stops trying rather than goes on to the other driver.

tatarize, 2021-07-22:
Okay, I checked and that error there is apparently a FileNotFoundError. I should have added a commit to catch that.

Solid chance this commit addresses it: https://github.com/meerk40t/meerk40t/commit/7eff28a8e3e2fed07bdc6a77e34c150ad4da33e5

gsboylan, 2021-07-23:
[image]

No good, and no logs out to console to clarify the error. Maybe something's swallowing the error?

gsboylan, 2021-07-23:
Got it - you're catching and throwing different error types. Let me put up a PR

gsboylan, 2021-07-23:
https://github.com/meerk40t/meerk40t/pull/460

tatarize, 2021-07-23:
The actual problem here is that it should throw a failed error and not work and default back to the other driver, but apparently the connection refused says the driver was fine but didn't connect. I need to check the flow since the in theory that kind of failure should result in trying the other driver rather than stopping with the connection refused. Let me take another stab at it.

tatarize, 2021-07-23:
https://github.com/meerk40t/meerk40t/commit/962ab67230e674754d5d407340707fd6cd73e301

This should maybe correct it. #459 is going to hit some serious troubles. The import errors for the driver running aren't correct. It needs to catch the failure to load and raise it as an importError.

The relevant section is:

```
        handlers = []
        try:
            from .libusb import Handler as LibUsbHandler

            handlers.append(LibUsbHandler(channel=self.channel, status=self.status))
        except ImportError:
            self.channel(_("PyUsb is not installed. Skipping."))
        try:
            from .windll import Handler as WinHandler

            handlers.append(WinHandler(channel=self.channel, status=self.status))
        except ImportError:
            self.channel(_("No Windll interfacing. Skipping."))
```

In `ch341.py` the function tries to load the windll handler and the libusbhandler and when it fails to load those correctly it's supposed to move on the next bit and fail correctly. So if I can't load the LibUsbHandler because I don't have the driver or something, then failing should fail to register the handler. Same for WinDll, either load it or fail with import. The issue here is that it failed the wrong way and didn't just say `No Windll interfacing. Skipping.` -- Which would be the expected result there.

tatarize, 2021-07-23:
The DLL CH341DLL.dll depends on KERNEL32, USER32, SETUPAPI. I'd think your windows would have *a lot* more problems if those weren't functional.

gsboylan, 2021-07-25:
I can confirm that 962ab67 fixes the issue.

gsboylan, 2021-07-28:
Okay - Actually, I was incorrect, this commit does not fix the issue. I can still repro the issue on your most recent commit 962ab67.

Sophist-UK, 2022-02-15:
@gsboylan Is this still an issue?

dorsi90, 2022-03-02:
@tiger12506 no. Platform.release() return value of 8.1 for windows 8.1 which is a floating number not integer so gives error for os version with a floating value when try to convert the string to integer

Sophist-UK, 2022-03-02:
Fix for ansi on 8.1 submitted as #872.

dorsi90, 2022-03-02:
@Sophist-UK python3 documentation says that starting from python 3.8 winmode parameter was added to ctypes.WinDLL class which has a default value of RTLD_LOCAL so i fixed it setting winmode to RTLD_GLOBAL changing the code below in windll.py
Line 1-> **from ctypes import c_byte, WinDLL**
Line 151-> self.driver = WinDLL("CH341DLL.dll",winmode=0)

With this changes worked for me on python 3.9.10 but maybe you need to add an if statement to check if the python version is  equals or greater then 3.8 since winmode does not exists on prior versions.

dorsi90, 2022-03-02:
@Sophist-UK sorry winmode has default value of None so i don't know if  it is ctypes.DEFAULT_MODE (wich is RTLD_LOCAL in windows OS) but it's certainly not RTLD_GLOBAL.
take a look https://docs.python.org/3.8/library/ctypes.html#ctypes.WinDLL

Sophist-UK, 2022-03-02:
@dorsi90 Guiseppe - I am looking at this now. Thank you for researching this for us - you first logged this in July 2021 so apologies for not having looked at this. I have tried your code but I cannot test it on my development machine which doesn't have a K40 attached.

My research suggests that you are on the right track here. But I am not sure what LoadLibraryEx flags Python uses by default, nor which specific flag is causing the problem and since the Python documentation says that these flags are for security I think we should do the minimum that fixes this by adding a full path or removing only the flag that is causing the issue.

Are you willing to do some further research and testing? (I will create a PR for this using the above code on the basis that we are going to tweak it if necessary once the research is complete.)

dorsi90, 2022-03-02:
@Sophist-UK I suggest not using the full path but keeping system variables search.
If it generate error for version prior to 3.8 because winmode not exists as parameter or just to or just to keep the old code then pathlevel.h defines PY_MAJOR_VERSION and PY_MINOR_VERSION so you can do something like this
**if PY_MAJOR_VERSION >= 3 and PY_MINOR_VERSION >=8 :
   """new code"""
else
   """old code**"""

Sophist-UK, 2022-03-02:
Ok - I found the code for WinDLL and it uses _LOAD_LIBRARY_SEARCH_DEFAULT_DIRS from module nt as the winmode.

The documentation for the flags can be found [here](https://docs.microsoft.com/en-us/windows/win32/api/libloaderapi/nf-libloaderapi-loadlibraryexw) or [here](https://docs.microsoft.com/en-us/windows/win32/api/libloaderapi/nf-libloaderapi-loadlibraryexa) (both essentially the same) and they say:
```
LOAD_LIBRARY_SEARCH_DEFAULT_DIRS (0x00001000) 

This value is a combination of LOAD_LIBRARY_SEARCH_APPLICATION_DIR, LOAD_LIBRARY_SEARCH_SYSTEM32, and LOAD_LIBRARY_SEARCH_USER_DIRS. Directories in the standard search path are not searched. This value cannot be combined with LOAD_WITH_ALTERED_SEARCH_PATH.
```

I took a look to see where CH341DLL.dll is located, and it appears that in the source it is located in /.github/workflows/win and included in the Windows executable with some clever build code. So I am not surprised that it is not found when running from source because:

a. Depending on the route you get the source, /.github is sometimes not included (i.e. Pypi/PIP install?)
b. Even if you have this directory, it is not included in the path.

So I am frankly surprised it works if you do `winmode=0` - have you moved or copied the file?

dorsi90, 2022-03-02:
It works because don't load the dll included in the .github folder but instead load the dll located in system32 which i think is installed when plug the k40 the first time.

LOAD_LIBRARY_SEARCH_DEFAULT_DIRS also search in system32 but are you sure that is the default paramenter value?

I had not seen that the dlls were also included in the github folder so at this point the full path to the github folder is the best choice as the doc suggest:
>The winmode parameter is used on Windows to specify how the library is loaded (since mode is ignored). It takes any value that is valid for the Win32 API LoadLibraryEx flags parameter. When omitted, the default is to use the flags that result in the most secure DLL load to avoiding issues such as DLL hijacking. Passing the full path to the DLL is the safest way to ensure the correct library and dependencies are loaded.

Later I do some tests

dorsi90, 2022-03-02:
@Sophist-UK to be sure i removed dll from system32 and from github folder and it also works without dll in folders so i don't know which path it use for load the ch341 dll :)
I need to write a code that print the path of the loaded dll

Sophist-UK, 2022-03-02:
You may be able to find the path using a ctype utility (see ["Finding shared libraries](https://docs.python.org/3.8/library/ctypes.html#finding-shared-libraries)]:
```
from ctypes.util import find_library
find_library("CH341DLL.dll")
```

dorsi90, 2022-03-02:
@Sophist-UK ctypes.util.find_library function returns None so can't find the dll but WinDLL("CH341DLL.dll",winmode=0) works like a charm i don't understand why

Sophist-UK, 2022-03-02:
Have you rebooted since you removed CH341DLL.dll from system32? It may still be loaded in memory and so not reloaded.

Sophist-UK, 2022-03-02:
I doubt that Python caches the file - but it is normal not to remove a DLL from memory immediately you say you want it unloaded and so if it has been loaded already, it probably stays loaded.

dorsi90, 2022-03-02:
After reboot the system is the same as you can see in picture the first line is find_library response which is None but dll is loaded.

[image]

dorsi90, 2022-03-02:
Tested again with old code **windll.LoadLibrary("CH341DLL.dll")** just to be sure and the error occurs.
[image]

dorsi90, 2022-03-03:
@Sophist-UK the dll was located in the Script folder of the virtualenv and if i delete the dll from this folder not working again.
So windll.LoadLibrary not work, WinDLL works only with winmode=0 but only if the dll is located in the Script folder and maybe in Lib folder of the virtualenv that starts the process.
I think the best solution is to add the full path as you suggested before

Sophist-UK, 2022-03-03:
1. There are only a few locations that it will find the DLL automatically. System32 is the obvious one. The scripts folder in a venv works because the python.exe is located there. Did you try putting the DLL back in system32 if it was originally there and seeing if it loaded with or without winmode=0?

2. The alternative is for MK to move it from /.github/workflow/win to somewhere like /meerk40t/extras where it will be included in Pypi installs / Download source zip/tar files / git, and adjust the windows executable build script accordingly.

dorsi90, 2022-03-03:
Yes i restored the dll in the system32 folder the problem is that winmode=0 search in system32 only if the app is located in the same partition where the OS is installed so if my OS is located in C and the meerk40t also is located in C it loads the dll as expected otherwise works only if dll is in the Script folder.

windll.LoadLibrary("CH341DLL.dll") does not work in any way.

The user friendly option is the full path but you can also use winmode=0 specifying in the documentation to place Meerk40t folder in the same partition where the OS in installed or copy CH341DLL.dll from system32 or github folder into Script folder generated by virtualenv.

Another option could be retrieve the partition where OS is installed and adding it to the searching path but i don't know how to do that.

Sophist-UK, 2022-03-03:
You will need to look at the documentation already linked to to see what flags to use. There is one for searching system32. But you first need to work out which of the flags is needed in addition the lack of which was not allowing it to be loaded at the start.

dorsi90, 2022-03-03:
Yes it say that LOAD_LIBRARY_SEARCH_DEFAULT_DIRS flag that is available in LoadLibraryEx can search in APPLICATION_DIR, USER_DIR and SYSTEM32 and this flag can also be used for WinDLL class in winmode parameter wich is perfect but doesn't work for me.
> The winmode parameter is used on Windows to specify how the library is loaded (since mode is ignored). It takes any value that is valid for the Win32 API LoadLibraryEx flags parameter. When omitted, [...]

The only parameter that for me search correctly in system32 is winmode=0 (only if application is placed anywhere in the partition where OS is installed).

Also note that winmode=0 is the default value for python 3.8 then is set to none starting from 3.9 version.

Which option do you prefer to use?
I can do some tests and if you want you can send me some code to test

dorsi90, 2022-03-03:
>import nt
self.driver = WinDLL("CH341DLL.dll", winmode=nt.LOAD_LIBRARY_SEARCH_DEFAULT_DIRS | nt.LOAD_IGNORE_CODE_AUTHZ_LEVEL)

[image]

self.driver = WinDLL("CH341DLL.dll", winmode = 0x00001000 | 0x00000010)
[image]

self.driver = WinDLL("CH341DLL.dll", winmode = 0)
[image]

All tests are made with CH341DLL.dll only present in system32 and github folders.

Sophist-UK, 2022-03-03:
Ok. I think we might now have enough information to get the driver to load providing that it is in a fixed number of places. I think that there might need to be a broader discussion about how to provide both CH341 support and LibUSB support on all platforms in both 32 and 64 bit o/s and 32/64 bit python, and when run from executables and from various types of source so we can provide a single unified experience for users.

dorsi90, 2022-03-03:
From executable i haven't encountered any problems maybe the dll is embedded in the EXE file.

However during my tests there have been several crashes not related to this that can be easily fixed. I can open new issues?

tatarize, 2022-03-04:
The DLL is totally built into the exe file. But, also it's using 32 bit python internally to make sure that stuff works, so it basically freezes a venv in the file.

Sophist-UK, 2022-03-04:
Yes. But the DLL is not available if you are running from downloaded source or pypi (can't recall which) because it is located in a directory which isn't included. And we need to provide 64 bit dll too so we can create a 64 bit version. And we should handle libusb similarly.

tatarize, 2022-03-04:
Yeah, the dll isn't really being distributed, just included at points. I'm not really sure about the 64 bit version stuff or whether that stuff is able to work. @tiger12506 might be pretty reasonable to figure such things out. Mac and Linux is still pretty consistent, and we might want to do some work on figuring out a pure 64 bit version. But basically you need the same bit driver as the python version, and it could be the case that the calls in the code itself are for the 32 bit bit python dll for the windll version. If that's the case then the dll calls would be different for 64 bit and 32 bit and so any 64 bit python would have an issue using the windows driver which could explain the issue here.

tatarize, 2022-03-06:
I think it checks libusb first, so in theory if you installed that it would work even if the windows driver thing is weird. And yeah the exec files has them built in since we didn't want to deal with intricate problems from folks who run the exe files.

closed 2022-07-20 by tatarize

Sophist-UK, 2022-07-20:
AFAIK this has not been completed.

The suggestion was to move the CH341A.dll to somewhere in the source tree so it is included in run from source and not only included in exes. Then the DLL can be loaded if run from source not just if run from exe.

Sophist-UK, 2022-07-20:
I am no longer developing so I am no longer running from source - and since I only ran from source on my normal laptop and not my K40 controller, I never had the need personally.

My comment was for the benefit of anyone else who might want to run MK from source on Windows - like the OP did.