meerk40t/meerk40t #462: [Long-term Suggestion] Support additional file formats

Sophist-UK, 2021-08-01:
MK already supports EPS, so extend this to include support for PDF, Adobe Illustrator (.ai, .line).

Also, support TIFF and compressed SVG (.svcz).

tatarize, 2021-08-03:
There are other formats that should likely be supported like gcode, which I think has some support but isn't well explored. Tiff might be supported in Pillow. EPS only works if ghostscript is installed. I'm not quite sure what that's about. But, I think TIFF support would take a very minor change.

tatarize, 2021-08-03:
https://github.com/meerk40t/meerk40t/commit/8f1b1fff2c03742e726bce4161240c99019a29a6 That might actually do it for TIFF.

.svgz would also be fairly easy. It's a zlib stream of the svg file format.

tatarize, 2021-08-03:
https://github.com/meerk40t/meerk40t/commit/ca1552bef1275c749d5f5641166ed1cadeeb6105 That should do it for svgz too.

tatarize, 2021-08-03:
* svgz is a gzip stream not zlib

Sophist-UK, 2021-08-04:
Wow - wasn't expecting it anywhere near that quick.

closed 2021-08-08 by tatarize