meerk40t/meerk40t #464: Add means of installing libusb drivers

Sophist-UK, 2021-08-06:
If you want LibUsb it is a pain to have to install whisperer.

Can we have a means in mk or perhaps in an installer or even another executable, to install the o/s libusb drivers?

tatarize, 2021-08-06:
Maybe. Meerk40t doesn't install though. I guess it could but dunno.

Sophist-UK, 2021-08-06:
Driver installs require elevation to administrator, so I would recommend either a separate executable just to do this or we build an installer for MK (by e.g. plagiarising another python project like Picard that builds an installer for a similar range of OS') that also installs LibUSB drivers at the same time - which would also allow creation of shortcuts and clean-up of previous registry values and other stuff at the same time.

tatarize, 2021-08-06:
Seems like this could maybe be a @joerlane thing. Not that he's a windows aficionado but it's compiling a kind of install thing. Unless there's some python call to start installing drivers and modifying the system settings. I really didn't look into how Whisperer did that. I just supported everything and rarely hit an issue. If you have MK and plugin the laser, it will control the laser... regardless what driver it uses. (or that's the plan).

Sophist-UK, 2021-08-06:
Ok - given what you just said perhaps the best way of dealing with this is as follows:

a. Wait until I have time, and I will plagiarise Picard to create multiple installers (for various O/S) for MK - but without the LibUSB drivers.
b. Someone else, maybe @joerlane can then add the drivers.

Mte90, 2023-05-26:
Ref: https://github.com/meerk40t/meerk40t/issues/1931

Maybe use the same code fro mthe k40 whisperer? in this way the project use the same version for windows in case.