meerk40t/meerk40t #465: v0.7.0 RC8: Please report any issues found here

Sophist-UK, 2021-08-06:
# PLEASE DO ALL FURTHER TESTING ON v0.7.0b33 RC-8
The latest beta release is v0.7.0beta32 Release Candidate 7. This issue is for reporting issues with this specific version of MeerK40t.

Please do all further testing on this release. Please do not report any further bugs with previous beta releases. Do give any other types of feedback, questions, concerns or comments.

https://github.com/meerk40t/meerk40t/releases/tag/0.7.0-beta33

# CAUTION
**CAUTION:** Please note that (depending on your design) if you use Merge Operations after you have used Duplicate Operation, or you use Merge Passes when you have set Operation Passes > 1, these can result in the laser beam burning the same path several times in quick succession without allowing time for the material to cool down in between. This can have unintended consequences. **Please do not use this feature of the beta whilst your K40 is unattended.**

Sophist-UK, 2021-08-06:
Review using mock-USB or simulation:
# Bugs
1. With Optimise, spurious travel moves to SVG start/end of closed path still evident in simulation (and presumably in real cuts causing significant twitching). [image] In RC7, in burns like this (with many small circles) twitches accounted for (perhaps) 1/2 of the total time of the burn - some were due to these, but others were likely hardware related.

Sophist-UK, 2021-08-07:
@tatarize : I have not tracked this down as yet but I have a feeling that the cause of the spurious travel moves may be that when a closed path is flattened, the "Z" that marks the path as closed (which I think is a Close object in SVGelements) is being yielded as some sort of extra node - or possibly that SVGelements is not checking whether the start and end points of the segments are the same before inserting an extra segment to close the path.

Sophist-UK, 2021-08-07:
@tatarize Just ran it from source with the forward preference code and it no longer has these extra travel moves showing in simulation. But they are showing in RC8 so either a commit since the release has fixed this or it is something about versions being slightly different or a glitch in executables.

Sophist-UK, 2021-08-07:
# BUG
1. RC7 and RC8 are taking minutes to close when I have an SVG with a reasonable number of elements loaded. RC6 did not have this problem.

tatarize, 2021-08-08:
Probably a reversed path thing. If it occurs only in sim it would strongly indicate that I am getting the "start" rather than the point that qualifies as start depending on the cutobject's backwards or forwards state.

Sophist-UK, 2021-08-08:
It occurred (on previous versions) in execute as well. I have definitely narrowed it down to the single-character forward path change from preventing it happening in the specific example I am testing with even though the direction of travel is the same in both cases, so it can only be addressing the symptom and not the root cause. Here is my current theory:

When you convert the closed circle which has (say) 4 quadrant segments into individual Cut code segments, the Z / Close SVGelements object is converted into a zero length segment with start and end points identical. Lets call these segments s1, s2, s3, s4 and sz in this sequence. Without my fix it burns s2, s3, s4, it then selects s1, and then has to do a move to burn sz. With my fix it selects s2, s3, s4, sz, s1 and doesn't need the travel move.

So we need two or three fixes:

a. Only create a close cut segment if the start and end points are not identical;
b. Change the new <= comparison in my fix so that if it is = then you select the shortest of the 2 forward paths - so it will always do sz before s1.
c. Possibly eliminate zero length segments - though I think you have special code for dots (which I think are zero length segments) so perhaps not.

Sophist-UK, 2021-08-08:
#471 submitted for b.

Sophist-UK, 2021-08-10:
# UI
1. Tree/Operations/Add Operation adds an unknown operation. UI would be better with a submenu for Cut/Engrave/Raster/Image.

Sophist-UK, 2021-08-11:
# UI
1. White operations displayed in tree with white text on a white background. Whiteish text on a white background needs background to be replaced with a black background.

tatarize, 2021-08-12:
```
  File "C:\Users\...\meerk40t-main\meerk40t-main\meerk40t\gui\wxmeerk40t.py", line 719, in __set_panes
    panel = SpoolerPanel(self, wx.ID_ANY, context=self.context)
  File "C:\Users\...\meerk40t-main\meerk40t-main\meerk40t\gui\panes\spoolerpanel.py", line 17, in __init__
    index = spools.index(selected_spooler)
ValueError: '?' is not in list
```
x3

```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
  File "mk40t.py", line 13, in <module>
  File "meerk40t\main.py", line 421, in run
  File "meerk40t\kernel.py", line 869, in bootstrap
  File "meerk40t\gui\wxmeerk40t.py", line 201, in plugin
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\gui\wxmeerk40t.py", line 4146, in window_open
  File "meerk40t\kernel.py", line 710, in <lambda>
  File "meerk40t\gui\wxmeerk40t.py", line 4139, in window_open
  File "meerk40t\kernel.py", line 481, in open
  File "meerk40t\kernel.py", line 515, in open_as
  File "meerk40t\gui\wxmeerk40t.py", line 390, in __init__
  File "meerk40t\gui\wxmeerk40t.py", line 1089, in __set_ribbonbar
  File "meerk40t\gui\icons.py", line 16, in GetBitmap
  File "wx\lib\embeddedimage.py", line 64, in GetImage
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(0, 0), "C") == 0" failed at ..\..\src\common\intl.cpp(1694) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!
```
x4

```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
...
  File "D:\MeerK40t-64\Source\meerk40t-main\meerk40t\core\elements.py", line 4536, in classify
    if Color.distance_sq("black", element_color) <= 2304 or isinstance(element, SVGText): # if element_colour == Color("black"):
  File "D:\MeerK40t-64\Source\meerk40t-main\meerk40t\svgelements.py", line 1903, in distance_sq
    red_mean = int((c1.red + c2.red) / 2.0)
TypeError: unsupported operand type(s) for +: 'int' and 'NoneType'
```
I think this is sophist crashing from source.

```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
...
  File "meerk40t\kernel.py", line 481, in open
  File "meerk40t\kernel.py", line 515, in open_as
  File "meerk40t\gui\wxmeerk40t.py", line 390, in __init__
  File "meerk40t\gui\wxmeerk40t.py", line 1089, in __set_ribbonbar
  File "meerk40t\gui\icons.py", line 16, in GetBitmap
  File "wx\lib\embeddedimage.py", line 64, in GetImage
wx._core.wxAssertionError: C++ assertion "wxString::Format("%.3f", 1.23).find(str) != wxString::npos" failed at ..\..\src\common\intl.cpp(1633) in `anonymous-namespace'::GetInfoFromLCID(): Decimal separator mismatch -- did you use setlocale()?If so, use wxLocale to change the locale instead.
```
x2

```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
  File "meerk40t\gui\executejob.py", line 638, in on_button_start
  File "meerk40t\kernel.py", line 134, in __call__
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\core\planner.py", line 721, in plan_blob
  File "meerk40t\core\planner.py", line 83, in blob
  File "meerk40t\core\cutcode.py", line 281, in __init__
  File "meerk40t\core\cutcode.py", line 225, in __init__
[... 126 lines cut ...]
  File "PIL\Image.py", line 2483, in transform
  File "PIL\Image.py", line 2682, in new
MemoryError
```

Big image I think.

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "wx\core.py", line 3407, in <lambda>
  File "meerk40t\gui\panes\camerapanel.py", line 156, in update_camera_frame
RuntimeError: Failed to gain raw access to bitmap data.
```

Not sure. I could likely catch it though.

```
MeerK40t crash log. Version: 0.7.0 RC-6 on darwin
Traceback (most recent call last):
  File "meerk40t/gui/wxmeerk40t.pyc", line 2732, in on_click_open
  File "meerk40t/core/elements.pyc", line 4503, in load_types
NameError: name '_' is not defined
```
X3 -- Old mac build.

Sophist-UK, 2021-08-12:
# Bug
1. Simulation - move play progress slider fully left and some stuff is incorrectly shown.

Sophist-UK, 2021-08-13:
# Bug (significant)
1. MK does not break apart compound paths i.e. `M stuff M stuff` before deciding which paths are outer paths.

Sophist-UK, 2021-08-14:
# Backwards Compatibility
1. Missing Step & Repeat

Sophist-UK, 2021-08-15:
# Missing functionality
1. Save and restore Execute optimisation options with Operations

Sophist-UK, 2021-08-16:
# Bugs
1. Today, several very slow travel moves
2. Today, several times it did inner cuts after outer cut despite Outer Cuts Last being set.

Sophist-UK, 2021-08-17:
Bugs
1. When burn is executing, UI is unresponsive. (Did a simulate - got stutters. Could not then close simulation window or do anything else until burn finished an hour later, when everything came back to life.)

Sophist-UK, 2021-08-18:
# UI
1. Execute Job - Add time estimate display after optimise step.

Sophist-UK, 2021-08-18:
# Bugs
1. Raster ops in Simulation scene are mis-scaled.
[image]
[image]
[Falcoa sign LASER7.zip](https://github.com/meerk40t/meerk40t/files/7006736/Falcoa.sign.LASER7.zip)

tatarize, 2021-08-18:
Yeah, it's an issue with the raster2 not being raster1 and is making the sim images half size when really they should be 2x that size. Certainly a great testcase though should be fixable.

Sophist-UK, 2021-08-18:
This last issue is fixed in #498.

Sophist-UK, 2021-08-22:
# Bug
1. Remove of file node when elements are in more than one operation does not remove elements from one of the operations. Note: This is in RC8 and when running from source with new classify. I suspect that there is an exception occurring which is not being reported. (Aaaaarrrrrggggghhhhhh!!!!!!!!!!!!!!!!)

Sophist-UK, 2021-08-22:
# Bug
1. Open SVG with saved notes and Notes pane and Automatically Open Notes checked - Notes window opens as expected with notes but notes pane is empty.

tatarize, 2021-08-23:
```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
  File "meerk40t\gui\mwindow.py", line 72, in on_close
  File "meerk40t\kernel.py", line 545, in close
  File "meerk40t\gui\mwindow.py", line 92, in finalize
  File "meerk40t\gui\wxmeerk40t.py", line 2173, in window_close
  File "meerk40t\kernel.py", line 134, in __call__
  File "meerk40t\kernel.py", line 1770, in console
  File "meerk40t\kernel.py", line 1856, in _console_parse
  File "meerk40t\kernel.py", line 1209, in inner
  File "meerk40t\kernel.py", line 2463, in shutdown
  File "meerk40t\kernel.py", line 917, in shutdown
  File "meerk40t\kernel.py", line 1684, in process_queue
  File "meerk40t\gui\navigation.py", line 578, in on_emphasized_elements_changed
  File "meerk40t\gui\navigation.py", line 583, in update_matrix_text
RuntimeError: wrapped C/C++ object of type TextCtrl has been deleted
```

```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2537, in on_drop_file
  File "meerk40t\gui\wxmeerk40t.py", line 2501, in load
  File "meerk40t\core\elements.py", line 4536, in load
  File "meerk40t\image\imagetools.py", line 1672, in load
  File "meerk40t\svgelements.py", line 7777, in load
  File "meerk40t\svgelements.py", line 7804, in load_file
  File "PIL\Image.py", line 3009, in open
  File "PIL\Image.py", line 2996, in _open_core
  File "PIL\Image.py", line 2906, in _decompression_bomb_check
PIL.Image.DecompressionBombError: Image size (253762860 pixels) exceeds limit of 178956970 pixels, could be decompression bomb DOS attack.
```

Very large image. 178 megapixel+

```
MeerK40t crash log. Version: 0.7.0 RC-6 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 3622, in on_item_activated
  File "meerk40t\gui\wxmeerk40t.py", line 3631, in activated_node
  File "meerk40t\kernel.py", line 472, in open
  File "meerk40t\kernel.py", line 494, in open_as
  File "meerk40t\gui\operationproperty.py", line 118, in restore
  File "meerk40t\gui\operationproperty.py", line 180, in set_widgets
RuntimeError: wrapped C/C++ object of type ComboBox has been deleted
```
X3

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\gui\executejob.py", line 611, in on_listbox_operation_dclick
  File "meerk40t\kernel.py", line 481, in open
  File "meerk40t\kernel.py", line 503, in open_as
  File "meerk40t\gui\operationproperty.py", line 118, in restore
  File "meerk40t\gui\operationproperty.py", line 182, in set_widgets
RuntimeError: wrapped C/C++ object of type ComboBox has been deleted
```
X1: similar crash as above, caused by double click rather than on activate. Both are old.

```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1460, in run
  File "meerk40t\core\drivers.py", line 122, in _driver_threaded
  File "meerk40t\core\drivers.py", line 133, in _process_spooled_item
  File "meerk40t\device\grbl\grbldevice.py", line 229, in plotplanner_process
  File "meerk40t\device\grbl\grbldevice.py", line 179, in move
  File "meerk40t\device\lhystudios\lhystudiosdevice.py", line 1570, in write
TypeError: must be str, not bytes
```

GRBL + Lhystudios = crash.

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1460, in run
  File "meerk40t\core\output.py", line 121, in _sending
  File "meerk40t\core\output.py", line 87, in connect
TimeoutError: [WinError 10060] Ein Verbindungsversuch ist fehlgeschlagen, da die Gegenstelle nach einer bestimmten Zeitspanne nicht richtig reagiert hat, oder die hergestellte Verbindung war fehlerhaft, da der verbundene Host nicht reagiert hat
```
X3 Timeout error. Old. I think I fixed this.

```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2794, in on_click_open
  File "meerk40t\gui\wxmeerk40t.py", line 2501, in load
  File "meerk40t\core\elements.py", line 4536, in load
  File "meerk40t\core\svg_io.py", line 284, in load
  File "meerk40t\svgelements.py", line 8378, in parse
  File "re.py", line 223, in findall
TypeError: expected string or bytes-like object
```

```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2794, in on_click_open
  File "meerk40t\gui\wxmeerk40t.py", line 2501, in load
  File "meerk40t\core\elements.py", line 4536, in load
  File "meerk40t\core\svg_io.py", line 284, in load
  File "meerk40t\svgelements.py", line 8333, in parse
AttributeError: 'NoneType' object has no attribute 'append'
```

```
MeerK40t crash log. Version: 0.7.0 RC-7 on win32
Traceback (most recent call last):
  File "meerk40t\kernel.py", line 1460, in run
  File "meerk40t\core\output.py", line 125, in _sending
OSError: [WinError 10057] Eine Anforderung zum Senden oder Empfangen von Daten wurde verhindert, da der Socket nicht verbunden ist und (beim Senden Ã¼ber einen Datagrammsocket mit einem sendto-Aufruf) keine Adresse angegeben wurde
```

X3 Again was caught. Old. Probably Sven.

```
MeerK40t crash log. Version: 0.7.0 RC-8 on win32
Traceback (most recent call last):
  File "meerk40t\gui\wxmeerk40t.py", line 2794, in on_click_open
  File "meerk40t\gui\wxmeerk40t.py", line 2501, in load
  File "meerk40t\core\elements.py", line 4536, in load
TypeError: load() takes 2 positional arguments but 3 were given
```

EGV file loading lacks the new style of positional elements. Fixing...

Sophist-UK, 2021-08-23:
> MeerK40t crash log. Version: 0.7.0 RC-6 on win32
> Traceback (most recent call last):
>   File "meerk40t\gui\wxmeerk40t.py", line 3622, in on_item_activated
>   File "meerk40t\gui\wxmeerk40t.py", line [...]

This might have been me deleting an operation when the OperationProperty window was open or similar.

Sophist-UK, 2021-08-25:
# Bug
1. Merge passes without Merge operations does not merge passes.

Sophist-UK, 2021-08-28:
# Bug
1. With shift home position set to 0,0, the machine does NOT home to the same position that the hardware homes to when it powers up. The Meerk40t home position is c. 0.058" to the right and 0.1" down. This may be a controller issue rather than MK. Is it possible to set negative offset for home?

Sophist-UK, 2021-08-28:
RC9 has now been released. Please do not report any further problems with RC8. Please try to repeat the issue with RC9 and then report it in https://github.com/meerk40t/meerk40t/issues/525. Thanks.

closed 2021-08-28 by Sophist-UK