meerk40t/meerk40t #480: Default Raster Step == 2?

tatarize, 2021-08-09:
I feel like users might be better off with a higher default raster step. In whisperer it's set to 2 by default. And so there's a few posts around that makes it seem like a MeerK40t issue. 

E.G.
https://www.facebook.com/100000948265949/videos/3095770987368225/

Where if the Raster's default step was 2, or even 3 it would stop this sort of naive hitch early on? It doesn't matter for rasterwizard images but when the raster operations render things like basic fills, it's *really* easy to hit a stutter condition.

tatarize, 2021-08-09:
The stutter ceiling is about 100 mm/s on default if set to 1 ppi. The default Raster Speed is 140mm/s so in theory the default Raster stutters.

Sophist-UK, 2021-08-11:
My vote would be to use a standard Step for rasters of 2 and to set the default Step in Raster Wizards to match.

tatarize, 2021-08-12:
RasterWizard sets the default to either 2 or 3 depending on the script. But, here it's just asking for trouble since the default raster setting is actually going to stutter.

Sophist-UK, 2021-08-13:
My point is that we should try to be consistent. The default for Raster steps should be the same as the default for Raster scripts.

Sophist-UK, 2021-08-16:
How fast do you need to Raster with step=2 to get stuttering? I suspect that instead of 2 bytes per mil, you now use 3 bytes per 2mil or 1.5 bytes per mil, so that would suggest that you can only raster at c. 150mm/s before stuttering at step=2. Is that the case?

tatarize, 2021-08-16:
Raster step 2 would stutter at about 208mm/s