meerk40t/meerk40t #481: operation settings window bugged

panp0d, 2021-08-09:
cant open engraving opions  [
[image]
]

tatarize, 2021-08-09:
If the window is somehow vanished I'd try tools, reset windows, and try again. I've not tried bugging a window, but sometimes when I run two screens it gets stuck at the other screen or something.

panp0d, 2021-08-09:
that fixed it thanks

tatarize, 2021-08-09:
The most likely reason is you did something weird with it and it remembered the setting there and hid under your toolbar or something weird. I'm not sure if there's other ways to break a window.

closed 2021-08-09 by panp0d

Sophist-UK, 2021-08-10:
As a note for the future, it would be good if MK could check that the window is actually positioned on the screen when it is opened and if it is no longer on a screen, do a reset just for that window.