meerk40t/meerk40t #482: Classification

tatarize, 2021-08-11:
https://github.com/meerk40t/meerk40t/wiki/Classification-Rules

Since several of these are questions and not strictly wiki-documentation which was the end goal of the classification articla. I'll answer here since it seems more appropriate. @Sophist-UK 

> Should we (programmatically) create Operations with non-zero transparency? I think we should always set Operation transparency to 0.

Meh, we could. I think 16 million colors is enough, but if we need a few billion more... I guess that'd be fine.

> How should we handle transparency in colour comparisons between elements and operations? I think we should only compare RGB and not transparency.

Also, a fine way to do that. I'm not sure how it's currently handled.

> What does transparency in fill do during Raster? I would hope that it adjusts the PPI in addition to the fill colour.

The transparency is treated according to the Porter-Goss image composting rules. A 50% transparent shape is rendered into the image exactly as it appears on screen with 50% the overlapping shape and 50% the shape under it, and that is rastered togetehr.

> What does transparency in stroke do during Cut / Engrave? At the moment, I don't think this makes a difference, however my vote would be to use this for PPI (enabled by default but with an Operation Properties Advanced option to disable it).
Alternative non-fill / fill rastering special cases

Hm. That might actually be actionable. I was always a bit reluctant since there are more features than attributes being mapped. But, the whole alpha==power makes both logical and reasonable sense. This wouldn't, however, require a change in operations though since it wouldn't be too hard to just combine it. So if you have a 50% transparent stroke in an item with 500 ppi you could give it, 25% power *and* you could set your power on a per-line basis regardless of color.

> Non-fill - I assume we raster a black Shape with no fill as if it has black fill.

Only if the raster operation is black. In which case it only gets put in that operation because it exactly matched the color.

> Does this make sense, or should we use this as a way of rastering a path rather than engraving it (by rastering the stroke and not rastering the non-existent fill) or making the user create two paths with a fill between?

You can give a line a stroke-width and make it somewhat non-trivial and rastering that makes a lot of sense.

Fill - If we have a stroke and fill of the same colour, the proposed algorithm below classifies the element to both Cut/Engrave and Raster operations. I propose that if the stroke and colour fills are the same then we only classify it to a Raster operation, and if the user wants it classified to both Cut/Engrave and Raster, then they need to change the stroke colour.

But, the color of the fill doesn't matter. It seems like if they wanted that they should get rid of the raster or something. The color of the fill matters for the rastering of the fill. The color of the stroke matters only if it matches the operation color.

---

Sophist-UK, 2021-08-11:
@tatarize I think you are right - an Issue **is** the best place to discuss this.

There is an issue with using element colour for classifying some Raster ops and ignoring it for others. The issue is this:

1. if you do exact matches for existing Raster operations as Phase 1, you really need to assume that these Raster ops are specific to that colour and add any remaining raster operations to different Raster operations
2. if you create a single Raster operation for remaining raster ops, it will have a color. Suppose some but not all of the of these remaining raster ops match this color, then if you add a new file or reclassify, this will then have an exact match for some and not for others and another Raster operation will be created of the same color. If you reclassify again, then the exact matches will be classified to both these Raster operations and yet another Raster operation will be created.

One (easily programmed classify) solution is to create Raster operations by element colour in the same way that we create vector operations by colour. But this then creates a major issue - which is that separate Raster ops do separate burns and you don't get fill overlays working in the same way. Example a blue raster with a white raster on top (with no strokes) will be classified into different colours and then it will then burn all the blue area rather than leaving the white area blank.

Another solution is to have a flag on Raster operations to define them as "default". Default Raster operations are never matched to element colour and are used for all rasters that are not explicitly matched to operations in Phase 1. The user will need to be able to set and clear this flag - so it would need to appear in Operation Properties. We could also then allow the default flag for vector operations too - so that vector operations that do not match in Phase 1 are added to default vector operations. **If** we had such a default flag, the algorithm would be:

1. Match element to existing operations exactly matching colors.
2. If not color matched to existing operations, add to default vector / raster ops with the same type if they already exist.
3. If a default operation does not exist, then new vector operations are created with exact colour, new raster operation is created as a default Raster operation. If user wants to classify vectors into a default operation, they would need to manually create one or convert an existing one to default by ticking the checkbox.

This would also have the benefit of having a way to prevent proliferation of large numbers of vector operations when the SVG has a large number of colours.

> > Non-fill - I assume we raster a black Shape with no fill as if it has black fill.
> 
> Only if the raster operation is black. In which case it only gets put in that operation because it exactly matched the color.

There are two use cases really:

1. User wants an easy way (using a color) to raster just the stroke rather than cut it.
2. User wants an east way (using a color) to raster the fill even though it is None.

What I can't really see is the justification for the second one - because setting a fill is easy. But it would be great to have a way to easily create rastered strokes rather than cutting/engraving them. (It is also more consistent - Redish stroke makes the stroke a Cut, Black (or blackish) makes the stroke a Raster, any other colour makes the stroke an Engrave. Nothing about Fill here complicating this.)

Personally I think Raster operations should burn elements as follows:

* Fill / No-Stroke - rasters the fill
* Fill / Stroke of different color - rasters the fill
* Fill / Stroke of same color - rasters the fill AND the stroke (using stroke width and endcap / corner shapes)
* Stroke/No fill - rasters the stroke (using stroke width and endcap / corner shapes)

So you if you want a blue doughnut, you can do it with one circular blue stroke with a big width rather than a big circle with no stroke and blue fill and another smaller circle with no stroke and white fill.

tatarize, 2021-08-11:
> if you do exact matches for existing Raster operations as Phase 1, you really need to assume that these Raster ops are specific to that colour and add any remaining raster operations to different Raster operations

Yes. Depending on the matches created. Black stroked objects creating Rasters would be a fine fallback. But, only because the object created would match all identical subsequent objects too.

> if you create a single Raster operation for remaining raster ops, it will have a color.

Probably black. Though I think None might be able to be set which means match nothing.

> Suppose some but not all of the of these remaining raster ops match this color, then if you add a new file or reclassify, this will then have an exact match for some and not for others and another [...]

If an operation matched *any* operation it is not eligible for fallback phase.

> One (easily programmed classify) solution is to create Raster operations by element colour in the same way that we create vector operations by colour. But this then creates a major issue - which is that separate Raster ops do separate burns and you don't get fill overlays working in the same way. 

The current methodology would be create a default black raster. But, if the stroke was non-black it would create an engrave and the fills would just be ignored and the engraves would have caught that color.

> This would also have the benefit of having a way to prevent proliferation of large numbers of vector operations when the SVG has a large number of colours.

But, this is already kind of expected behaviors. If you wanted them in the same operation they should have been the same color.

> Personally I think Raster operations should burn elements as follows:

This would be a pretty breaking change. The strokes of shapes are currently rastered. Specifically because the shape on screen is used as an image and that image is rastered. That's how rastering works. I use wxPython to make the image exactly like the one you see on screen.

tatarize, 2021-08-11:
I think classification is a lot of work for something that is by definition imperfect and cannot catch all or even most parts. It's a reasonable first pass at trying to guess user intent. And these should be changed by the user to elicit the desired end-results.

I think wider range red fallback colors are fine for classification, it's a guess and it's a guess for a color they don't have.

I think black lines creating a raster would be fine, or filled shapes creating a black-stroke raster would be fine. It would be perhaps disruptive but it wouldn't be a disaster.

And adding an Image op for images is fine. It might change some stuff, but it's not too big of a problem.

Adding stroke-color dots if they happen to have dots is certainly fine. They are really rare as an operation and wouldn't render anything anyway.

I think if we wanted to implement these changes without disruption we could add the new operations in Disabled mode which would give them a home without changing anything about how it worked before. 

I sort of like the ppi-stroke-opacity thing.

--- 

Ultimately the classification schemes are not great. They are not intended to catch everything, just create a seemingly reasonable guess. This should be augmented by more radical ideas rather than more hyper-specific classifications.

For example, RDWorks and Lightburn use a colored assigned bar. Where the bottom bar has a bunch of colors which are linked to specific operations. You select items in the scene. Hit the color button there and the elements there are now part of that operation. Some folks prefer the simplicity of Whisperer which calls red and only exact red a cut and blue and only exact blue an engrave and to raster literally everything else. --- It would not be impossible to *also* support these in addition to MeerK40t's current methods.

It would be a little strange but all this would need to do is the same sorts of thing and then send that data to the simulator or planner accordingly. But, generally if we had a better methodology than we currently have we could switch over to that without compatibility issues since it could still use operations+classification and Whisperer's simple buttons. The only problem would be the switch from getting the data from the tree vs. getting the data from the other methodology.

And if the goal was to improve classification we would be better off reducing it's usable by making save commands work more correctly and save the element assignments. That would make the classification into those elements *much* less important since you could put them where-ever you wanted them.

I think trying to make it wildly better when it's always going to be imperfect isn't worth that much effort. It's a guess at an automatic set of assignments when you can literally have *anything* set there. It gets you started, it's not supposed to be perfect, just sort of good.

Sophist-UK, 2021-08-11:
> I think classification is a lot of work for something that is by definition imperfect and cannot catch all or even most parts. It's a reasonable first pass at trying to guess user intent. And these should be changed by the user to elicit the desired end-results.

I disagree - I think classification is one of the **key** usability features that users will be looking for.

tatarize, 2021-08-11:
> I disagree - I think classification is one of the key usability features that users will be looking for.

I think that ultimately the better method is to make it easier for users to assign objects to particular operations, or give them a proper choice. You can't catch everything because you still need to assign speeds etc, to particular operations, etc. The reason I'm not very keen on changing things to produce any breaking changes is that people who already have some work might want it continue using that work without interruption. So larger changes in classification might mess that up.

I think it might be best if I corrected the saving routines and from there we could maybe better gauge how useful classification would be. Since one if the largest problems it has is being used as the default methodology for reloading files with operations. The failure to do this correctly results makes classification have a larger role than it maybe should.

Sophist-UK, 2021-08-11:
As I have pointed out, my proposed approach maintains the maximum amount of backwards compatibility, changing only things that are inconsistent and limiting functionality.

0.7.0 is already sufficiently different to 0.6.22 already to make backwards compatibility not an issue.

0.7.0 RC9 backwards compatibility with previous betas is important, but they are all still betas where things change, so IMO backwards compatibility should not be a sacred cow, and it is better to make changes now whilst we are still in beta than once we have a final 0.7.0 release.

Sophist-UK, 2022-02-15:
Closing as this seems to be stable.

closed 2022-03-28 by Sophist-UK