meerk40t/meerk40t #486: [Suggestion] Status bar usage

Sophist-UK, 2021-08-12:
The status bar is currently barely used.

This issue has been opened to prompt a discussion on how it should be used. Immediately 2 uses come to mind:

a. Status of executing burns - Job x of y, Progress x%, Time xx:xx remaining of yy:yy - USB error messages etc.
b. Description of last user action - e.g. Selected x paths which are in y operations, Raster (Gravy) with step 3 executed on "label" etc.   
If the GUI window is too small to show both simultaneously, switch back and forth every 5 secs.

tatarize, 2021-08-12:
I did some work towards a. I made the spooler not pre-eat the item it's working on. So the first item will be in the spooler while it's being worked on. I also added a spooler channel that will say when spooled items get pulled which might help towards the timing of such things in progress. I'm not sure how to get the rest of A finished. I'm sure it needs some additional hooks.

B sounds really hard. Make things give a bunch of information that they aren't currently giving.

Sophist-UK, 2021-08-13:
B. is a nice to have - got the idea from Inkscape.

tatarize, 2021-08-16:
Everything would need to send a signal if it did some functional change. And while something like that exists notifying things if they were modified, it would have to describe the change. This would also be useful for a proper undo-redo stack. Since it would inform the user when a functional change was made and another restore point would be needed. This would still be somewhat difficult. I think the best method of doing this is to store soft-references to basically all the states and give them names.

There's other methods of using the status bar, I think signals update them but it's rarely done. And might not have survived the devices update since they would all have origins now or whatnot.

closed 2022-07-20 by tatarize