meerk40t/meerk40t #488: SVG save converts non-Path Shapes to Paths

Sophist-UK, 2021-08-12:
When MK saves an SVG the following shapes are converted from native SVG shapes to Paths:
* Rect
* Ellipse
* Circle
* SimpleLine
* Polyline
* Polygon
Can we try to preserve these as the original type of Shape rather than converting them?

tatarize, 2021-08-12:
Probably in a save routine if they are still in that form it could be doable. I'm not sure the utility of that but yeah, it *should* be doable more-or-less.

tatarize, 2021-08-12:
Likely the save revamp would be more about assigning elements by `id` to particular operations. So they load and save correctly. And including things like command ops into the saving routine.

Sophist-UK, 2021-08-13:
in svg-io save the code explicitly converts Shapes that are not Paths to Paths.

This is **nothing** to do with assigning elements by id to particular operations (ids are not guaranteed in the svg spec, so you would have to assign them (uniquely) if they don't already have ids. This needs fixing regardless of saving elements into operations.

closed 2022-07-20 by tatarize